raspberrypi/linux #869: Need a place to document all the overlays

msperl, 2015-03-02:
As the number of overlays is growing we need a location where we document each overlay.

pelwell, 2015-03-02:
What's wrong with the existing location?

pelwell, 2015-03-02:
... which is found here: https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README
and referenced here: https://github.com/raspberrypi/documentation/blob/master/configuration/device-tree.md#35-supported-overlays-and-parameters

msperl, 2015-03-02:
It sits outside of the code where the overlay dts files are located (not the same repository), so adding a new overlay to linux would mean another merge request for the firmware to get documented

pelwell, 2015-03-02:
That sounds to me like a different issue. Who would you say the documentation is for?

msperl, 2015-03-03:
Well, if you want some documentation added when overlays get added (which typically require modules to get enabled) then all should sit in the same repo (which is linux) - and that is mostly my concern: incomplete documentation!

The firmware repo is mostly generated, so why not move those overlay documentation into the linux tree under Documentation and copy the bits and pieces over to firmware when you build the kernel?

Actually you could also just copy the overlay sources as well as the binaries to /boot/overlay and then have the requirement to have the documentation sit in a comment section in those sources.

notro, 2015-03-03:
This makes me remember that it has become quite easy to make PR's from the github webgui.
And this fits fine with textfiles that don't need compile testing.
Fork https://github.com/raspberrypi/firmware
Go to boot/overlays/README
Click pencil icon to edit
Click 'Preview changes' to see the diff.
Choose 'Create a new branch for this commit and start a pull request'
Add commit message and hit 'Propose file change'
(I didn't actually try that step, but I've done this when it was slightly different, for the pi docu repo)
Then when it's merged, just kill the fork.

pelwell, 2015-03-03:
@msperl As discussed elsewhere, I do intend to move the overlays and the documentation into one common home, and that probably shouldn't be in the kernel tree. But that task is a lower priority than some others at the moment, so until then I suggest you work with what is there now.

Ruffio, 2016-08-14:
@msperl has this issue been resolved? If yes, please close this issue. Thanks.

msperl, 2016-08-15:
Has that mentioned move by pelwell happened?

closed 2016-08-15 by msperl