raspberrypi/linux #870: Please enable CONFIG_ATA_OVER_ETH=m in bcmrpi_defconfig

Ferroin, 2015-03-04:
This enables building the ATA over Ethernet (AoE) module.  This protocol is used in some higher end SAN systems, as a higher efficiency alternative to iSCSI and NBD, and also for low-end shared storage setups because it requires no extra userspace to work.  The module itself won't get auto-loaded by default, and there are no dependencies that aren't already being built, so there should be no impact for most users.

At the moment, this is the last thing keeping me using a custom build of the kernel, it would be nice to be able to just use the stock one.

popcornmix, 2015-03-04:
Added to linux tree. Will be in next firmware update.

Ferroin, 2015-03-04:
Thanks, greatly appreciated.

closed 2015-03-04 by Ferroin

popcornmix, 2015-03-05:
This should be in latest rpi-update firmware.