raspberrypi/linux #871: ARCH_BCM2709: FREE_IRQS too small, fails to allocate all irqs

notro, 2015-03-05:
FREE_IRQS is only 32 in ARCH_BCM2709 compared to 128 in ARCH_BCM2708.
This leads too irq allocation failure for pinctrl-bcm2835 and others:
```
[    0.000000] NR_IRQS:480

<snip>

[    2.400490] irq: irq_create_mapping(0xb9a93000, 0x1f)
[    2.405553] irq: -> using domain @b9a93000
[    2.409664] irq: irq 31 on domain /soc/gpio mapped to virtual irq 479

[    2.416135] irq: irq_create_mapping(0xb9a93000, 0x20)
[    2.421211] irq: -> using domain @b9a93000
[    2.425320] irq: -> virq allocation failed

<snip>

[    2.736443] gpiochip_add: registered GPIOs 0 to 53 on device: pinctrl-bcm2835
```
Full boot log including stmpe device: https://gist.github.com/notro/7bbe1be9a7c71081756f

Setting FREE_IRQS to 128 in arch/arm/mach-bcm2709/include/mach/irqs.h solves the problem.
This is also solves the problem of getting the stmpe touch controller to work on the Pi2.

I have a pending PR, so I can't (easily) make a PR for this.

popcornmix, 2015-03-05:
SPARE_ALLOC_IRQS is 64 on 2708 and 32 on 2709 - make them both 64 too?

notro, 2015-03-05:
No harm in doing so, that space is used in non DT mode. Can't tell if it is absolutely necessary though.

popcornmix, 2015-03-05:
https://github.com/raspberrypi/linux/commit/56941b1cd8b6766d755a7252ebf7a5c6e17ec000

notro, 2015-03-05:
Impossible to beat that reaction time :-)
Thanks.

ladyada, 2015-03-05:
@notro @popcornmix you guys are *AMAZING* thank you so much!!!!

closed 2015-03-05 by notro