raspberrypi/linux #876: RAM management issue in pi 2

ric96, 2015-03-09:
This might not affect a lot of users but according to some reports Pi2 seems to hang while compiling software using the -j4 tag with Free RAM ammout dipping to as low as 15mb in some cases.
although -j2 works good.
Could the ram management be optimised somehow to reduce such instances?
Links:
  http://www.raspberrypi.org/forums/viewtopic.php?f=63&t=102680
http://www.raspberrypi.org/forums/viewtopic.php?f=63&t=103141

popcornmix, 2015-03-09:
Compiling (and linking in particular) is very memory intensive.
Potentially trying to run 4 of these in parallel may well exhaust memory.
(check demsg after problems to see if there are out-of-memory reports)

I suggest you enable a lot of swap (and expect slowdowns) or compile with fewer parallel jobs.

lurch, 2015-03-11:
You can (obviously) also edit `config.txt` and give less memory to the GPU, if you want to give a few more megabytes of RAM to the CPU.

closed 2015-03-22 by popcornmix