raspberrypi/linux #881: dwc_otg can't coexist with other USB drivers

ali1234, 2015-03-11:
dwc_otg exports usb_gadget_probe_driver function and patches out the udc-core module which normally exports it. This breaks other USB device controller drivers, most notably the virtual dummy_hcd ones, and causes the kernel compile to throw a load of errors if you try to build dwc_otg and gadget at the same time.

OpenWRT patched this issue:

https://lists.openwrt.org/pipermail/openwrt-devel/2012-November/017427.html

TheSin-, 2015-09-29:
I'm also seeing this issue in the latest 4.1.8, has anything been found here to allow use to build both together?  The referenced commit doesn't seems to even be part of the issue.

popcornmix, 2015-09-29:
Are you talking about raspberrypi/linux or some other tree? This linked openwrt patch doesn't apply to raspberrypi/linux (the files aren't even named the same).

The Pi kernel doesn't define CONFIG_USB_GADGET so just removing usb_gadget_probe_driver/usb_gadget_unregister_driver won't work.
Something like http://paste.ubuntu.com/12617280/ may be possible though. Does that help your case?

TheSin-, 2015-09-29:
I'm just saying that I can not build USB_GADGET with dwc_otg, if I build both when I try to load g_mass_storage I get a warning about udc_core exports duplicate symbol usb_gadget_probe_driver and g_mass_storage won't load.

I'm not saying the openwrt patch is a solution I didn't even want to try that cause I'm sure there is reason for it on the Pi, and it seems to be heavily dependant on dwc_otg.  One of the things I wanted to mess around with was g_mass_storage, and since you helped me with building my own kernel it's one of the first new modules I tried to build, so I'm using the default config I'm just adding
```
CONFIG_USB_MASS_STORAGE=m
CONFIG_USB_GADGET=m
```
builds fine, boots fine, but when I run modprobe g_mass_storage
```
udc_core: exports duplicate symbol usb_gadget_probe_driver (owned by kernel)
modprobe: ERROR: could not insert 'g_mass_storage': Exec format error
```
I'm still working on debuging/figuring things out, to be fair I'm also trying to figure out hidapi so I can control the brightness not the display I bought at the same time so my thoughts are a little scattered. Let me try a few more things and get back to you.

popcornmix, 2015-09-29:
Are you hoping to enable gadget mode for the in built USB ports or an external device of some sort?
Gadget mode is not possible on a model B Pi/Pi2 as there is a USB hub on the board.

ali1234, 2015-09-29:
Original bug reporter here. I want to use dummy_hcd. This is a virtual loopback driver that allows you to write develop gadget code that can optionally run in userspace, while looking like real hardware to the kernel. None of that works on the Pi kernel though.

popcornmix, 2015-09-29:
Does it work with the patch I linked that avoids the duplicate symbol?

TheSin-, 2015-09-29:
@popcornmix I was hoping to do some testing with external devices, I wasn't sure if I could get an external one to work if it was attached to the Pi Hub, I knew the Pi Hub was not capable, but was hoping to play around and see what if anything was possible.  Either way I can not build both due to that error.

I'm testing the patch right now

TheSin-, 2015-09-29:
okay so I tried with the patch and using

```
CONFIG_USB_MASS_STORAGE=m
CONFIG_USB_GADGET=y
```
built fine, booted fine and looks like it's loading fine (just need a device to test with now, but no major errors at least)
```
modprobe: ERROR: could not insert 'g_mass_storage': No such device
```
Going to make an other pass with gadget set to m next

hh, 2015-12-02:
Do we know if this or #882 or #883 or #884 get us anywhere with regards to gadget support on the RPi zero?

popcornmix, 2015-12-02:
I've not heard of success with gadget mode. The original (from 3 years ago) dwc_otg driver should support gadget mode (although it's never been used). It's quite likely that some of the fiq optimisations don't support gadget mode, so disabling the fiq would be advisable.
You'll want to revert https://github.com/raspberrypi/linux/commit/59e76bb7e2936acd74938bb385f0884e34b91d72 to avoid forcing host mode (it should then use OTG cable to determine host/gadget mode).

Ruffio, 2016-08-14:
@ali1234 has your issue been resolved? If yes, then please close this issue.

P33M, 2017-05-04:
Gadget is primarily supported by the upstream dwc2 driver. dwc_otg should only be used for host mode.

closed 2017-05-04 by P33M