raspberrypi/linux #882: dwc_otg -DDEBUG option breaks build.

ali1234, 2015-03-11:
In drivers/usb/host/dwc_otg/Makefile uncomment this line:

```
#ccflags-y	+= -DDEBUG
```

Result: kernel fails to build:

```
al@al-desktop:~/piusb/linux$ make
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
make[1]: `include/generated/mach-types.h' is up to date.
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  CHK     kernel/config_data.h
  CC      drivers/usb/host/dwc_otg/dwc_otg_driver.o
  CC      drivers/usb/host/dwc_otg/dwc_otg_attr.o
drivers/usb/host/dwc_otg/dwc_otg_attr.c: In function ‘spramdump_show’:
drivers/usb/host/dwc_otg/dwc_otg_attr.c:921:27: warning: unused variable ‘otg_dev’ [-Wunused-variable]
  CC      drivers/usb/host/dwc_otg/dwc_otg_cil.o
  CC      drivers/usb/host/dwc_otg/dwc_otg_cil_intr.o
  CC      drivers/usb/host/dwc_otg/dwc_otg_pcd_linux.o
  CC      drivers/usb/host/dwc_otg/dwc_otg_pcd.o
  CC      drivers/usb/host/dwc_otg/dwc_otg_pcd_intr.o
  CC      drivers/usb/host/dwc_otg/dwc_otg_hcd.o
  CC      drivers/usb/host/dwc_otg/dwc_otg_hcd_linux.o
  CC      drivers/usb/host/dwc_otg/dwc_otg_hcd_intr.o
  CC      drivers/usb/host/dwc_otg/dwc_otg_hcd_queue.o
  CC      drivers/usb/host/dwc_otg/dwc_otg_hcd_ddma.o
  CC      drivers/usb/host/dwc_otg/dwc_otg_adp.o
  CC      drivers/usb/host/dwc_otg/dwc_otg_fiq_fsm.o
  LD      drivers/usb/host/dwc_otg/dwc_otg.o
  LD      drivers/usb/host/dwc_otg/built-in.o
  LD      drivers/usb/host/built-in.o
  LD      drivers/usb/built-in.o
  LD      drivers/built-in.o
  LINK    vmlinux
  LD      vmlinux.o
  MODPOST vmlinux.o
  GEN     .version
  CHK     include/generated/compile.h
  UPD     include/generated/compile.h
  CC      init/version.o
  LD      init/built-in.o
drivers/built-in.o: In function `dwc_otg_driver_remove':
:(.text+0xc2b18): undefined reference to `__DWC_DEBUG'
:(.text+0xc2b60): undefined reference to `__DWC_DEBUG'
:(.text+0xc2b74): undefined reference to `__DWC_DEBUG'
:(.text+0xc2bd8): undefined reference to `__DWC_DEBUG'
:(.text+0xc2bec): undefined reference to `__DWC_DEBUG'
drivers/built-in.o::(.text+0xc2c0c): more undefined references to `__DWC_DEBUG' follow
make: *** [vmlinux] Error 1
```

Ruffio, 2016-08-14:
@ali1234 has your issue been resolved? If yes, then please close this issue.

ali1234, 2016-08-23:
Not fixed.

popcornmix, 2016-08-24:
dwc_otg depends on dwc_common_port. You need to enable debug in there too.
https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/usb/host/dwc_common_port/Makefile#L8

ali1234, 2016-08-24:
Indeed. Fixed.

closed 2016-08-24 by ali1234