raspberrypi/linux #883: dwc-otg option -DDWC_DEVICE_ONLY breaks build

ali1234, 2015-03-11:
In drivers/usb/host/dwc_otg/Makefile uncomment the DWC_DEVICE_ONLY line:

```
# Use one of the following flags to compile the software in host-only or
# device-only mode.
#ccflags-y        += -DDWC_HOST_ONLY
#ccflags-y        += -DDWC_DEVICE_ONLY
```

Result: kernel build fails:

```
al@al-desktop:~/piusb/linux$ make
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
make[1]: `include/generated/mach-types.h' is up to date.
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  CHK     kernel/config_data.h
  CC      drivers/usb/host/dwc_otg/dwc_otg_driver.o
  CC      drivers/usb/host/dwc_otg/dwc_otg_attr.o
drivers/usb/host/dwc_otg/dwc_otg_attr.c: In function ‘spramdump_show’:
drivers/usb/host/dwc_otg/dwc_otg_attr.c:921:27: warning: unused variable ‘otg_dev’ [-Wunused-variable]
  CC      drivers/usb/host/dwc_otg/dwc_otg_cil.o
  CC      drivers/usb/host/dwc_otg/dwc_otg_cil_intr.o
drivers/usb/host/dwc_otg/dwc_otg_cil_intr.c:1321:112: error: unknown type name ‘dwc_otg_hcd_t’
drivers/usb/host/dwc_otg/dwc_otg_cil_intr.c: In function ‘dwc_otg_handle_common_intr’:
drivers/usb/host/dwc_otg/dwc_otg_cil_intr.c:1433:3: error: implicit declaration of function ‘dwc_otg_read_common_intr’ [-Werror=implicit-function-declaration]
drivers/usb/host/dwc_otg/dwc_otg_cil_intr.c:1537:17: error: ‘fiq_enable’ undeclared (first use in this function)
drivers/usb/host/dwc_otg/dwc_otg_cil_intr.c:1537:17: note: each undeclared identifier is reported only once for each function it appears in
cc1: some warnings being treated as errors
make[4]: *** [drivers/usb/host/dwc_otg/dwc_otg_cil_intr.o] Error 1
make[3]: *** [drivers/usb/host/dwc_otg] Error 2
make[2]: *** [drivers/usb/host] Error 2
make[1]: *** [drivers/usb] Error 2
make: *** [drivers] Error 2
```

Ruffio, 2016-08-15:
@ali1234 has your issue been resolved? If so, please close this issue. Thanks.

ali1234, 2016-08-24:
Not fixed.

P33M, 2017-05-04:
Using the upstream dwc2 driver is the recommended way of using OTG-device mode on the Pi. dwc_otg is pretty much only a host mode driver.

closed 2017-05-04 by P33M