raspberrypi/linux #885: Buffer I/O Error on external USB HDD during fsck.ext4 on Pi 2

Blindfreddy, 2015-03-12:
It seems that there is a problem with the USB subsystem using a USB HDD;   latest firmware on Pi.
No problem using same HDD attached to a ThinkPad running ubuntu 14.04.

steps to reproduce:

attach USB HDD to Pi 2
run fsck.ext4 as follows:

andrev@rpi4 ~ $ sudo fsck.ext4 -fpc /dev/sda1
Error reading block 159383557 (Attempt to read block from filesystem resulted in short read) while reading inode and block bitmaps.  

/dev/sda1: UNEXPECTED INCONSISTENCY; RUN fsck MANUALLY.
	(i.e., without -a or -p options)
^Candrev@rpi4 ~ $ sudo fsck.ext4 -fc /dev/sda1
e2fsck 1.42.5 (29-Jul-2012)
fsck.ext4: Attempt to read block from filesystem resulted in short read while trying to open /dev/sda1
Could this be a zero-length partition?

dmesg
....
[ 3130.315542] usb 1-1.3: reset high-speed USB device number 4 using dwc_otg
[ 3281.907729] sd 0:0:0:0: [sda]  
[ 3281.907786] Result: hostbyte=0x00 driverbyte=0x08
[ 3281.907806] sd 0:0:0:0: [sda]  
[ 3281.907819] Sense Key : 0x2 [current] 
[ 3281.907845] sd 0:0:0:0: [sda]  
[ 3281.907856] ASC=0x4 ASCQ=0x1
[ 3281.907875] sd 0:0:0:0: [sda] CDB: 
[ 3281.907886] cdb[0]=0x28: 28 00 4c 00 00 67 00 00 20 00
[ 3281.907951] blk_update_request: I/O error, dev sda, sector 1275068519
[ 3283.927722] sd 0:0:0:0: timing out command, waited 180s
[ 3283.927791] sd 0:0:0:0: [sda]  
[ 3283.927808] Result: hostbyte=0x00 driverbyte=0x08
[ 3283.927825] sd 0:0:0:0: [sda]  
[ 3283.927838] Sense Key : 0x2 [current] 
[ 3283.927863] sd 0:0:0:0: [sda]  
[ 3283.927876] ASC=0x4 ASCQ=0x1
[ 3283.927894] sd 0:0:0:0: [sda] CDB: 
[ 3283.927905] cdb[0]=0x28: 28 00 4c 00 00 e7 00 00 20 00
[ 3283.927971] blk_update_request: I/O error, dev sda, sector 1275068647
[ 3463.706079] sd 0:0:0:0: [sda]  
[ 3463.706132] Result: hostbyte=0x00 driverbyte=0x08
[ 3463.706151] sd 0:0:0:0: [sda]  
[ 3463.706165] Sense Key : 0x2 [current] 
[ 3463.706190] sd 0:0:0:0: [sda]  
[ 3463.706203] ASC=0x4 ASCQ=0x1
[ 3463.706221] sd 0:0:0:0: [sda] CDB: 
[ 3463.706232] cdb[0]=0x28: 28 00 4c 00 00 87 00 00 38 00
[ 3463.706298] blk_update_request: I/O error, dev sda, sector 1275068551
[ 3465.726032] sd 0:0:0:0: timing out command, waited 180s
[ 3465.726103] sd 0:0:0:0: [sda]  
[ 3465.726119] Result: hostbyte=0x00 driverbyte=0x08
[ 3465.726136] sd 0:0:0:0: [sda]  
[ 3465.726150] Sense Key : 0x2 [current] 
[ 3465.726176] sd 0:0:0:0: [sda]  
[ 3465.726188] ASC=0x4 ASCQ=0x1
[ 3465.726206] sd 0:0:0:0: [sda] CDB: 
[ 3465.726217] cdb[0]=0x28: 28 00 4c 00 00 67 00 00 08 00
[ 3465.726283] blk_update_request: I/O error, dev sda, sector 1275068519
[ 3465.726306] Buffer I/O error on dev sda1, logical block 637534228, async page read
[ 3465.726331] Buffer I/O error on dev sda1, logical block 637534229, async page read
[ 3465.726349] Buffer I/O error on dev sda1, logical block 637534230, async page read
[ 3465.726368] Buffer I/O error on dev sda1, logical block 637534231, async page read
[ 3601.300484] INFO: task fsck.ext4:3106 blocked for more than 120 seconds.
[ 3601.300514]       Not tainted 3.18.9-v7+ #767
[ 3601.300526] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 3601.300540] fsck.ext4       D 8051e698     0  3106   3105 0x00000000
[ 3601.300607] [<8051e698>] (__schedule) from [<8051eb88>] (schedule+0x40/0x8c)
[ 3601.300637] [<8051eb88>] (schedule) from [<8051ee7c>] (io_schedule+0xb0/0x11c)
[ 3601.300664] [<8051ee7c>] (io_schedule) from [<8051f630>] (bit_wait_io+0x44/0x68)
[... 123 lines cut ...]
[    4.232829] usb 1-1.3: Manufacturer: Western Digital
[    4.240402] usb 1-1.3: SerialNumber: 57583831454232564C363239
[    4.251416] usb-storage 1-1.3:1.0: USB Mass Storage device detected
[    4.261708] scsi host0: usb-storage 1-1.3:1.0
[    4.530750] usbcore: registered new interface driver uas
[    5.268798] scsi 0:0:0:0: Direct-Access     WD       My Passport 0748 1019 PQ: 0 ANSI: 6
[    5.284858] sd 0:0:0:0: [sda] Spinning up disk...
[    5.295901] scsi 0:0:0:1: Enclosure         WD       SES Device       1019 PQ: 0 ANSI: 6
[    5.337968] sd 0:0:0:0: Attached scsi generic sg0 type 0
[    5.345884] scsi 0:0:0:1: Attached scsi generic sg1 type 13
[    5.937560] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    6.231084] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    6.298017] ....
[   10.226008] random: dd urandom read with 127 bits of entropy available
[   10.337997] .
[   10.376331] random: nonblocking pool is initialized
[   10.682491] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   11.347980] .ready
[   11.348967] sd 0:0:0:0: [sda] 3906963456 512-byte logical blocks: (2.00 TB/1.81 TiB)
[   11.349481] sd 0:0:0:0: [sda] Write Protect is off
[   11.349505] sd 0:0:0:0: [sda] Mode Sense: 47 00 10 08
[   11.349946] sd 0:0:0:0: [sda] No Caching mode page found
[   11.349966] sd 0:0:0:0: [sda] Assuming drive cache: write through
[   11.533084]  sda: sda1
[   11.536837] sd 0:0:0:0: [sda] Attached SCSI disk

P33M, 2015-03-12:
A WD passport + fsck is a lot easier to troubleshoot than transferring heaps of data and then waiting for it to fall over.

What's on the disk? Is it a root filesystem? Lots of files?

Blindfreddy, 2015-03-12:
There is just one filesystem on the disk, type ext4, fully allocates all blocks.
fdisk:

Disk /dev/sdc: 2000.4 GB, 2000365289472 bytes
256 heads, 63 sectors/track, 242247 cylinders, total 3906963456 sectors
Units = sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disk identifier: 0x00000000

   Device Boot      Start         End      Blocks   Id  System
/dev/sdc1               1  3906963455  1953481727+  ee  GPT

Yes, there are lots of files on the disk.

Ruffio, 2016-08-15:
@Blindfreddy has your issue been resolved? If so, please close this issue. Thanks.

P33M, 2017-05-04:
Closing USB issues as OP has not updated the thread. Further comments by the OP stating that the issue is still present on latest rpi-update kernel will lead to a review.

closed 2017-05-04 by P33M