raspberrypi/linux #886: MT7601 USB WIFI driver support

bluenote73, 2015-03-12:
I'm not sure what needs to be included here, but it would be super helpful if the MT7601 had support in the kernel for newbs like me.  I got mine to work, but it's going to be a constant battle every time an upgrade happens, etc.  FWIW it was sold to me as one of the properly supported chipsets :/

Here's a link to their source:
http://www.mediatek.com/en/downloads/ MT7601

Hope this can be done.  TIA

mvadu, 2015-03-20:
There are two version of the firmware with "improved" code quality 
1. https://github.com/kuba-moo/mt7601u
2. https://github.com/porjo/mt7601 (gives the background on the chipset as well)

It may be wise to include one of these distributions than mediatek one, as they have not updated their source in years!

popcornmix, 2015-03-20:
If the dongle is not supported by the upstream linux kernel, then we're not going to include custom patches. The correct solution if for the driver to be submitted upstream where it can be peer reviewed. When accepted we can pull it into out tree.
https://www.kernel.org/doc/Documentation/SubmittingPatches

Until that happens, there seem to be instructions for building a suitable kernel here:  
http://groenholdt.net/Computers/RaspberryPi/MediaTek-MT7601-USB-WIFI-on-the-Raspberry-Pi/MediaTek-MT7601-USB-WIFI-on-the-Raspberry-Pi.html

closed 2015-03-20 by popcornmix

bluenote73, 2015-03-20:
Thanks for the info popcornmix.    Could you give the benefit of your knowledge and tell me where a user might most profitably lobby about this?     Do I need to be harassing the manufacturer?  Or is there a place where this stuff is discussed for the linux kernel?

thx

popcornmix, 2015-03-21:
If you want to push the driver upstream yourself, then the "Submitting Patches" link I gave is correct.
You should probably start by asking the manufacturer to submit upstream, and if they are not willing see if they will provide permission for you to do so.
The driver looks to be GPLv2, so is fine for going upstream, but you may need a "Signed Off By" acknowledgement from a MediaTek employee for it to be accepted upstream.

mvadu, 2015-04-12:
I finally managed to get to working toolset. I shared my instructions at https://github.com/porjo/mt7601/issues/18#issuecomment-90389343