raspberrypi/linux #888: USB Audio broken in kernel 3.18.7+

Halloran, 2015-03-15:
I run an RPi as an audio server: mpd 0.19.7 using ALSA to output to a Schiit Modi USB DAC, serving MP3's and FLAC files.

Recently, I ran apt-get update, apt-get upgrade, and my kernel went from 3.12.28+ to 3.18.7+, and this completely broke the audio output. Audio was being output more or less as white noise, not as slightly distorted, and this was the case with the aplay utility as well as with MPD. After a lot of troubleshooting with MPD, ALSA, etc., I used rpi-update to downgrade to the 3.12.28+ kernel, and all is working fine....

No idea it this is limited to USB audio, or if the problem might be deeper, but wanted to report it. Am happy to help with more system information, etc.

popcornmix, 2015-03-15:
```sudo rpi-update f74b92120e0d469fc5c2dc85b2b5718d877e1cbb```
is the latest 3.12 and 
```sudo rpi-update cad071ac50ddf8a9f3c7a5d808cfa0daf5effb45```
is the oldest 3.18 kernel.
Can you test those two and report which work?

Halloran, 2015-03-15:
Looks like the latest 3.12 works, the oldest 3.18 does not:

Ran:
```
sudo rpi-update f74b92120e0d469fc5c2dc85b2b5718d877e1cbb
```
got:
```
pi@framboise ~ $ uname -a
Linux framboise 3.12.36+ #737 PREEMPT Wed Jan 14 19:40:07 GMT 2015 armv6l GNU/Linux
```
Audio works

Ran
```
sudo rpi-update cad071ac50ddf8a9f3c7a5d808cfa0daf5effb45
```
got:
```
pi@framboise ~ $ uname -a
Linux framboise 3.18.3+ #739 PREEMPT Tue Jan 20 17:45:32 GMT 2015 armv6l GNU/Linux
```
Audio broken

popcornmix, 2015-03-15:
Are you able to test the USB on a different platform (e.g. x86) with 3.18 kernel?

The point where it broke was switching to a newer upstream kernel rather than a change to any Pi specific code. It may be this is a common problem for that sound card with newer firmware versions. 

Could be worth searching for similar problems on other platforms, in case there are known patches or workarounds.

Halloran, 2015-03-15:
Testing on x86 is not something that I can do anytime soon, though I'll try to figure out a way to do that.... I have looked through a lot of Audio sites, and it's hard to isolate the issues mentioned there, b/c getting USB audio working on Ubuntu seems to be pretty confusing in the first place....

But thanks for that feedback. I will try to do more digging, and see if I can make any progress.

In my case, the 3.12.x kernel is not really a problem, so I'll admit that my motivation here is pretty low.

Again, let me know if there are other things you'd like me to check, and if I can get an x86 machine running 3.18 I'll post results here.

P33M, 2015-03-16:
For reference, can you post a `lsusb -v` for the card in question?

At this point I'm not convinced this is a Pi-specific issue, namely because recent updates should have absolutely no effect on Pi 1 USB performance.

Halloran, 2015-03-16:
Cannot figure out which device it is, as the manufacturer is not showing up, but here are the 2 possibilities:

Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         0 
  bDeviceProtocol         1 
  bMaxPacketSize0        64
  idVendor           0x0424 Standard Microsystems Corp.
  idProduct          0xec00 
  bcdDevice            2.00
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol    255 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               4
[... 219 lines cut ...]
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      24
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               4

pssc, 2015-03-16:
USB audio working
Bus 001 Device 004: ID 08bb:2902 Texas Instruments Japan PCM2902 Audio Codec

Hardware        : BCM2708
Revision        : 000f

Linux raspberrypi 3.18.9

P33M, 2015-03-16:
@Halloran With audio playing (either MPD or aplay) can you post the output of `cat /proc/asound/card1/stream0` - and compare the 3.12 version with 3.18 to see if it's any different?

P33M, 2015-03-16:
@Halloran I've just noticed that the device is likely to use a High-Speed chipset (CM6631A). Can you post a `lsusb -t` to confirm?

sanjibm, 2015-03-16:
I can second this USB audio issue. I had to downgrade my kernel to 3.12 to get USB audio to work. BTW, I am also using a Schiit Modi USB DAC. Anything beyond 3.12 breaks the audio into crackles and corruption/noise.

This might have a connection with the nrpacks option in the snd-usb-audio module. I noticed that this option is no longer supported in the latest kernels. According to modinfo, this sets the: Max. number of packets per URB. (int). Setting this to "nrpacks=1" should allow latencies in the area of 4-6 msecs. 

So if it shows up in the kernel as a valid parameter, then all is well:

```shell
modinfo snd-usb-audio | grep ^parm:
```
If nrpacks isn't available, then USB audio doesn't seem to work with the Schiit Modi DAC; this option has been recommended for other popular USB DACs too.

I am not sure, but it appears that the nrpacks option was removed as the configuration for setting the number of packets per URB has been made automatic, hence this is no longer needed. Clearly something seems to be going wrong with this new implementation, or perhaps I am missing a replacement bit of configuration for the new kernels. Need to investigate a bit more

pssc, 2015-03-16:
> "Anything beyond 3.12 breaks the audio into crackles and corruption/noise."

This would seem to be related to buffer settings in the audio subsystem have you tried upping buffer time and period count / buffer size for reliable usb audio try periodCount=5 bufferTime=113378,
I had to significantly up these for USB audio (internal pc=3 bt=50).

P33M, 2015-03-16:
@pssc @sanjibm It it unlikely that these module parameters need altering from defaults. With 3.18 in the wild and packaged for the latest release of Raspbian, if these changes caused failures across *all* usb-audio devices then the entire world would be screaming on the forums. As of 3.18 the nrpacks parameter is redundant because the driver has a notionally saner algorithm for calculating the number of URBs required.

Both the reported problems are with a specific type of DAC. This DAC is unusual in being a high-speed device. There are differences with handling of packet intervals for high-speed Isochronous devices, which a move to 3.18 may have caused to change.

pssc, 2015-03-16:
ok well event with my low speed devices I had to alter the application settings (not kernel) to get sane usb audio, if the system was under any load.  I could also distort the USB audio by altering the spi bus frequency for the piftft screen (but I think that’s a different issue)
The above settings where tuned to cause ~3-4k buffer update for a given period, reducing the strain on the system, before this the some of the USB audio device I tried sounded like the world had ended. Other USB sound devices where tolerant of lower settings(but caused horribly distorted audio on others). but still significantly more than the defaults alsa settings for internal output.

Halloran, 2015-03-16:
@P33M I've had to travel so do not have access to the machine at the moment. Will get you the info you requested above, but it might be a couple of days.

pssc, 2015-03-16:
Looking at one of my home PI's the least tolerant device I have was a
Bus 001 Device 005: ID 0d8c:0103 C-Media Electronics, Inc. CM102-A+/102S+ Audio Controller
which looks like it may me similar to the chip set in the above mentioned DAC. 
With this device I also found a reboot didn’t recover from the distorting and it need a complete power removal from the pi to restore reasonable audio.

notro, 2015-03-16:
@pssc 
> I could also distort the USB audio by altering the spi bus frequency for the piftft screen

PiTFT (fbtft driver) and USB audio has been problematic: https://forums.adafruit.com/viewtopic.php?f=47&t=63696
Depending on use and config, the fbtft drivers can, during movie playback for instance, continuously send 4k chunks using DMA. Reducing the buffer size was shown to improve usb audio (txbuflen=512), but it reduced overall display fps.

pssc, 2015-03-16:
@notro Yes this tallies with my experance I have reliable operation uning usb audio and pitft as long as I dont alter the spi frequency ifrom 25Mhz and drop the fps=~12 to reflect this altering the spi frquency allways produced bad audio event with low fps options.  I am running a squeezeplay variant for audio and visual as well as fbcp(altered to copy for the fps the pitft is running at) to mirror onto the pitft and external usb sound of differnt variaties as well as via hdmi though and amp on one of my pi's

pssc, 2015-03-17:
@notro my though was that the ALSA or the  USB driver was reliant on the interrupt from the 25Mhz timer for some operation.

pssc, 2015-03-22:
I have the HDMI active for both audio and visual , as well as the fbtft and external USB sound, so that may well be enough to trigger a SRAM congestion issue msperl/spi-bcm2835#13 increasing the timer speed without raising the fps from 12, triggers the issues.

Halloran, 2015-03-22:
With kernel at 3.18.3+ (audio is broken)
```
pi@framboise-sud ~ $ cat /proc/asound/card1/stream0
Schiit Schiit USB Audio Device at usb-bcm2708_usb-1.2, high speed : USB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 2
    Packet Size = 378
    Momentary freq = 44131 Hz (0x5.8433)
    Feedback Format = 7.17
  Interface 1
    Altset 1
    Format: S16_LE
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000
    Data packet interval: 1000 us
  Interface 1
    Altset 2
    Format: S24_3LE
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000
    Data packet interval: 1000 us
```

With kernel at 3.12.36+ (audio working fine)
```
pi@framboise-sud ~ $ cat /proc/asound/card1/stream0
Schiit Schiit USB Audio Device at usb-bcm2708_usb-1.2, high speed : USB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 2
    Packet Size = 378
    Momentary freq = 44131 Hz (0x5.8433)
    Feedback Format = 7.17
  Interface 1
    Altset 1
    Format: S16_LE
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000
    Data packet interval: 1000 us
  Interface 1
    Altset 2
    Format: S24_3LE
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000
    Data packet interval: 1000 us
```

Don't see any difference here.
[note that this is a different host, but the hardware is essentially the same, and the problem seems exactly the same.]

Halloran, 2015-03-22:
Here is the lsusb -t:
```
pi@framboise-sud ~ $ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=hub, Driver=hub/3p, 480M
        |__ Port 1: Dev 3, If 0, Class=vend., Driver=smsc95xx, 480M
        |__ Port 2: Dev 4, If 0, Class=audio, Driver=snd-usb-audio, 480M
        |__ Port 2: Dev 4, If 1, Class=audio, Driver=snd-usb-audio, 480M
        |__ Port 2: Dev 4, If 2, Class=HID, Driver=usbhid, 480M
        |__ Port 3: Dev 5, If 0, Class=stor., Driver=usb-storage, 480M
```

Halloran, 2015-03-22:
@P33M your comment:
> There are differences with handling of packet intervals for high-speed Isochronous devices, which a 
> move to 3.18 may have caused to change.
seems the logical path to me. Is there any way to see packet handling intervals? Or to manually set them?

P33M, 2015-03-24:
The claimed interval is displayed in the output of the sysfs node (alternate setting for the device). Whether this actually what's coming out of the USB port is a different matter entirely. As you can see, there's no difference in the playback settings between 3.12 and 3.18.

I think at this point it needs one of these devices + the USB analyser to work out what's going on.

Halloran, 2015-03-25:
OK. I definitely do not have access to a USB analyzer. Sorry. Let me know if there's anything further I can do to help.

P33M, 2015-03-25:
I was meaning R-Pi's resident analyzer. I've ordered one of these DACs for inspection, as it may be common to all USB2.0 HS USB audio devices.

inadsan, 2015-03-25:
Try change use_mmap yes to no
mpd.conf
audio_output {
   type			"alsa"
   name			"My ALSA Device"
   use_mmap		"no"
}
with an application using libao it worked to me

Halloran, 2015-03-25:
The use_mmap param is not present in my mpd.conf, and the mpd documentation suggests that the default is no, but I will give it a try and post the results.

In the meantime, thanks for the suggestion!

P33M, 2015-03-30:
I now have a Schitt Modi 2.

Interestingly, there's a switch on the back that puts the device in "expert" mode.

It changes the descriptors like so:
```

Bus 001 Device 004: ID 0d8c:0319 C-Media Electronics, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x0d8c C-Media Electronics, Inc.
  idProduct          0x0319 
  bcdDevice            1.05
  iManufacturer           1 Schiit Audio
  iProduct                2 USB Modi Device
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          298
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          1 Audio
      bFunctionSubClass       0 
      bFunctionProtocol      32 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol     32 
      iInterface              2 USB Modi Device
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               2.00
        bCategory              10
        wTotalLength          256
        bmControl            0x00
      AudioControl Interface Descriptor:
        bLength                17
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bCSourceID             18
        bNrChannels             0
        bmChannelConfig   0x00000000
[... 266 lines cut ...]
          wDescriptorLength      24
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               4
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)
```

P33M, 2015-03-30:
```

Bus 001 Device 005: ID 0d8c:1319 C-Media Electronics, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0d8c C-Media Electronics, Inc.
  idProduct          0x1319 
  bcdDevice            1.05
  iManufacturer           1 Schiit Audio
  iProduct                2 USB Modi Device
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          214
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              2 USB Modi Device
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength           40
        bInCollection           1
        baInterfaceNr( 0)       1
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      3 (OUTPUT_TERMINAL)
        bTerminalID             7
        wTerminalType      0x0301 Speaker
        bAssocTerminal          0
        bSourceID              13
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                10
[... 161 lines cut ...]
          wDescriptorLength      24
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               4
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)
```

P33M, 2015-03-30:
This is the output on a B+ with 3.18:
[image]
And on 3.12:
[image]

On 3.12, I note that the audio actually plays ridiculously fast.

P33M, 2015-03-30:
@Halloran Can you try using the device with dwc_otg.fiq_fsm_enable=0?

Halloran, 2015-03-30:
Happy to give that a try. Where/how do I configure that?

P33M, 2015-03-30:
Add the parameter to /boot/cmdline.txt and reboot.

Halloran, 2015-04-01:
Yup. That works.

What are the implications of that?

P33M, 2015-04-01:
The fiq_fsm handler is incorrectly performing transfers to isoc OUT endpoints. It appears that it's blindly transferring data every microframe instead of strided per 2^(n-1) microframes. With Isoc IN endpoints, you can request data every microframe and the device just responds as-and-when.

This is why nrpacks=1 worked - one isochronous transfer per URB which hid the bug.

Without FIQ support, microframe jitter will cause variations in timing of the OUT packets which will probably cause sample rate skew. I'll look at implementing strided transfer support.

Halloran, 2015-04-01:
OK. Going back to the 3.12 kernel. Happy to test things out to help. And many thanks for working on this.

Ruffio, 2016-08-15:
@Halloran has your issue been resolved? If so, please close this issue. Thanks.

Halloran, 2016-08-15:
@Ruffio am I correct that you are asking if the issue is resolved in the 4.1 kernel? Or is there a 3 series kernel that I should be testing?

Ruffio, 2016-08-15:
@Halloran I would test with latest version (4.4)

Halloran, 2016-08-15:
@Ruffio will try to give that a go sometime this week. Sorry to have left this open so long.

Halloran, 2016-08-16:
Updated kernel to 4.4.17+ and the issue is resolved.

closed 2016-08-16 by Halloran