raspberrypi/linux #890: aplay and arecord

kciszewski, 2015-03-16:
In 3.12.33+ aplay and arecord (command line) don't work.

popcornmix, 2015-03-16:
3.12 is no linger supported. 3.18 is the stable version.
This is unlikely to be a kernel issue. You may be better off asking questions on the forum.

kciszewski, 2015-03-16:
So where do I get a 3.18 version?  I tried downloading from the Raspberry PI page last week, and the download didn't work.
--------------------------------------------

 

 
 
 it on GitHub.

popcornmix, 2015-03-16:
```
sudo apt-get update && sudo apt-get upgrade
```
is the usual way of updating an image. You can also:
http://www.raspberrypi.org/downloads/

and click "download zip" next to Raspbian which is working fine for me.

closed 2015-03-22 by popcornmix