raspberrypi/linux #891: Buildroot, R-Pi2 and linux 23c76b7f608e51258c6fca02aebeb5d588583149

Nathilion, 2015-03-16:
Hi,

I've been trying to create my own custom image with buildroot, using the new raspberrypi2_defconfig I just discovered. I have added a few packages to the config (iptables, samba and iptools) and started a build. Everything goes fine right up to point where it starts to clone your repository. In particular the revision with the signature 23c76b7f608e51258c6fca02aebeb5d588583149 . When it does the whole repository gets downloaded neatly in a tar.gz ball and subsequently unpacked. However, as soon as the whole cloning process is finished I get the error 'fatal: not a tree object'. After which the repo is deleted and buildroot tries to download an alternative from its own site and fails of course, thus ending the build process.

Here's what gets put onto the console:
<pre>>>> linux 23c76b7f608e51258c6fca02aebeb5d588583149 Downloading
Doing full clone
Cloning into bare repository 'linux-23c76b7f608e51258c6fca02aebeb5d588583149'...
POST git-upload-pack (gzip 1090 to 585 bytes)
remote: Counting objects: 4088791, done.
remote: Total 4088791 (delta 0), reused 0 (delta 0), pack-reused 4088791
Receiving objects: 100% (4088791/4088791), 1.14 GiB | 4.90 MiB/s, done.
Resolving deltas: 100% (3371132/3371132), done.
Checking connectivity... done.
fatal: not a tree object
--2015-03-15 21:20:57--  http://sources.buildroot.net/linux-23c76b7f608e51258c6fca02aebeb5d588583149.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 176.9.16.109
Connecting to sources.buildroot.net (sources.buildroot.net)|176.9.16.109|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2015-03-15 21:20:58 ERROR 404: Not Found.</pre>

I tried to overcome this by manually downloading the repository to an identically named directory in the 'output/build/' directory  and then switching back to the revision indicated by that hash. This trick usually works when sources fail to download (due to proxies, dead links, etc.) so I was hopeful it would work this time as well. For this I used the following commands:

<pre>git clone https://github.com/raspberrypi/linux.git linux-23c76b7f608e51258c6fca02aebeb5d588583149
cd linux-23c76b7f608e51258c6fca02aebeb5d588583149/
git reset --hard 23c76b7f608e51258c6fca02aebeb5d588583149
</pre>

The last command failed with the following error:
<pre>fatal: Could not parse object '23c76b7f608e51258c6fca02aebeb5d588583149'.</pre>
Trying a checkout also fails but with the error: <pre>fatal: reference is not a tree</pre>

This pretty much sums up everything I tried. The particular Buildroot revision is c95c8c928e2d822c155984e84b4c22f2d0388cc9 (March 10th).

popcornmix, 2015-03-16:
I'm not quite sure where your git hash came from but try this:
https://github.com/raspberrypi/linux/commit/11f2a1f8db91523fbcf10f8428f407cf3b37f936
or
```git reset --hard 11f2a1f8db91523fbcf10f8428f407cf3b37f936```

Nathilion, 2015-03-16:
It was this patch against the Buildroot repo that introduced that hash: http://git.buildroot.net/buildroot/commit/?id=79e8d730ad6d414f6a67724a68057e2025d3dda6 

I'll try out the new hash once I get back from work. And will update this issue with the results.

popcornmix, 2015-03-16:
Ah - they were using a has from the 3.19 tree.
The 3.18 tree is stable and gets merged, so hashes are permanent.
The 3.19 tree is unstable and will be rebased with each point update. It is unwise to use git hashes from the unstable tree. If a stable 3.19 is required I suggest you fork and point at your own tree.

Nathilion, 2015-03-16:
Thanks for the help! The reset to your hash worked. Like you said, though, I did feel more comfortable using the 3.18 stable tree so I switched the kernel headers to 3.18 in the toolchain and supplied the current head's hash (now that I know what to look for) in the kernel section in buildroot and that worked like a charm too. My first raspberry pi image just came out of the buildroot oven. Bon appetit!

closed 2015-03-16 by Nathilion

Eternity-theIndie, 2015-04-08:
Is there a working Raspberry 2 release? If so, how can I get it? I don't find the 11f2a1f8d... hash.
I guess that the latest stable release does not support RPi2(?).

Thanks!