raspberrypi/linux #894: zram compressed ramdisk for swap

pssc, 2015-03-17:
It makes checking out the kernel possible on my rev2 B pi and compilation somewhat quicker.
This has been requested before but wasn’t available as a module in 3.12 and had to be complied in.

CONFIG_ZSMALLOC=m
CONFIG_PGTABLE_MAPPING=y

CONFIG_ZRAM=m
CONFIG_ZRAM_LZ4_COMPRESS=y

CONFIG_LZ4_COMPRESS=m
CONFIG_LZ4_DECOMPRESS=m

popcornmix, 2015-03-17:
Do you have any benchmarks showing the improvement?
(e.g. time to compile an app with and without compressed ramdisk?)

I'll look into it. Enabling as a module on 3.18 should be possible.

pssc, 2015-03-17:
I only properly benchmarked to see if it was worth enabling  CONFIG_PGTABLE_MAPPING for this arch. it was a factor of ~two quicker.

pssc, 2015-03-17:
my git pull for the kernel falls over OOM without the swap enabled , with this enabled I don't end up using up by SDcard swapping.

jbruchon, 2015-03-22:
I am using Xbian and based on my experience, Xbian has some glitches and slowdowns because it swaps to and from the SD card occasionally. The swap is never more than about 105MB total. zram "I/O" performance is extremely high compared to even my Class 10 SD card. If I had access to the zram module, I'd be able to use it to eliminate my playback issues. There is no good reason to *not* have the zram module available, especially since the Pi is the type of system that needs zram swap the most.

Regarding benchmarks, compilation performance is a poor benchmark because it is not a typical load for most users (if compilation performance matters, you are cross-compiling anyway) and hitting swap may not occur at all. A much better test would be running various tests using a web browser such as Firefox which often consumes all available RAM and gets OOM killed often if swap is disabled.

popcornmix, 2015-03-22:
Should be enabled now. Can you rpi-update and test?

jbruchon, 2015-03-23:
I don't personally have rpi-update (xbian doesn't appear to have it.) Any alternative ways to obtain and test?

popcornmix, 2015-03-23:
@jbruchon You'll have to ask in xbian forums. I suspect they build their own kernel, so this change will have no effect (they can of course make a similar change themselves).

pssc, 2015-03-24:
I have this ok and working on one of my rev 2 B's that's under heavy load.

popcornmix, 2015-03-24:
Okay to close?

PeterPablo, 2015-03-24:
I have also updated. Though not being able to give quantitative benchmark results, I have the impression that performance has improved (apache server, busy with owncloud). No problematic messages in `syslog` as far as I observed.

closed 2015-03-31 by popcornmix

yutayu, 2020-12-11:
https://github.com/armbian/build/pull/2377
https://github.com/armbian/build/pull/2377/commits/e90a488f24d4ba7f0947930f485c4e69a09e9374

CONFIG_PGTABLE_MAPPING=y
It seems it causes error.
Like this ,https://github.com/raspberrypi/linux/issues/3839.