raspberrypi/linux #895: kernel message on every ioctl(console_fd, FBIO_WAITFORVSYNC, 0)

pssc, 2015-03-17:
```
diff --git a/drivers/video/fbdev/bcm2708_fb.c b/drivers/video/fbdev/bcm2708_fb.c
index a443666..6b811d7 100644
--- a/drivers/video/fbdev/bcm2708_fb.c
+++ b/drivers/video/fbdev/bcm2708_fb.c
@@ -424,10 +424,10 @@ static int bcm2708_ioctl(struct fb_info *info, unsigned int cmd, unsigned long a
 
                bcm_mailbox_property(&p, p[0]);
 
-               pr_info("bcm2708_fb_ioctl %x,%lx returns=%d p[1]=0x%x\n", cmd, arg, p[5], p[1]);
-
                if ( p[1] == VCMSG_REQUEST_SUCCESSFUL )
                        result = 0;
+               else
+                       pr_info("bcm2708_fb_ioctl %x,%lx returns=%d p[1]=0x%x\n", cmd, arg, p[5], p[1]);
        }
        return result;
 }

```
Thanks

Phill.

popcornmix, 2015-03-17:
Okay

popcornmix, 2015-03-22:
Can you rpi-update and test?

closed 2015-03-31 by popcornmix