raspberrypi/linux #897: Raspberry Pi 2 USB disconnect

Saudumm, 2015-03-17:
Every time I connect a USB soundcard to my Raspberry Pi 2 Model B, the USB controller disconnects after a random time, if sound is played via the USB soundcard. Also, you can hear a popping noise every now and then from the speakers when sound is played. Raspbian continues to run, except for every USB device or the Ethernet port.
I found dozens of threads for "Failed to read reg index 0x00000114" but they all are a year old and there the problem got fixed in April 2014 with a firmware update. Maybe this has something to do with the USB controller on the Raspberry Pi 2 specifically?

The problem does not occur when I connect the speakers directly to the headphone jack and disconnect the sound card.
The sound card is connected to a powered USB-Hub (but problem does also occur if i connect it directly to the Pi)
Raspberry receives enough power (constantly 5V)
Doesn't matter if a WiFi Stick or Ethernet cable is connected.
Tested two different USB-soundcards that are confirmed working (one from adafruit and a Creative Play sound card)
I'm at a loss here. Headphone jack is working fine, but sound isn't that good.

I uploaded the interesting part of the syslog to pastebin: http://pastebin.com/yn4PUUwm

clivem, 2015-03-18:
This is a random problem on both bcm2708 and bcm2709 flavours. I've experienced it on A+, B, B+, and 2B. It seems to be load related and USB just appears to shut down.

[87935.712410] usb usb1-port1: disabled by hub (EMI?), re-enabling...
[87935.718747] usb 1-1: USB disconnect, device number 2
[87935.725009] usb 1-1.1: USB disconnect, device number 3
[87935.730712] smsc95xx 1-1.1:1.0 eth0: unregister 'smsc95xx' usb-bcm2708_usb-1.1, smsc95xx USB 2.0 Ethernet
[87935.740569] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[87935.773439] usb 1-1.3: USB disconnect, device number 4
[87935.972407] rtw_cmd_thread: DriverStopped(0) SurpriseRemoved(1) break at line 482
[87936.172255] Indeed it is in host mode hprt0 = 00001901
[87936.292237] Indeed it is in host mode hprt0 = 00001501
[87936.562374] Indeed it is in host mode hprt0 = 00001101
[87936.872274] Indeed it is in host mode hprt0 = 00001101
[87937.142283] Indeed it is in host mode hprt0 = 00001101
[87937.412247] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[87937.419121] Indeed it is in host mode hprt0 = 00001101
[87937.542288] Indeed it is in host mode hprt0 = 00001101
[87937.812271] Indeed it is in host mode hprt0 = 00001101
[87938.082305] Indeed it is in host mode hprt0 = 00001101
[87938.352329] Indeed it is in host mode hprt0 = 00001101
[87938.622300] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[87938.629168] Indeed it is in host mode hprt0 = 00001101
[87938.752612] Indeed it is in host mode hprt0 = 00001101
[87939.072319] Indeed it is in host mode hprt0 = 00001101
[87939.342351] Indeed it is in host mode hprt0 = 00001101
[87939.612386] Indeed it is in host mode hprt0 = 00001101
[87939.882444] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[87939.902430] Indeed it is in host mode hprt0 = 00001101
[87940.072381] Indeed it is in host mode hprt0 = 00001101
[87940.382463] Indeed it is in host mode hprt0 = 00001101
[87940.672372] Indeed it is in host mode hprt0 = 00001101
[87940.942367] Indeed it is in host mode hprt0 = 00001101
[87941.213600] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[87941.231396] usb usb1-port1: unable to enumerate USB device

pssc, 2015-03-18:
I to have a Pi rev2 B that gives up on USB overnight, I Further interaction via some gpio based buttons usually causes it to wedge, Its displaying a clock so  I can tell when its stopped.

P33M, 2015-03-18:
@Saudumm is your problem specific to Pi 2? Do you have a B / B+ that you can test with?

Saudumm, 2015-03-18:
Yes, I have a B and B+ here. I'll test it as soon as I'm at home.

Saudumm, 2015-03-18:
The problem also occurs on my B and B+ (used the same sd-card) with both USB soundcards.

Here's the output from `lsusb -v` for the Creative Play

```
Bus 001 Device 011: ID 041e:30d3 Creative Technology, Ltd Sound Blaster Play!
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x041e Creative Technology, Ltd
  idProduct          0x30d3 Sound Blaster Play!
  bcdDevice            1.00
  iManufacturer           0 
  iProduct                0 
  iSerial                 2 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          253
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength          100
        bInCollection           2
        baInterfaceNr( 0)       1
        baInterfaceNr( 1)       2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             2
        wTerminalType      0x0201 Microphone
[... 208 lines cut ...]
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      50
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              32
```

mrpetrov, 2015-04-14:
I am also having some USB trouble - I have a D-Link DWA-125 Wireless dongle which stays connected for some time, then this happens:

Apr 13 12:07:00 solarmanpi kernel: [    5.330644] Btrfs loaded
Apr 13 12:07:00 solarmanpi kernel: [    5.389043] BTRFS: device fsid af6a7450-c7ec-4ff8-a8f0-db4178e07b5f devid 1 transid 14817 /dev/mmcblk0p2
Apr 13 12:07:00 solarmanpi kernel: [    5.404039] BTRFS: device label STORAGE devid 1 transid 175 /dev/mmcblk0p3
Apr 13 12:07:00 solarmanpi kernel: [    5.434117] BTRFS info (device mmcblk0p2): disk space caching is enabled
Apr 13 12:07:00 solarmanpi kernel: [    5.461855] BTRFS: detected SSD devices, enabling SSD mode
Apr 13 12:07:00 solarmanpi kernel: [    7.377374] Driver for 1-wire Dallas network protocol.
Apr 13 12:07:00 solarmanpi kernel: [    7.398684] w1-gpio onewire@0: gpio pin 4, external pullup pin -1, parasitic power 0
Apr 13 12:07:00 solarmanpi kernel: [    7.733938] cfg80211: Calling CRDA to update world regulatory domain
Apr 13 12:07:00 solarmanpi kernel: [    8.008540] usb 1-1.2: reset high-speed USB device number 4 using dwc_otg
Apr 13 12:07:00 solarmanpi kernel: [    8.128879] ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 3070, rev 0201 detected
Apr 13 12:07:00 solarmanpi kernel: [    8.165984] random: nonblocking pool is initialized
Apr 13 12:07:00 solarmanpi kernel: [    8.177019] ieee80211 phy0: rt2x00_set_rf: Info - RF chipset 0005 detected
Apr 13 12:07:00 solarmanpi kernel: [    8.220019] usbcore: registered new interface driver rt2800usb
Apr 13 12:07:00 solarmanpi kernel: [    9.752352] BTRFS info (device mmcblk0p2): disk space caching is enabled
Apr 13 12:07:00 solarmanpi kernel: [    9.948862] BTRFS info (device mmcblk0p2): disk space caching is enabled
Apr 13 12:07:00 solarmanpi kernel: [   12.769514] BTRFS info (device mmcblk0p3): disk space caching is enabled
Apr 13 12:07:00 solarmanpi kernel: [   12.799406] BTRFS: detected SSD devices, enabling SSD mode
Apr 13 12:07:00 solarmanpi kernel: [   14.420785] ieee80211 phy0: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
Apr 13 12:07:00 solarmanpi kernel: [   14.426323] ieee80211 phy0: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.29
Apr 13 12:07:00 solarmanpi kernel: [   15.152042] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Apr 13 12:07:00 solarmanpi kernel: [   17.080403] wlan0: authenticate with 64:70:02:5c:78:ba
Apr 13 12:07:00 solarmanpi kernel: [   17.180202] wlan0: send auth to 64:70:02:5c:78:ba (try 1/3)
Apr 13 12:07:00 solarmanpi kernel: [   17.182696] wlan0: authenticated
Apr 13 12:07:00 solarmanpi kernel: [   17.187926] wlan0: associate with 64:70:02:5c:78:ba (try 1/3)
Apr 13 12:07:00 solarmanpi kernel: [   17.191527] wlan0: RX AssocResp from 64:70:02:5c:78:ba (capab=0x431 status=0 aid=1)
Apr 13 12:07:00 solarmanpi kernel: [   17.202498] wlan0: associated
Apr 13 12:07:00 solarmanpi kernel: [   17.202770] cfg80211: Calling CRDA to update world regulatory domain
Apr 14 05:41:04 solarmanpi kernel: [63242.481248] NYET/NAK/ACK/other in non-error case, 0x00000002
Apr 14 05:41:04 solarmanpi kernel: [63242.481286] NYET/NAK/ACK/other in non-error case, 0x00000002
Apr 14 05:41:04 solarmanpi kernel: [63242.481312] NYET/NAK/ACK/other in non-error case, 0x00000002
Apr 14 05:41:04 solarmanpi kernel: [63242.481369] smsc95xx 1-1.1:1.0 eth0: Failed to read reg index 0x00000114: -71
Apr 14 05:41:04 solarmanpi kernel: [63242.481382] smsc95xx 1-1.1:1.0 eth0: Error reading MII_ACCESS
Apr 14 05:41:04 solarmanpi kernel: [63242.481394] smsc95xx 1-1.1:1.0 eth0: MII is busy in smsc95xx_mdio_read
Apr 14 05:41:04 solarmanpi kernel: [63242.481431] NYET/NAK/ACK/other in non-error case, 0x00000002
Apr 14 05:41:04 solarmanpi kernel: [63242.481459] NYET/NAK/ACK/other in non-error case, 0x00000002
Apr 14 05:41:04 solarmanpi kernel: [63242.481485] NYET/NAK/ACK/other in non-error case, 0x00000002
Apr 14 05:41:04 solarmanpi kernel: [63242.481525] smsc95xx 1-1.1:1.0 eth0: Failed to read reg index 0x00000114: -71
Apr 14 05:41:04 solarmanpi kernel: [63242.481537] smsc95xx 1-1.1:1.0 eth0: Error reading MII_ACCESS
Apr 14 05:41:04 solarmanpi kernel: [63242.481548] smsc95xx 1-1.1:1.0 eth0: MII is busy in smsc95xx_mdio_read
Apr 14 05:41:04 solarmanpi kernel: [63242.495647] usb 1-1: USB disconnect, device number 2
Apr 14 05:41:04 solarmanpi kernel: [63242.495660] usb 1-1.1: USB disconnect, device number 3
Apr 14 05:41:04 solarmanpi kernel: [63242.495875] smsc95xx 1-1.1:1.0 eth0: unregister 'smsc95xx' usb-bcm2708_usb-1.1, smsc95xx USB 2.0 Ethernet
Apr 14 05:41:04 solarmanpi kernel: [63242.495931] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Apr 14 05:41:04 solarmanpi kernel: [63242.526200] usb 1-1.2: USB disconnect, device number 4
Apr 14 05:41:04 solarmanpi kernel: [63242.605586] wlan0: deauthenticating from 64:70:02:5c:78:ba by local choice (Reason: 3=DEAUTH_LEAVING)
Apr 14 05:41:04 solarmanpi kernel: [63242.667420] cfg80211: Calling CRDA for country: BG
Apr 14 05:41:05 solarmanpi kernel: [63242.955725] Indeed it is in host mode hprt0 = 00001901
Apr 14 05:41:05 solarmanpi kernel: [63243.075514] Indeed it is in host mode hprt0 = 00001501
Apr 14 05:41:05 solarmanpi kernel: [63243.405496] usb 1-1: new high-speed USB device number 5 using dwc_otg
Apr 14 05:41:05 solarmanpi kernel: [63243.405631] Indeed it is in host mode hprt0 = 00001101
[... cut ...]
Apr 14 06:48:19 solarmanpi kernel: [67277.510799] Indeed it is in host mode hprt0 = 00001501

As you can see, wifi worked fine for 18 hours. Then I tried manually unplugging the wifi stick and then plugging it back in to no avail. Other USB port did not work either. And this is the only USB device my Pi has - its a custom controller done with GPIO, powered by a 5V 2A adapter, and uses the wifi stick to communicate with the rest of the world.

Communication break down does not affect it doing its GPIO thing, but I would like to know why the wifi stick is failing...

mrpetrov, 2015-04-16:
Interesting stuff - all I did was move the wifi stick directly to on of my Pi 2 USB ports - so the change was to remove the extension cable I used, and now I have alomst 2 days of link time and still counting!
Kernel has had interesting stuff to say, probably:

[   15.261737] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   17.158603] wlan0: authenticate with 64:70:02:5c:78:ba
[   17.258013] wlan0: send auth to 64:70:02:5c:78:ba (try 1/3)
[   17.260290] wlan0: authenticated
[   17.267754] wlan0: associate with 64:70:02:5c:78:ba (try 1/3)
[   17.271304] wlan0: RX AssocResp from 64:70:02:5c:78:ba (capab=0x431 status=0 aid=1)
[   17.279525] wlan0: associated
[   17.279790] cfg80211: Calling CRDA to update world regulatory domain
[ 2070.522040] wlan0: deauthenticating from 64:70:02:5c:78:ba by local choice (Reason: 3=DEAUTH_LEAVING)
[ 2070.601978] cfg80211: Calling CRDA for country: BG
[ 2075.774311] wlan0: authenticate with 64:70:02:5c:78:ba
[ 2075.866153] wlan0: send auth to 64:70:02:5c:78:ba (try 1/3)
[ 2075.869975] wlan0: authenticated
[ 2075.880698] wlan0: associate with 64:70:02:5c:78:ba (try 1/3)
[ 2075.884127] wlan0: RX AssocResp from 64:70:02:5c:78:ba (capab=0x431 status=0 aid=1)
[ 2075.890457] wlan0: associated
[26151.683095] ieee80211 phy0: rt2800usb_fill_rxdone: Error - Bad frame size 8250, forcing to 0
[26151.683123] ieee80211 phy0: rt2x00lib_rxdone: Error - Wrong frame size 0 max 3840
[57288.179263] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 1 timed out, invoke forced kick
[57289.179333] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 1 timed out, invoke forced kick
[57290.209230] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 1 timed out, invoke forced kick
[57291.209223] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 1 timed out, invoke forced kick
[57292.239472] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 1 timed out, invoke forced kick
[57293.239288] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 1 timed out, invoke forced kick
[57294.179314] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 1 timed out, invoke forced kick
[57302.209237] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 1 timed out, invoke forced kick
[57303.209244] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 1 timed out, invoke forced kick
[57310.179305] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 0 timed out, invoke forced kick
[57311.179253] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 0 timed out, invoke forced kick
[57312.209342] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 0 timed out, invoke forced kick
[57313.209165] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 0 timed out, invoke forced kick
[57313.209737] ieee80211 phy0: rt2x00usb_watchdog_tx_dma: Warning - TX queue 2 DMA timed out, invoke forced forced reset
[57314.209052] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 0 timed out, invoke forced kick
[57314.209622] ieee80211 phy0: rt2x00usb_watchdog_tx_dma: Warning - TX queue 2 DMA timed out, invoke forced forced reset
[57315.209028] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 0 timed out, invoke forced kick
[57315.209573] ieee80211 phy0: rt2x00usb_watchdog_tx_dma: Warning - TX queue 2 DMA timed out, invoke forced forced reset
[57316.239421] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 0 timed out, invoke forced kick
[57316.240050] ieee80211 phy0: rt2x00usb_watchdog_tx_dma: Warning - TX queue 2 DMA timed out, invoke forced forced reset
[57317.239240] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 0 timed out, invoke forced kick
[57317.239808] ieee80211 phy0: rt2x00usb_watchdog_tx_dma: Warning - TX queue 2 DMA timed out, invoke forced forced reset
[57318.239051] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 0 timed out, invoke forced kick
[57318.239611] ieee80211 phy0: rt2x00usb_watchdog_tx_dma: Warning - TX queue 2 DMA timed out, invoke forced forced reset
[57319.239229] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 0 timed out, invoke forced kick
[57319.239819] ieee80211 phy0: rt2x00usb_watchdog_tx_dma: Warning - TX queue 2 DMA timed out, invoke forced forced reset
[57352.179225] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 1 timed out, invoke forced kick
[57353.179210] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 1 timed out, invoke forced kick
[57354.189122] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 1 timed out, invoke forced kick
[57355.239274] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 1 timed out, invoke forced kick
[57355.239714] ieee80211 phy0: rt2x00usb_watchdog_tx_dma: Warning - TX queue 2 DMA timed out, invoke forced forced reset
[57356.149515] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 1 timed out, invoke forced kick
[57356.149954] ieee80211 phy0: rt2x00usb_watchdog_tx_dma: Warning - TX queue 2 DMA timed out, invoke forced forced reset
[57357.149336] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 1 timed out, invoke forced kick
[57357.149781] ieee80211 phy0: rt2x00usb_watchdog_tx_dma: Warning - TX queue 2 DMA timed out, invoke forced forced reset
[57358.209314] ieee80211 phy0: rt2800usb_watchdog: Warning - TX HW queue 1 timed out, invoke forced kick
[57358.209741] ieee80211 phy0: rt2x00usb_watchdog_tx_dma: Warning - TX queue 2 DMA timed out, invoke forced forced reset
[... cut ...]
[121653.862482] ieee80211 phy0: rt2800usb_fill_rxdone: Error - Bad frame size 8250, forcing to 0
[121653.862511] ieee80211 phy0: rt2x00lib_rxdone: Error - Wrong frame size 0 max 3840
[127842.783765] ieee80211 phy0: rt2800usb_fill_rxdone: Error - Bad frame size 8250, forcing to 0
[127842.783792] ieee80211 phy0: rt2x00lib_rxdone: Error - Wrong frame size 0 max 3840
[127844.084077] ieee80211 phy0: rt2800usb_fill_rxdone: Error - Bad frame size 8250, forcing to 0
[127844.084104] ieee80211 phy0: rt2x00lib_rxdone: Error - Wrong frame size 0 max 3840

agherzan, 2015-09-01:
I encounter the same issue using an Edimax WIFI dongle:

root@raspberrypi2:~# [   43.914648] audit_printk_skb: 18 callbacks suppressed
[   43.919988] audit: type=1130 audit(1441107952.750:18): pid=451 uid=0 auid=4294967295 ses=4294967295 msg=' comm="systemd-journald" exe="/lib/systemd/systemd" hostname=? addr=? terminal=pts/0 re'
[   44.075980] audit: type=1130 audit(1441107952.910:19): pid=451 uid=0 auid=4294967295 ses=4294967295 msg=' comm="kmod-static-nodes" exe="/lib/systemd/systemd" hostname=? addr=? terminal=pts/0 r'
[   44.096293] audit: type=1130 audit(1441107952.930:20): pid=451 uid=0 auid=4294967295 ses=4294967295 msg=' comm="systemd-modules-load" exe="/lib/systemd/systemd" hostname=? addr=? terminal=pts/'
[   44.117562] audit: type=1130 audit(1441107952.950:21): pid=451 uid=0 auid=4294967295 ses=4294967295 msg=' comm="systemd-random-seed" exe="/lib/systemd/systemd" hostname=? addr=? terminal=pts/0'
[   44.142485] audit: type=1130 audit(1441107952.980:22): pid=451 uid=0 auid=4294967295 ses=4294967295 msg=' comm="systemd-tmpfiles-setup" exe="/lib/systemd/systemd" hostname=? addr=? terminal=pt'
[   44.223938] audit: type=1130 audit(1441107953.060:23): pid=451 uid=0 auid=4294967295 ses=4294967295 msg=' comm="systemd-journal-flush" exe="/lib/systemd/systemd" hostname=? addr=? terminal=pts'
[   44.269052] audit: type=1127 audit(1441107953.100:24): pid=500 uid=0 auid=4294967295 ses=4294967295 msg='init exe="/lib/systemd/systemd-update-utmp" hostname=? addr=? terminal=? res=success'
[   44.290837] audit: type=1131 audit(1441107953.130:25): pid=451 uid=0 auid=4294967295 ses=4294967295 msg=' comm="systemd-journal-flush" exe="/lib/systemd/systemd" hostname=? addr=? terminal=pts'
[   44.321889] audit: type=1130 audit(1441107953.160:26): pid=451 uid=0 auid=4294967295 ses=4294967295 msg=' comm="systemd-sysctl" exe="/lib/systemd/systemd" hostname=? addr=? terminal=pts/0 res='
[   44.388381] systemd-journald[24]: Received request to flush runtime journal from PID 1
[   44.399652] audit: type=1130 audit(1441107953.240:27): pid=451 uid=0 auid=4294967295 ses=4294967295 msg=' comm="systemd-tmpfiles-setup-dev" exe="/lib/systemd/systemd" hostname=? addr=? termina'
[   44.648510] systemd-udevd[34]: starting version 215
[   51.987880] rtw_cmd_thread: DriverStopped(0) SurpriseRemoved(1) break at line 482
[   56.306232] w1_master_driver w1_bus_master1: Family 0 for 00.800000000000.8c is not registered.
[   58.748662] usb 1-1.1: USB disconnect, device number 3
[   58.754198] smsc95xx 1-1.1:1.0 eth0: unregister 'smsc95xx' usb-bcm2708_usb-1.1, smsc95xx USB 2.0 Ethernet
[   58.768862] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   58.875357] usb 1-1.2: USB disconnect, device number 4
[   58.946291] Indeed it is in host mode hprt0 = 00001101
[   59.124427] usb 1-1: reset high-speed USB device number 2 using dwc_otg
[   59.131432] Indeed it is in host mode hprt0 = 00001101
[   59.314263] usb 1-1: device descriptor read/64, error -71
[   59.424553] Indeed it is in host mode hprt0 = 00001101
[   59.604239] usb 1-1: device descriptor read/64, error -71
[   59.714100] Indeed it is in host mode hprt0 = 00001101
[   59.894014] usb 1-1: reset high-speed USB device number 2 using dwc_otg
[   59.901065] Indeed it is in host mode hprt0 = 00001101
[   60.083915] usb 1-1: device descriptor read/64, error -71
[   60.194126] Indeed it is in host mode hprt0 = 00001101
[   60.373754] usb 1-1: device descriptor read/64, error -71
[   60.483722] Indeed it is in host mode hprt0 = 00001101
[   60.663609] usb 1-1: reset high-speed USB device number 2 using dwc_otg
[   61.083338] usb 1-1: device not accepting address 2, error -71
[   61.089291] Indeed it is in host mode hprt0 = 00001101
[   61.273288] usb 1-1: reset high-speed USB device number 2 using dwc_otg
[   61.693045] usb 1-1: device not accepting address 2, error -71
[   61.699149] usb 1-1: USB disconnect, device number 2
[   61.802995] hub 1-1:1.0: activate --> -19
[   61.962948] Indeed it is in host mode hprt0 = 00001501
[   62.142831] usb 1-1: new high-speed USB device number 5 using dwc_otg
[   62.149647] Indeed it is in host mode hprt0 = 00001101
[   62.332743] usb 1-1: device descriptor read/64, error -71
[   62.443018] Indeed it is in host mode hprt0 = 00001101
[   62.622665] usb 1-1: device descriptor read/64, error -71
[   62.732609] Indeed it is in host mode hprt0 = 00001101
[   62.912453] usb 1-1: new high-speed USB device number 6 using dwc_otg
[   62.919280] Indeed it is in host mode hprt0 = 00001101
[   63.102362] usb 1-1: device descriptor read/64, error -71
[   63.212611] Indeed it is in host mode hprt0 = 00001101
[   63.392213] usb 1-1: device descriptor read/64, error -71
[   63.502264] Indeed it is in host mode hprt0 = 00001101
[   63.682112] usb 1-1: new high-speed USB device number 7 using dwc_otg
[   64.101847] usb 1-1: device not accepting address 7, error -71
[   64.107808] Indeed it is in host mode hprt0 = 00001101
[   64.291773] usb 1-1: new high-speed USB device number 8 using dwc_otg
[   64.382585] IPv6: ADDRCONF(NETDEV_UP): rce0: link is not ready
[   64.711527] usb 1-1: device not accepting address 8, error -71
[   64.717466] usb usb1-port1: unable to enumerate USB device
[  115.614516] w1_master_driver w1_bus_master1: Family 0 for 00.400000000000.46 is not registered.
[  162.984440] w1_master_driver w1_bus_master1: Family 0 for 00.c00000000000.ca is not registered.

stdoubleu, 2015-09-29:
Still an open issue for the Rpi 2:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=118891

Quite disappointing that this problem is not addressed in any way.

vrabac, 2015-12-20:
I have also same problem on my Raspberry Pi 2 Model B, the USB controller disconnects after a random time on archlinux-arm.

```
[   10.174136] input: Sundtek Ltd. Remote Control as /devices/virtual/input/input0
[   11.494877] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   11.502826] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC5E1
[   29.619982] usb 1-1.5: usbfs: process 298 (mediasrv) did not claim interface 2 before use
[   72.767628] usb usb1-port1: disabled by hub (EMI?), re-enabling...
[   72.773965] usb 1-1: USB disconnect, device number 2
[   72.779113] usb 1-1.1: USB disconnect, device number 3
[   72.784996] smsc95xx 1-1.1:1.0 eth0: unregister 'smsc95xx' usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet
[   72.794934] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   72.878485] usb 1-1.5: USB disconnect, device number 4
[   72.889941] usb 1-1.5: usbfs: USBDEVFS_CONTROL failed cmd mediasrv rqt 64 rq 2 len 1 ret -108
[   73.047648] Indeed it is in host mode hprt0 = 00001501
[   73.167501] Indeed it is in host mode hprt0 = 00001101
[   73.437546] Indeed it is in host mode hprt0 = 00001101
[   73.707521] Indeed it is in host mode hprt0 = 00001101
[   73.977525] Indeed it is in host mode hprt0 = 00001101
[   74.247533] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[   74.254271] Indeed it is in host mode hprt0 = 00001101
[   74.377515] Indeed it is in host mode hprt0 = 00001101
[   74.647517] Indeed it is in host mode hprt0 = 00001101
[   74.917530] Indeed it is in host mode hprt0 = 00001101
[   75.187538] Indeed it is in host mode hprt0 = 00001101
[   75.457521] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[   75.464244] Indeed it is in host mode hprt0 = 00001101
[   75.587525] Indeed it is in host mode hprt0 = 00001101
[   75.857507] Indeed it is in host mode hprt0 = 00001101
[   76.127515] Indeed it is in host mode hprt0 = 00001101
[   76.397517] Indeed it is in host mode hprt0 = 00001101
[   76.667525] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[   76.674229] Indeed it is in host mode hprt0 = 00001101
[   76.797511] Indeed it is in host mode hprt0 = 00001101
[   77.067515] Indeed it is in host mode hprt0 = 00001101
[   77.337503] Indeed it is in host mode hprt0 = 00001101
[   77.607512] Indeed it is in host mode hprt0 = 00001101
[   77.877511] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[   77.884195] usb usb1-port1: unable to enumerate USB device
```

lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 5: Dev 4, If 0, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 5: Dev 4, If 1, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 5: Dev 4, If 2, Class=Vendor Specific Class, Driver=usbfs, 480M
        |__ Port 5: Dev 4, If 3, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 5: Dev 4, If 4, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 5: Dev 4, If 5, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 5: Dev 4, If 6, Class=Vendor Specific Class, Driver=, 480M

Again last night:
```
[Sat Dec 19 20:28:30 2015] input: Sundtek Ltd. Remote Control as /devices/virtual/input/input0
[Sat Dec 19 20:28:30 2015] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[Sat Dec 19 20:28:30 2015] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[Sat Dec 19 20:28:32 2015] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[Sat Dec 19 20:28:32 2015] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC5E1
[Sat Dec 19 20:28:50 2015] usb 1-1.5: usbfs: process 295 (mediasrv) did not claim interface 2 before use
[Sat Dec 19 23:20:47 2015] usb 1-1-port5: disabled by hub (EMI?), re-enabling...
[Sat Dec 19 23:20:47 2015] usb 1-1.5: USB disconnect, device number 4
[Sat Dec 19 23:20:47 2015] usb 1-1.5: new high-speed USB device number 5 using dwc_otg
[Sat Dec 19 23:20:48 2015] usb 1-1.5: New USB device found, idVendor=2659, idProduct=1502
[Sat Dec 19 23:20:48 2015] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Sat Dec 19 23:20:48 2015] usb 1-1.5: Product: SkyTV Ultimate V 2015
[Sat Dec 19 23:20:48 2015] usb 1-1.5: Manufacturer: Sundtek
[Sat Dec 19 23:20:48 2015] usb 1-1.5: SerialNumber: XXXXXXX
[Sat Dec 19 23:20:49 2015] input: Sundtek Ltd. Remote Control as /devices/virtual/input/input1
```

Kernel: Linux vdrrpi 4.1.15-1-ARCH
Firmware:
$ pacman -Qi raspberrypi-firmware
Name           : raspberrypi-firmware
Version        : 20151215-1

stdoubleu, 2015-12-20:
I finally return my rpi2 to the distributor; he checked it, reproduced the problem and sent me a new one.
Seems to be a quite rare problem.

agherzan, 2015-12-20:
So you state that this is a hw dependent issue?

Saudumm, 2015-12-20:
I had the problem on a RPi 2, B+ and a B (same SD card), so I don't think it's a hw problem with the Pi itself.

stdoubleu, 2015-12-24:
I tried different SD cards, power units, cables, locations, fresh images, thus in my case it was a hw dependent issue on my rpi2. 
Although it seems to appear also on other pi's it still can be an hw issue on those too. Maybe something is not isolated properly or something is not shielded correctly during manufaturing and thus sensible for interferences?
Still a rare event but from my point not a software issue.

willyt, 2016-01-22:
I'm seeing this issue on Rpi 2 B. Just after a backup script runs I get:
```
Jan 22 02:37:24 mail kernel: [52059.816040] usb 1-1.5: reset high-speed USB device number 4 using dwc_otg
```
and then I get a whole bunch of btrfs write errors on the attached external HD. Let me know if there is more useful information I can send to help debug.

pasnox, 2016-03-26:
Same here... On high load use of the disk it get disconnect and reconnect with another device name (/dev/sda1 become /dev/sdb1)

Any idea ?

bcutter, 2016-05-03:
Same here. I have tons of those messages and I don´t know what to do. It might also be this is the root cause of periodic reboots of the Pi 2. Every 0,8 to 2 days it reboots, the only suspicios thing in the logs are those USB relevant ones.

```
	Line  528: May  2 00:40:11 raspberrypi kernel: [66640.136492] usb 1-1.2.3: reset high-speed USB device number 8 using dwc_otg
	Line 1027: May  2 05:06:12 raspberrypi kernel: [13851.488587] usb 1-1.2.3: reset high-speed USB device number 8 using dwc_otg
	Line 1028: May  2 05:06:13 raspberrypi kernel: [13852.098586] usb 1-1.2.3: reset high-speed USB device number 8 using dwc_otg
	Line 1205: May  2 11:46:13 raspberrypi kernel: [37852.587610] usb 1-1.2.3: reset high-speed USB device number 8 using dwc_otg
	Line 1486: May  2 21:06:12 raspberrypi kernel: [71452.186613] usb 1-1.2.3: reset high-speed USB device number 8 using dwc_otg
	Line 1497: May  2 21:26:12 raspberrypi kernel: [72652.199174] usb 1-1.2.3: reset high-speed USB device number 8 using dwc_otg
	Line 1771: May  3 06:36:12 raspberrypi kernel: [105652.491623] usb 1-1.2.3: reset high-speed USB device number 8 using dwc_otg
	Line 1859: May  3 10:16:12 raspberrypi kernel: [118852.676307] usb 1-1.2.3: reset high-speed USB device number 8 using dwc_otg
	Line 1860: May  3 10:16:13 raspberrypi kernel: [118853.486309] usb 1-1.2.3: reset high-speed USB device number 8 using dwc_otg
	Line 1861: May  3 10:16:14 raspberrypi kernel: [118854.096317] usb 1-1.2.3: reset high-speed USB device number 8 using dwc_otg
	Line 1917: May  3 12:06:12 raspberrypi kernel: [125452.706595] usb 1-1.2.3: reset high-speed USB device number 8 using dwc_otg
```

**How can I check which USB device is the mentioned "usb 1-1.2.3" ?**

stdoubleu, 2016-05-03:
FYI: After spending days to find the error, I returned mine to the store. It took them some time to reproduce the problem, but also couldnt provide a solution.
Thus, I got a new one and since then it runs like charm.

I recommend to return it instead of wasting time.

bcutter, 2016-05-06:
I´ll give another hardware a shot. Before those errors (and even worse: the Pi randomly reboots every half to 2 days since using it as a NAS with Samba!) 3 of 4 usb ports were not in use. I´ll try to track it down from the endpoints (sata to USB adapters) over the USB hubs to the Pi´s USB port.

bcutter, 2016-05-10:
Switching one SATA to USB 3.0-adapter seems to have fixed the error (and the reboots), at least nothing in logs for about 3 days. Fingers crossed.

Ruffio, 2016-08-14:
@Saudumm has your issue been resolved? If yes, then please close this issue.

P33M, 2017-05-04:
Closing USB issues as OP has not updated the thread. Further comments by the OP stating that the issue is still present on latest rpi-update kernel will lead to a review.

closed 2017-05-04 by P33M