raspberrypi/linux #898: bcm2835-v4l2 waiting for frame completion

danger89, 2015-03-17:
Hi,

I'm using Raspbian (latest version:  3.18.9+ #768). Together with MotionEye.
What I did:
 - Install the Raspbian image via dd
 - In raspi-config, I enabled 'Enable support for Raspberry Pi camera'.
 - Plug the Raspberry Pi Camera module in the Pi
 - Edit: /etc/modules, append the file with: 'bcm2835-v4l2'
 - I got /dev/video0 correctly
 - I installed MotionEye (git clone git@bitbucket.org:ccrisan/motioneye.git)
 - Used the default settings.py. Starting service: ./motioneye.py
 - Add camera (mmal service 16.1)

Now, after several seconds (max. 1 minute) the following errors/warnings will appear in /var/log/kern.log:
```
Mar 17 21:33:30 raspcam kernel: [  263.204615] bcm2835-v4l2: error 0 waiting for frame completion
Mar 17 21:33:56 raspcam kernel: [  288.874612] bcm2835-v4l2: error 0 waiting for frame completion
Mar 17 21:34:21 raspcam kernel: [  313.434586] bcm2835-v4l2: error 0 waiting for frame completion
Mar 17 21:35:07 raspcam kernel: [  360.184571] bcm2835-v4l2: error 0 waiting for frame completion
Mar 17 21:35:59 raspcam kernel: [  412.324656] bcm2835-v4l2: error 0 waiting for frame completion
Mar 17 21:36:36 raspcam kernel: [  448.824562] bcm2835-v4l2: error 0 waiting for frame completion
Mar 17 21:37:12 raspcam kernel: [  484.814579] bcm2835-v4l2: error 0 waiting for frame completion
Mar 17 21:37:48 raspcam kernel: [  520.774628] bcm2835-v4l2: error 0 waiting for frame completion
```

Every time a new line appears in the kern.log, the camera LED will turn off for abotu 2 seconds. And then it will turn on again. I didn't had this issue on a previous Raspbian release and older firmware.

Yes, ...:
 - I can create images (screenshots) via v4l2-ctl without problems
 - I can use raspistill/raspivid commands without issues

Firmware version: /opt/vc/bin/vcgencmd version
```
Mar 15 2015 19:51:38 
Copyright (c) 2012 Broadcom
version fef371701296941551f3b8375d5680a427c04f48 (clean) (release)
```

Debug output. /var/log/debug:
```
Mar 17 21:33:33 raspcam motion: [1] Started stream webcam server in port 8081
Mar 17 21:33:55 raspcam motion: [1] Calling vid_close() from motion_cleanup
Mar 17 21:34:03 raspcam motion: [0] httpd thread exit
Mar 17 21:34:03 raspcam motion: [1] Thread 1 started
Mar 17 21:34:03 raspcam motion: [1] mmap information:
Mar 17 21:34:03 raspcam motion: [1] frames=4
Mar 17 21:34:03 raspcam motion: [1] 0 length=115200
Mar 17 21:34:03 raspcam motion: [1] 1 length=115200
Mar 17 21:34:03 raspcam motion: [1] 2 length=115200
Mar 17 21:34:03 raspcam motion: [1] 3 length=115200
Mar 17 21:34:04 raspcam motion: [1] Started stream webcam server in port 8081
Mar 17 21:34:20 raspcam motion: [1] Calling vid_close() from motion_cleanup
Mar 17 21:34:36 raspcam motion: [0] httpd thread exit
Mar 17 21:34:36 raspcam motion: [1] Thread 1 started
Mar 17 21:34:36 raspcam motion: [1] mmap information:
Mar 17 21:34:36 raspcam motion: [1] frames=4
Mar 17 21:34:36 raspcam motion: [1] 0 length=115200
Mar 17 21:34:36 raspcam motion: [1] 1 length=115200
Mar 17 21:34:36 raspcam motion: [1] 2 length=115200
Mar 17 21:34:36 raspcam motion: [1] 3 length=115200
Mar 17 21:34:36 raspcam motion: [1] Started stream webcam server in port 8081
Mar 17 21:35:06 raspcam motion: [1] Calling vid_close() from motion_cleanup
Mar 17 21:35:15 raspcam motion: [0] httpd thread exit
Mar 17 21:35:15 raspcam motion: [1] Thread 1 started
Mar 17 21:35:15 raspcam motion: [1] mmap information:
Mar 17 21:35:15 raspcam motion: [1] frames=4
Mar 17 21:35:15 raspcam motion: [1] 0 length=115200
Mar 17 21:35:15 raspcam motion: [1] 1 length=115200
Mar 17 21:35:15 raspcam motion: [1] 2 length=115200
Mar 17 21:35:15 raspcam motion: [1] 3 length=115200
[... 82 lines cut ...]
Mar 17 21:54:32 raspcam motion: [0] httpd thread exit
Mar 17 21:54:32 raspcam motion: [1] Thread 1 started
Mar 17 21:54:32 raspcam motion: [1] mmap information:
Mar 17 21:54:32 raspcam motion: [1] frames=4
Mar 17 21:54:32 raspcam motion: [1] 0 length=115200
Mar 17 21:54:32 raspcam motion: [1] 1 length=115200
Mar 17 21:54:32 raspcam motion: [1] 2 length=115200
Mar 17 21:54:32 raspcam motion: [1] 3 length=115200
Mar 17 21:54:33 raspcam motion: [1] Started stream webcam server in port 8081
Mar 17 21:55:05 raspcam motion: [1] Calling vid_close() from motion_cleanup
Mar 17 21:55:08 raspcam motion: [0] httpd thread exit
Mar 17 21:55:08 raspcam motion: [1] Thread 1 started
Mar 17 21:55:08 raspcam motion: [1] mmap information:
Mar 17 21:55:08 raspcam motion: [1] frames=4
Mar 17 21:55:08 raspcam motion: [1] 0 length=115200
Mar 17 21:55:08 raspcam motion: [1] 1 length=115200
Mar 17 21:55:08 raspcam motion: [1] 2 length=115200
Mar 17 21:55:08 raspcam motion: [1] 3 length=115200
Mar 17 21:55:08 raspcam motion: [1] Started stream webcam server in port 8081
```

Thanks in advance!

Kind regards,
Melroy van den Berg

6by9, 2015-03-18:
I'd still recommend the forums. Posting on here will get me looking at it in spare time. The forums will get many people looking into it.

Please also post the results from ```v4l2-ctl -V``` after your system has run, as that will tell us the format that it is asking for.

Ruffio, 2016-08-14:
@danger89 has your issue been resolved? If yes, then please close this issue.

closed 2016-08-14 by danger89