raspberrypi/linux #900: Raspbian has Ruby already installed, why? And an older version to boot 1.9.3

Basilmir, 2015-03-18:
I'm trying out a Ruby gem, and it took a lot of work to get up and running on Raspbian, It seems Ruby 1.9.3 was present, tried:

apt-get remove ruby1.9.3

Why is it there?

popcornmix, 2015-03-19:
This repo if for linux kernel issues. Your issue appears to be a raspbian image issue.
I'm guessing ruby is a dependency of one of the installed packages.

Probably best to ask in the raspbian forum: http://www.raspberrypi.org/forums/viewforum.php?f=66
Or report it here: https://github.com/asb/spindle which is the repo used create raspbian images.

closed 2015-03-19 by popcornmix

Basilmir, 2015-03-23:
Thought I'd come back and post my results. More on the subject here: http://www.raspberrypi.org/forums/viewtopic.php?f=66&t=104861&sid=c9a89d0005ee5cd2ff63cd137911a9f5

Finally I gave up just installed a clean Raspbian and used the 'old' packages (it worked by the way, it's a good thing):

sudo apt-get netatalk - for mDNS and file sharing so I can contact my Pi on raspberrypi.local, if you only need mDNS then just install sudo apt-get avahi, I don't see why this is not included in Raspbian by default, no point in having a hostname if you can't use it to find your Pi on the local network.

sudo apt-get update - update the repository list.
sudo apt-get upgrade - see if there are any newer versions for packages you already have.

sudo apt-get ruby1.9.1-dev - actually install Ruby 1.9.3-p227-dev the actual name of the package stayed the same while the contents were updated. You need it to compile stuff, especially on the Pi 2 where there are no prebuilt packages available.

sudo gem install passenger - get Passenger (link between Ruby apps and Nginx) installed.
* you also might need sudo apt-get install libcurl4-gnutls-dev - a dependency needed for the next step, it will warn you and ask for you to install it anyway if you forget about it.
sudo passenger-install-nginx-module - compile and install Nginx with the Passenger module.

sudo gem install your_app_of_choice - your app should compile and install it's own dependencies, follow on-screen instructions if it fails.