raspberrypi/linux #901: Rather clean install of Raspbian via NOOBS 1.4 doesn't shutdown cleanly on the Pi 2?

Basilmir, 2015-03-19:
Connecting via ssh to the Raspberry Pi 2 and doing a "sudo shutdown now" starts off the shutdown sequence only to show a root shell after:

INIT: Sending processes the KILL signal

PS. All I installed is an "sudo apt-get install netatalk"

popcornmix, 2015-03-20:
I think this is expected behaviour. See:
http://askubuntu.com/questions/578144/why-doesnt-running-sudo-shutdown-now-shutdown

Try
```
sudo shutdown -h now
```
or
```
sudo halt
```

Basilmir, 2015-03-20:
Oh! I'm not exactly a newbie but I've never encountered this before. Mainly used GNU Linux on PCs and those usually have Power On/Off buttons that do that for you. Interesting! I guess we can close this then.

closed 2015-03-21 by popcornmix