raspberrypi/linux #903: USB isosynchronous - zeroed input buffer returned

TheTechnobear, 2015-03-20:
Hi,
Ive got an issue with a USB device (1.1) that uses iso transfers.
all completions back from the device are filled with zeros, even though they are marked as successful completions. It appears the data is not being copied back into the buffer by the kernel.

some background:
Im using libusb,  and have exactly the same code, running for 2 devices, a usb 1.1device and a usb 2.0 device. both devices work on a debian/pc fine, and the usb 2.0 device works fine on the PI2, its only the 1.1 device that has this issue of a 'zeroed buffer'

observations:
- when libusb debug is on, everything is shown as successful there are no errors
- Ive done a usbmon trace on the 1.1 device on the PI2, and can see that the data is being sent from the device correctly.

usually no flags are present in cmdline.txt for this, but Ive tried the various dwc_otg flags to disable/enable fiq/fiq_fsm etc etc, and none seem to make any difference. 

Ive run out to 'things to try', but i need to get this device working :( 

Given the above, it seems likely its a kernel issue, is there something I can do to get more information to the likely cause, or possible fix?
(Id be happy to test any ideas anyone has... )

platform: PI2: 3.18.9-v7
I'm using a separate powered MTT hub, and Ive tried 3 different hubs, also the same hubs work fine on the debian PC installation.

P33M, 2015-03-20:
Hi.

Some more logs would be useful.

- Please post a `lsusb -v` with the device attached.
- Please post the output of dmesg after running your program for communicating with the device (with default settings of `/boot/cmdline.txt`)?

By what metric are you measuring the fact that the completions are "successful"?

Note that historically it has been virtually impossible to have working/reliable Isochronous transports from a USB1.1 device. For each frame (1ms) of data, you have to precisely and correctly schedule the start-split and complete-split transfers (of which there are multiple complete-split transfers) or the translating hub simply drops the data. The fiq_fsm code substantially resolved most of the issues surrounding this set of devices but there are certain edge cases that may still result in failure.

TheTechnobear, 2015-03-20:
completion success... libusb reports status complete on both the overall transfer and on the individual packets. also usbmon status field also shows the iso completion as been 'successful'. 

Im not quite sure what you mean by "its been virtually impossible to have reliable iso transports" on 1.1 devices,this device and code is working fine on Windows/Mac OSX and also other Linux distributions (the last of which is using exactly the same code as on the PI).

of course saying this, Id don't doubt the difficultly of the task, nor the great job that has been done on the code in recent times - both of which I highly respect, and appreciate.

Naive questions.. is there a large amount of difference in the code for a 2.0 and 1.1 device, because as I mentioned, the 2.0 device works fine with the same libusb code.

Is there anything I can do with the dwc module, to get it to indicate what the issue is
(as you will see from the logs, there really is nothing in them which indicates any kind of error)

I'll go get the rest of the details now :)

TheTechnobear, 2015-03-20:
(lsusb for device, ive removed product name/manfacture, as Id prefer not to have this public at this time)

Bus 001 Device 006: ID 2139:0101  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         0 
  bDeviceProtocol       255 
  bMaxPacketSize0        64
  idVendor           0x2139 
  idProduct          0x0101 
  bcdDevice            0.01
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           25
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              488mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol    255 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0300  1x 768 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

TheTechnobear, 2015-03-20:
[   73.108067] usb 1-1.5.6: new full-speed USB device number 6 using dwc_otg
[   73.217684] usb 1-1.5.6: New USB device found, idVendor=2139, idProduct=0101
[   73.217703] usb 1-1.5.6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   73.217715] usb 1-1.5.6: Product: vxxxx
[   73.217726] usb 1-1.5.6: Manufacturer: xxxxxx

TheTechnobear, 2015-03-20:
boot/cmdline.txt
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait

P33M, 2015-03-20:
To elaborate, support for USB1.1 Isochronous on Raspberry Pi has been troublesome due to limitations with the hardware on BCM2835. It's not a conventional USB host. In fact word from the original IP implementors is that it was never intended to be used in this fashion, so Raspberry Pi are basically "on their own" as far as support is concerned.

I see from your lsusb output that your endpoint is a 768 byte packet every 1ms frame. Does your device transmit a packet that is always the maximum packet size for the endpoint?

TheTechnobear, 2015-03-20:
ah, interesting info about the BCM2835... I can appreciate the problems being faced!

768... that is interesting, I believed it should*... BUT I noticed on the usbmon data,  Im only getting back 384. (sorry should have mentioned this in orig post).
BUT I checked on the Debian/PC where it IS working, and that is the same (its returning 384 bytes), 
so I suspect that 384 is normal.
to be 100% certain, I would have to check on the Mac implementation... to see if also receives 384.
( my code supplies a 768 buffer for it)

* the spec and header files says its 768, but Im wondering if this is just incorrect... hence the need to if in practice its using 768.

TheTechnobear, 2015-03-20:
okay, I can confirm, on Mac OSX it also returns 384... so this appears to be normal.

P33M, 2015-03-21:
So with USBmon you can see correct data - i.e. when URBs still belong to the kernel, data comes across the wire and appears to be received correctly, but the data retrieved from libusb is erroneous?

TheTechnobear, 2015-03-21:
Yes, USBmon shows correct data, but libusb does not give me the correct data.

unfortunately, no I don't have a Pi1, only a Pi2

again, just a small reminder, this device works perfectly with libusb on Debian/Pc, so not sure this can be an application/libusb fault

TheTechnobear, 2015-03-21:
ok, had a look at the above issue, this was an issue in the bcm2835-v4l2 kernel module, (bytesused=0)
and seems specific to that module (unless there is another issue) which I don't use...

Im not using any others module, as this is a custom usb device - I just use libusb which i assume goes to dec_otg directly? ... or is there something else?

I also do already check the 'actual length' of the data in each packet returned and handle appropriately, this includes receiving zero BUT I don't get zero, I get 384 bytes returned of zeros.

P33M, 2015-03-23:
To be honest, unless I either find or create a device that replicates your use case (using libusb to talk to a USB1.1 isochronous endpoint) I can't do much to diagnose the bug - especially if USBMON is reporting correct data being returned from the driver. The only devices that i've seen with the necessary features are audio or webcams - both are non-trivial to get working with just libusb talking to the device.

I would recommend comparing performance of the device on a Pi 1 with 3.12 / 3.18 kernel.

TheTechnobear, 2015-03-23:
Unfortunately, Ive not got access to a PI1... and it wouldn't have the capacity to run my application anyway.

Yeah, I can understand that... bit hard debugging something without being able to reproduce it.

Perhaps you could point me, to some areas of code I could look in the dwc module that are responsible for copying this data, or that decides that the data should not be copied?
then I could perhaps do some test with some debug code, to see if I can narrow it down...
(remember I have a working 2.0 device, so hopefully I could see where the two devices 'deviate')

Im assuming, that whats happening is that for some reason, the module is deciding not to copy the data, perhaps because it thinks there is an 'error' (but not reporting it), or its 'stale' data, or something similar.
(it seems unlikely if usbmon is getting the data, that the dwc module isn't)

(my issue, is I don't know the dwc code, so really would need some kind of 'starting point')

perhaps if I do this... i might not be able to fix it, but at lease shed some more light on what is going on.

Thanks
Mark

P33M, 2015-03-23:
Usbmon operates at the layer above dwc_otg - the tracepoints are at usb_submit_urb and the corresponding usb_hcd_giveback_urb.
http://lxr.free-electrons.com/source/drivers/usb/core/hcd.c#L1543
http://lxr.free-electrons.com/source/drivers/usb/core/hcd.c#L1665

These gateways are through which all transfer requests and associated data passes. Therefore if you can see data in usbmon packets, then the bug is above dwc_otg - unless there is something about our completed iso_desc structures (c.f. http://lxr.free-electrons.com/source/include/linux/usb.h#L1215) that differs that causes libusb to do something weird.

TheTechnobear, 2015-03-25:
Thanks I'll take a look, when I get a bit more time.

MrReeMachine, 2017-02-04:
I was wondering if there is any update on this issue as I am seeing this too.

I am trying to read isochronos audio stream from a chat headset connected to an Xbox One game controller connected to a Pi 3 via a USB cable - as with others above everything appears to be working correctly except the buffers are filled with zeros and not with the data I'm seeing being sent over the USB  in my USB protocol analyser.

TheTechnobear, 2017-02-11:
Ok...I have some extra information:

- rPI2,  I now 2 devices that have this same issue.   
- the Asus Tinkerboard (Rockchip RK3288 , kernel 4.4) has the same issue as the PI2

- the Beaglebone Black does NOT have the issue, (nor any 'full size pc' ive tried running linux) 

just like the rPI2, the tinkerboard also shows the data at the usbmon level, but further up the usb stack.
 
my understanding is the KK3288 is also using the dwc_otg driver

I guess, I'm going to need to dig into the kernel/libusb..
one question  I have (as Ive not done any work with the kernel before), perhaps for @P33M 

so in the above source links for hcd.c, line 1745, in __usb_hcd_giveback_urb, is calling , usbmon_urb_complete(&hcd->self, urb, status);

so naively, I assuming this is a callback to usbmon, which will display the data, which we know is correct... it appears only a couple of lines later, that the usb->complete() is called... which I (again naively ) assume is the completion function specified

if this is the case, then it would appear my best bet is probably to put a trace inside libusb where it gets the completion callback, and see if the data is present there... and to find if there are any 'edge' cases, where it doesn't copy the data...

basically to pursue as @P33M  said
"unless there is something about our completed iso_desc structures (c.f. http://lxr.free-electrons.com/source/include/linux/usb.h#L1215) that differs that causes libusb to do something weird."

(I guess I'm a bit more comfortable doing this rather than building/debugging installing kernels which Ive little experience of)

mcerveny, 2017-02-11:
I probably found it. Kernel does not copy output of ISO transfers to userspace (buffer is untouched) but  "actual_length" per ISO is set. The problem is [here](https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/usb/core/devio.c#L1663) because it test "global" "actual_length" and does not trigger copy [here](https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/usb/core/devio.c#L426). The other USB HCI like UHCI ([here](https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/usb/host/uhci-q.c#L1394)) increments "global" "actual_length" from ISO "actual_length".  DWC ODG HCI should do the same [here](https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/usb/host/dwc_otg/dwc_otg_hcd_linux.c#L329). Something like:
```
diff --git a/drivers/usb/host/dwc_otg/dwc_otg_hcd_linux.c b/drivers/usb/host/dwc_otg/dwc_otg_hcd_linux.c
index 2ceed42..5011750 100644
--- a/drivers/usb/host/dwc_otg/dwc_otg_hcd_linux.c
+++ b/drivers/usb/host/dwc_otg/dwc_otg_hcd_linux.c
@@ -325,10 +325,12 @@ static int _complete(dwc_otg_hcd_t * hcd, void *urb_handle,
 		int i;
 
 		urb->error_count = dwc_otg_hcd_urb_get_error_count(dwc_otg_urb);
+		urb->actual_length = 0;
 		for (i = 0; i < urb->number_of_packets; ++i) {
 			urb->iso_frame_desc[i].actual_length =
 			    dwc_otg_hcd_urb_get_iso_desc_actual_length
 			    (dwc_otg_urb, i);
+			urb->actual_length += urb->iso_frame_desc[i].actual_length;
 			urb->iso_frame_desc[i].status =
 			    dwc_otg_hcd_urb_get_iso_desc_status(dwc_otg_urb, i);
 		}

```

TheTechnobear, 2017-02-12:
thank you @mcerveny  
Ive applied this patch (to 4.4.48 kernel) and test with 2 problematic devices, and both now appear to work.

this is fantastic... do you know how to get it put into the main source tree, as its not just the rPI affected by this.

thanks again.

P33M, 2017-02-12:
It's good that you found the bug - but in my opinion the bug is in devio.c.

The URB actual_length field is meaningless for isochronous transfers, as each frame may be of variable length so the individual iso_desc array elements have actual_length per-frame.

http://lxr.free-electrons.com/source/include/linux/usb.h#L1317
```
1317  * @actual_length: This is read in non-iso completion functions, and
1318  *      it tells how many bytes (out of transfer_buffer_length) were
1319  *      transferred.  It will normally be the same as requested, unless
1320  *      either an error was reported or a short read was performed.
1321  *      The URB_SHORT_NOT_OK transfer flag may be used to make such
1322  *      short reads be reported as errors.
```
The devio.c processcompl function should ignore actual_length if dealing with an isochronous transfer.

But for compatibility, I suppose we can duplicate the information.

MrReeMachine, 2017-02-13:
That's excellent news thanks for the hard work everyone!

@TheTechnobear - I'm a relative novice with Unix (though a very experienced software developer on other platforms) - I desperately need to get this issue resolved to complete a project I'm working on - is it at all possible you could talk me through, or point me in the right direction to apply the patch as you did as I don't really have time to wait for this fix to filter through a proper Raspbian release?

Many thanks,
Rich.

P33M, 2017-02-13:
I've done a brief survey of various HCDs and it looks like they all sum urb->actual_length for isochronous transfers.

I'd be tempted to go with the majority behaviour rather than to the letter of a kerneldoc comment.

@mcerveny can you submit a pull request with this patch please?

closed 2017-02-13 by popcornmix

popcornmix, 2017-02-13:
Latest rpi-update firmware contains the patch from @mcerveny.

R030t1, 2018-06-30:
I am experiencing similar symptoms on a x86_64 host with a USB 2.0 FS device. The issue mostly goes away if I continuously receive packets. Receiving a few returns zeros, or data with a few transfers at the start filled with zeros.

I realize this bug went in a device-specific direction but can anyone comment on the presence of this bug on x86_64 hosts?

JamesH65, 2018-06-30:
I doubt it - this is a github issue tracker for the Raspberry Pi, a usually 32bit ARM system...