raspberrypi/linux #904: dtoverlay=w1-gpio issue at A+ board  with piCore-6.1alpha4 kernel 3.18.9

jgrulich, 2015-03-21:
The dtoverlay=w1-gpio is not working at A+ board and booting hangs up at the first "rainbow" screen.
At the B+ board the same SD card works normally. The only way how to bring the A+ board working with this kernel is to disable the DT  with "device_tree=" and use the "traditional modprobe" to load the modules. Than everything is working well.

pelwell, 2015-03-21:
What is in your config.txt and cmdline.txt?

pelwell, 2015-03-21:
It might also help if, for the DT disabled case, you show the modules you are loading and the values of any parameters.

jgrulich, 2015-03-21:
cmdline.txt:
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/ram0 elevator=deadline rootwait quiet nortc loglevel=3 noembed

config.txt:
initramfs 6.1alpha4.gz followkernel
dtparam=i2c=on
dtoverlay=w1-gpio

The last line cause the issue.

I need the i2c for the RTC PCF8563 device which I load with:
echo pcf8563 0x51 > /sys/bus/i2c/devices/i2c-1/new_device
And w1 for the DS18B20 temperature sensor is loaded automatically when used dtoverlay=w1-gpio.
It works for B+ and RPi 2 boards, but NOT for A+ board.
When the DT is disabled, than i2c is enabled by default, means I don't need to do anything with it and I only load the w1 module with modprobe w1-gpio.

pelwell, 2015-03-21:
Your configuration looks sensible. I will add support for the 8563 to the i2c-rtc overlay, which will replace the dtparam=i2c and the echo with dtoverlay=i2c-rtc,pcf8563.

I can't think of a reason why it would behave differently on the A+, unless it is to do with RAM size. We don't really have the bandwidth to support other kernel builds - that's down to the provider of the kernel, although I see he has already bounced you here - but I will give it a try to see if it is anything obvious. However, I don't have an A+ here to test it on, so it will have to wait until Monday.

jgrulich, 2015-03-21:
I've tested kernel 3.18.6 which seems to have the similar overlay as 3.18.9 and w1-gpio is not working there with A+ board too.
Than I've tested the first kernel with DT which is 3.18.5 and there it is working well. This overlay version is quite simple and may be simply identified, because doesn't have the RED PWR LED control, means that the RED LED is ON during operation. The not working overlay has RED LED control and it's normally OFF.

pelwell, 2015-03-21:
Which firmware (start.elf) version are you using? If it isn't too much hassle, I would recommend doing an rpi-update to get the latest, then restoring your kernel and modules add needed. Actually, you could try booting the stock kernel before restoring your custom one.

jgrulich, 2015-03-21:
Sure, I've tested all available firmware packages, included the one which is here: https://github.com/raspberrypi/firmware/tree/master/boot. The same result. B+ is working, A+ not.
 Seems to me that there is some issue with the RAM mapping and it's related only to the kernel version. The latest working is 3.18.5.
 BTW I've found another overlay which is not working: hifiberry-dacplus.

jgrulich, 2015-03-22:
I've tested another board to be sure that it's not the HW issue. The non-functionality is probably related to the HW difference between the A+ and B+. There is another RAM and no USB/LAN hub (SMSC), but don't know how this may have influence on the GPIO port pins.

pelwell, 2015-03-23:
So far I've been able to reproduce the problem - a kernel panic - but I don't yet know what is causing it. Not having a build system for the kernel makes it harder...

When it crashes you get this output on the serial console:
```
Uncompressing Linux... done, booting the kernel.
[    1.339594] Unable to handle kernel NULL pointer dereference at virtual address 0000000d
[    1.347717] pgd = c0004000
[    1.350431] [0000000d] *pgd=00000000
[    1.354031] Internal error: Oops: 17 [#1] PREEMPT ARM
[    1.359092] Modules linked in:
[    1.362166] CPU: 0 PID: 1 Comm: swapper Not tainted 3.18.9-piCore+ #1
[    1.368621] task: cac78000 ti: cac3c000 task.ti: cac3c000
[    1.374043] PC is at pcpu_alloc+0xf0/0x804
[    1.378152] LR is at pcpu_alloc+0xac/0x804
[    1.382257] pc : [<c0105fb0>]    lr : [<c0105f6c>]    psr: 00000093
[    1.382257] sp : cac3dcf0  ip : cac3dcf0  fp : cac3dd44
[    1.393751] r10: 00000001  r9 : cac3c000  r8 : 00000000
[    1.398983] r7 : c08193c4  r6 : 00000008  r5 : 00000004  r4 : 00000001
[    1.405519] r3 : cbfefca0  r2 : cbfefca8  r1 : 00000004  r0 : 00000001
[    1.412057] Flags: nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
[    1.419464] Control: 00c5387d  Table: 0a2c8008  DAC: 00000015
[    1.425218] Process swapper (pid: 1, stack limit = 0xcac3c1b0)
[    1.431059] Stack: (0xcac3dcf0 to 0xcac3e000)
[    1.435428] dce0:                                     00000000 00000000 cac3c030 00000004
[    1.443625] dd00: c086c136 80000013 00000000 00000000 c086c136 00000000 00000000 cad81900
[    1.451821] dd20: 00000004 c06b34e4 c00172f8 00000000 ca422000 cac3c010 cac3dd54 cac3dd48
[    1.460018] dd40: c0106708 c0105ecc cac3dd7c cac3dd58 c00dbaa4 c01066f4 00000001 cac3dd74
[    1.468215] dd60: cac3dd9c cad81900 c0016f3c 00000004 cac3dd9c cac3dd80 c001784c c00dba80
[    1.476426] dd80: cad81900 c0869b90 cad81900 c081e32c cac3ddcc cac3dda0 c0018768 c00177b4
[    1.484630] dda0: c0018670 c08f8f30 cad28610 c081d938 fffffdfb 00000000 ca422000 cac3c010
[    1.492833] ddc0: cac3ddec cac3ddd0 c0374dac c001867c c08f8f30 cad28610 00000000 c081d938
[    1.501035] dde0: cac3de14 cac3ddf0 c0372eb8 c0374d64 00000007 cad28610 c081d938 cad28644
[    1.509237] de00: 00000000 c081b7f0 cac3de34 cac3de18 c0373158 c0372e3c c03730b0 00000000
[    1.517440] de20: c081d938 c03730b0 cac3de5c cac3de38 c03711b0 c03730bc cac2370c cad07e10
[    1.525642] de40: ca2ce034 c081d938 ca2ce000 c08504b8 cac3de6c cac3de60 c0372a40 c0371140
[    1.533845] de60: cac3de94 cac3de70 c03726c4 c0372a24 c06b3624 cac3de80 c081d938 c081b7f0
[    1.542047] de80: c07bee24 00000000 cac3deac cac3de98 c0373af8 c037257c c080ecfc c081b7f0
[    1.550250] dea0: cac3debc cac3deb0 c0374ccc c0373a7c cac3decc cac3dec0 c07bee3c c0374c80
[    1.558452] dec0: cac3df54 cac3ded0 c0008750 c07bee30 00000000 00000000 cac3df04 00000000
[    1.566654] dee0: cac3df0c cac3def0 cac3df0c cac3def8 c07ba5ec 00000000 cbfff3f9 c05b1bd8
[    1.574856] df00: cac3df54 cac3df10 c003e998 c07ba5f8 00000000 00000006 00000006 cbfff400
[    1.583058] df20: c06f90d4 00000000 00000000 c080ecfc 00000006 c0869960 000000ed c0869960
[    1.591260] df40: c07f2b40 c07ba5ec cac3df94 cac3df58 c07badec c00086cc 00000006 00000006
[    1.599462] df60: c07ba5ec 00000001 cac3df94 00000000 c0562b68 00000000 00000000 00000000
[    1.607665] df80: 00000000 00000000 cac3dfac cac3df98 c0562b80 c07bad00 cac3c000 00000000
[    1.615865] dfa0: 00000000 cac3dfb0 c000ec28 c0562b74 00000000 00000000 00000000 00000000
[    1.624066] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    1.632267] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000 d67b7fff ffffdffa
[    1.640495] [<c0105fb0>] (pcpu_alloc) from [<c0106708>] (__alloc_percpu+0x20/0x24)
[    1.648104] [<c0106708>] (__alloc_percpu) from [<c00dbaa4>] (perf_pmu_register+0x30/0x2b8)
[    1.656414] [<c00dbaa4>] (perf_pmu_register) from [<c001784c>] (armpmu_register+0xa4/0xd4)
[    1.664723] [<c001784c>] (armpmu_register) from [<c0018768>] (cpu_pmu_device_probe+0xf8/0x39c)
[    1.673387] [<c0018768>] (cpu_pmu_device_probe) from [<c0374dac>] (platform_drv_probe+0x54/0xb4)
[    1.682215] [<c0374dac>] (platform_drv_probe) from [<c0372eb8>] (really_probe+0x88/0x21c)
[    1.690430] [<c0372eb8>] (really_probe) from [<c0373158>] (__driver_attach+0xa8/0xac)
[    1.698295] [<c0373158>] (__driver_attach) from [<c03711b0>] (bus_for_each_dev+0x7c/0xb0)
[    1.706508] [<c03711b0>] (bus_for_each_dev) from [<c0372a40>] (driver_attach+0x28/0x30)
[    1.714547] [<c0372a40>] (driver_attach) from [<c03726c4>] (bus_add_driver+0x154/0x200)
[    1.722586] [<c03726c4>] (bus_add_driver) from [<c0373af8>] (driver_register+0x88/0x108)
[    1.730715] [<c0373af8>] (driver_register) from [<c0374ccc>] (__platform_driver_register+0x58/0x6c)
[    1.739803] [<c0374ccc>] (__platform_driver_register) from [<c07bee3c>] (register_pmu_driver+0x18/0x20)
[    1.749239] [<c07bee3c>] (register_pmu_driver) from [<c0008750>] (do_one_initcall+0x90/0x1dc)
[    1.757801] [<c0008750>] (do_one_initcall) from [<c07badec>] (kernel_init_freeable+0xf8/0x1bc)
[    1.766450] [<c07badec>] (kernel_init_freeable) from [<c0562b80>] (kernel_init+0x18/0xfc)
[    1.774671] [<c0562b80>] (kernel_init) from [<c000ec28>] (ret_from_fork+0x14/0x20)
[    1.782272] Code: e5944000 e0822006 e1540002 0a0000d1 (e594300c)
[    1.788433] ---[ end trace 09fc774a00078b77 ]---
[    1.793076] note: swapper[1] exited with preempt_count 1
[    1.798586] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[    1.798586]
[    1.807771] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[    1.807771]
```

It isn't clear why this would be different on the A+ than the B+.

It is tempting to say this a problem in the kernel configuration, but I'm not confident of that fact yet.

jgrulich, 2015-03-24:
Just mentioned that the i2c-rtc-overlay was updated, I've replaced it and added into the config.txt and it's strange, but this brings the w1-gpio-overlay working. But only these two together.
config.txt is now this:

initramfs 6.1alpha4.gz followkernel
dtparam=pwr_led_activelow
dtoverlay=w1-gpio
dtoverlay=i2c-rtc,pcf8563

Other issues, like a hifiberry-dac-overlay, or hifiberry-dacplus-overlay alone,means without w1 and i2c-rtc remain not working with the kernel panic.

pelwell, 2015-03-24:
I'm happy that the situation has improved somewhat, but I can't make any further contributions at the moment. This issue has already consumed too much of my time - working with just a binary image of a kernel built by someone else does not make sense.

jgrulich, 2015-03-24:
Thank's for your effort. For now I'll use it with DT dissabled and will wait for the new kernel.
Maybe that than it will work better. Once again thank's a lot for the i2c-rtc update with new devices.

closed 2015-03-24 by jgrulich

pelwell, 2015-03-25:
The latest firmware, containing the fix for the issue I found, has been made available via rpi-update.