raspberrypi/linux #905: direct snd-soc-rpi-wsp Wolfson Cirrus Logic support for 3.18.9 kernel

jgrulich, 2015-03-21:
I was looking for some ready made support for the Wolfson Sound Card support for the 3.18.9 kernel in piCore-6.1 alpha4, but no lucky. I don't like to build it, or test some unofficial ways.

popcornmix, 2015-03-22:
This is up to wolfson. They submitted a PR https://github.com/raspberrypi/linux/pull/705, but it is not finished, and we are waiting for an update. Please report this to them.

jgrulich, 2015-03-23:
Hence the PR #705 was closed and that Wolfson is now part of Cirrus Logic, really don't know how to contact them for reporting.

popcornmix, 2015-03-23:
Perhaps start here:
http://www.element14.com/community/community/raspberry-pi/raspberry-pi-accessories/cirrus_logic_audio_card/blog/2015/01/16/cirrus-logic-audio-card-2015-update
and
https://github.com/CirrusLogic/rpi-linux

But without a driver pull request, we're not able to support this.

closed 2015-03-23 by popcornmix