raspberrypi/linux #912: rpi-3.19.y: CONFIG_IKCONFIG changed from builtin to module

notro, 2015-03-27:
From 3.18 to 3.19, CONFIG_IKCONFIG has changed from builtin to module.
Both bcmrpi_defconfig and bcm2709_defconfig.

I use /proc/config.gz often so for me it's a step back.
Was this change intentional?

popcornmix, 2015-03-27:
It was intentional (but I'm willing to listen to other viewpoints).

It seems having ~30K always loaded as part of kernel.img and sitting in memory for something very rarely used is inefficient.
```
sudo modprobe configs && zcat /proc/config.gz
```
will get you the information you need. You could also add configs to your list of default modules, if you'd like it even more accessible. Is this not sufficient?

notro, 2015-03-27:
I wondered if it could be about size. I can live with this.

closed 2015-03-27 by notro