raspberrypi/linux #914: Kernel panic on Pi 1 B+ when using an initramfs

diederikdehaas, 2015-03-31:
I'm the maintainer of [raspbian-ua-netinst](https://github.com/debian-pi/raspbian-ua-netinst) and we've had several reports about kernel panics on the Pi 1 B/B+ (see https://github.com/debian-pi/raspbian-ua-netinst/issues/199 and https://github.com/debian-pi/raspbian-ua-netinst/issues/201).
By default it's using the raspbian.org kernel and an initramfs and I have multiple times 'fixed' the issue by disabling the initramfs (which is generated/used by default with the raspbian.org kernel).

But that kernel is not (exactly) the same as the kernel build from this repo.
Plugwash told me on IRC that he was able to get a kernel panic by using an initramfs with a rpf kernel (this repo), so I tried whether I could accomplish the same ... and I have.

Procedure: 
- install a basic system using the raspbian-ua-netinst and reboot
- this may or may not result in a kernel panic right away
- assuming it boots correctly, log in to the system with user `root` and password `raspbian` (the default)
- install the `raspberrypi-bootloader` package to get the rpf kernel (it will uninstall some other packages)
- reboot into the new kernel
- generate an initramfs by executing `/etc/kernel/postinst.d/initramfs-tools $(uname -r)`
- modify `/boot/config.txt` so it contains the line `initramfs initrd.img-3.18.7+ followkernel` (I have it under a [pi1] header)
- reboot and see it has a kernel panic with the rpf kernel/initramfs too

If it doesn't give a kernel panic, re-generating the initramfs with the earlier mentioned command often/sometimes gives a kernel panic on reboot. The 'problem' is that it doesn't consistently fail.

Log from start to kernel panic (which is also [stored here](https://gist.github.com/diederikdehaas/9a67b6afd902ce75788e), more dumps in https://github.com/debian-pi/raspbian-ua-netinst/issues/199):
```
Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.18.7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #755 PREEMPT Thu Feb 12 17:14:31 GMT 2015
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine model: Raspberry Pi Model B+
[    0.000000] cma: Reserved 8 MiB at 0x1b400000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 113792
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708.boardrev=0x10 bcm2708.serial=0x9dee0f68 smsc95xx.macaddr=B8:27:EB:EE:0F:68 bcm2708_fb.fbswap=1 bcm2708.disk_led_gpio=47 bcm2708.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 elevator=deadline root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 433896K/458752K available (5926K kernel code, 358K rwdata, 1876K rodata, 340K init, 734K bss, 24856K reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
[    0.000000]     vmalloc : 0xdc800000 - 0xff000000   ( 552 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc07a6ad8   (7803 kB)
[    0.000000]       .init : 0xc07a7000 - 0xc07fc000   ( 340 kB)
[    0.000000]       .data : 0xc07fc000 - 0xc085588c   ( 359 kB)
[    0.000000]        .bss : 0xc085588c - 0xc090d128   ( 735 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:522
[    0.000026] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483648000ns
[    0.000076] Switching to timer-based delay loop, resolution 1000ns
[    0.000356] Console: colour dummy device 80x30
[    0.001441] console [tty1] enabled
[    0.001487] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.001560] pid_max: default: 32768 minimum: 301
[    0.001941] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.002008] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.003018] Initializing cgroup subsys memory
[    0.003112] Initializing cgroup subsys devices
[    0.003173] Initializing cgroup subsys freezer
[    0.003227] Initializing cgroup subsys net_cls
[    0.003277] Initializing cgroup subsys blkio
[    0.003402] CPU: Testing write buffer coherency: ok
[    0.003516] ftrace: allocating 19479 entries in 58 pages
[    0.111593] Setting up static identity map for 0x55d058 - 0x55d0b4
[    0.114413] devtmpfs: initialized
[    0.131909] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.135104] pinctrl core: initialized pinctrl subsystem
[    0.137859] NET: Registered protocol family 16
[    0.143394] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[... 101 lines cut ...]
[    2.230284] [<c02f9954>] (strcmp) from [<c045af54>] (of_bus_isa_match+0x20/0x30)
[    2.237719] [<c045af54>] (of_bus_isa_match) from [<c045ad94>] (of_match_bus+0x2c/0x70)
[    2.245675] [<c045ad94>] (of_match_bus) from [<c045b310>] (__of_translate_address+0xa0/0x294)
[    2.254239] [<c045b310>] (__of_translate_address) from [<c045ba14>] (of_address_to_resource+0x7c/0x100)
[    2.263683] [<c045ba14>] (of_address_to_resource) from [<c0315d84>] (bcm2835_pinctrl_probe+0x58/0x400)
[    2.273049] [<c0315d84>] (bcm2835_pinctrl_probe) from [<c0367698>] (platform_drv_probe+0x54/0xb4)
[    2.281969] [<c0367698>] (platform_drv_probe) from [<c03656e8>] (really_probe+0x88/0x21c)
[    2.290187] [<c03656e8>] (really_probe) from [<c0365988>] (__driver_attach+0xa8/0xac)
[    2.298056] [<c0365988>] (__driver_attach) from [<c0363a84>] (bus_for_each_dev+0x70/0xa4)
[    2.306270] [<c0363a84>] (bus_for_each_dev) from [<c0365274>] (driver_attach+0x28/0x30)
[    2.314311] [<c0365274>] (driver_attach) from [<c0364ef8>] (bus_add_driver+0x154/0x200)
[    2.322354] [<c0364ef8>] (bus_add_driver) from [<c0366328>] (driver_register+0x88/0x108)
[    2.330484] [<c0366328>] (driver_register) from [<c03675b8>] (__platform_driver_register+0x58/0x6c)
[    2.339573] [<c03675b8>] (__platform_driver_register) from [<c07ce124>] (bcm2835_pinctrl_driver_init+0x18/0x20)
[    2.349703] [<c07ce124>] (bcm2835_pinctrl_driver_init) from [<c0008760>] (do_one_initcall+0x90/0x1e4)
[    2.358979] [<c0008760>] (do_one_initcall) from [<c07a7e08>] (kernel_init_freeable+0xf8/0x1b8)
[    2.367636] [<c07a7e08>] (kernel_init_freeable) from [<c05513d0>] (kernel_init+0x18/0xfc)
[    2.375860] [<c05513d0>] (kernel_init) from [<c000e988>] (ret_from_fork+0x14/0x20)
[    2.383462] Code: e24cb004 ea000001 e3530000 0a000007 (e4d03001)
[    2.389809] ---[ end trace 4c6abb27342ccfdb ]---
[    2.394525] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[    2.394525]
[    2.403702] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[    2.403702]
```

popcornmix, 2015-03-31:
Can you test if it does it with the latest "rpi-update" firmware/kernel?

diederikdehaas, 2015-03-31:
I've managed to also produce this kernel panic starting from the `2015-02-16-raspbian-wheezy.img` image:
- Write image to sd card and boot up
- do `sudo apt-get update` and `sudo apt-get dist-upgrade` and reboot
- generate initramfs with `/etc/kernel/postinst.d/initramfs-tools $(uname -r)`
- modify `/boot/config.txt` so it has the line `initramfs initrd.img-3.18.7+ followkernel`. This time I did not put it under a `[pi1]` header
- reboot ... and see a kernel panic

Log:
```
Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.18.7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #755 PREEMPT Thu Feb 12 17:14:31 GMT 2015
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine model: Raspberry Pi Model B+
[    0.000000] cma: Reserved 8 MiB at 0x1b400000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 113792
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708.boardrev=0x10 bcm2708.serial=0x9dee0f68 smsc95xx.macaddr=B8:27:EB:EE:0F:68 bcm2708_fb.fbswap=1 bcm2708.disk_led_gpio=47 bcm2708.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 433244K/458752K available (5926K kernel code, 358K rwdata, 1876K rodata, 340K init, 734K bss, 25508K reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
[    0.000000]     vmalloc : 0xdc800000 - 0xff000000   ( 552 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc07a6ad8   (7803 kB)
[    0.000000]       .init : 0xc07a7000 - 0xc07fc000   ( 340 kB)
[    0.000000]       .data : 0xc07fc000 - 0xc085588c   ( 359 kB)
[    0.000000]        .bss : 0xc085588c - 0xc090d128   ( 735 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:522
[    0.000024] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483648000ns
[    0.000075] Switching to timer-based delay loop, resolution 1000ns
[    0.000355] Console: colour dummy device 80x30
[    0.001430] console [tty1] enabled
[    0.001476] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.001552] pid_max: default: 32768 minimum: 301
[    0.001932] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001998] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.002999] Initializing cgroup subsys memory
[    0.003093] Initializing cgroup subsys devices
[    0.003154] Initializing cgroup subsys freezer
[    0.003208] Initializing cgroup subsys net_cls
[    0.003259] Initializing cgroup subsys blkio
[    0.003382] CPU: Testing write buffer coherency: ok
[    0.003496] ftrace: allocating 19479 entries in 58 pages
[    0.111557] Setting up static identity map for 0x55d058 - 0x55d0b4
[    0.114386] devtmpfs: initialized
[    0.131854] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.135042] pinctrl core: initialized pinctrl subsystem
[    0.137794] NET: Registered protocol family 16
[    0.143290] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[    0.171337] cpuidle: using governor ladder
[    0.201399] cpuidle: using governor menu
[    0.201916] bcm2708.uart_clock = 3000000
[    0.205005] No ATAGs?
[    0.205071] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.205133] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.205199] mailbox: Broadcom VideoCore Mailbox driver
[    0.205365] bcm2708_vcio: mailbox at f200b880
[    0.205857] bcm_power: Broadcom power driver
[    0.205915] bcm_power_open() -> 0
[    0.205945] bcm_power_request(0, 8)
[... 90 lines cut ...]
[    2.302064] [<c02f9954>] (strcmp) from [<c045af54>] (of_bus_isa_match+0x20/0x30)
[    2.309503] [<c045af54>] (of_bus_isa_match) from [<c045ad94>] (of_match_bus+0x2c/0x70)
[    2.317460] [<c045ad94>] (of_match_bus) from [<c045b310>] (__of_translate_address+0xa0/0x294)
[    2.326023] [<c045b310>] (__of_translate_address) from [<c045ba14>] (of_address_to_resource+0x7c/0x100)
[    2.335467] [<c045ba14>] (of_address_to_resource) from [<c0315d84>] (bcm2835_pinctrl_probe+0x58/0x400)
[    2.344834] [<c0315d84>] (bcm2835_pinctrl_probe) from [<c0367698>] (platform_drv_probe+0x54/0xb4)
[    2.353753] [<c0367698>] (platform_drv_probe) from [<c03656e8>] (really_probe+0x88/0x21c)
[... cut ...]
[    2.439425] [<c07a7e08>] (kernel_init_freeable) from [<c05513d0>] (kernel_init+0x18/0xfc)
[    2.447654] [<c05513d0>] (kernel_init) from [<c000e988>] (ret_from_fork+0x14/0x20)
[    2.455256] Code: e24cb004 ea000001 e3530000 0a000007 (e4d03001)
[    2.461607] ---[ end trace 682122b6d7e0f900 ]---
[    2.466322] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[    2.466322]
[    2.475500] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[    2.475500]
```

diederikdehaas, 2015-03-31:
> Can you test if it does it with the latest "rpi-update" firmware/kernel?

Will do that now.

diederikdehaas, 2015-03-31:
I have now regenerated the initramfs 5 times and I haven't got a kernel panic (yet?).
Often times it needs to fail in the first try, so I'll setup the system from scratch a few more times.
If you want me to test other things and/or commit id, let me know.
But there is a real possibility that 3.18.10+ fixes the kernel panic.

diederikdehaas, 2015-04-02:
I'm quite sure the issue is fixed with the 3.18.10+ kernel. 
Plugwash has made a kernel which is similar to 3.18.10+ and it's fixed there too.
So, feel free to close this issue.

diederikdehaas, 2015-04-04:
I cheered too soon, I have just gotten a kernel crash with with the latest kernel from `rpi-update` on a Pi 1 B+ (with an initramfs).
I'm now also trying to get it to crash on a Pi 1 B, but the hard thing about this issue is that it doesn't happen consistently.

I have done the installation with the raspbian-ua-netinst with some custom configuration (release=jessie) and a script which is run near the end of the installation which installs and runs `rpi-update`, generates an initramfs for that kernel and specifies that in `/boot/config.txt` so that the first boot of the new system is with the `rpi-update` kernel.

I can provide the configuration files for the installer if you like with which I have gotten the kernel panic.

diederikdehaas, 2015-04-04:
And it happened on a Pi 1 B too, this time with release=jessie

diederikdehaas, 2015-04-04:
Shiftplusone/@XECDesign asked for an image which would have the crashing kernel and I have just uploaded one: http://cknow.org/rpi/raspbian-wheezy-rpi1b-kernel-panic.img.tar.gz

I have shrunk the image so it should fit on any 8GB card. You can log into the image with user `root` and password `raspbian`. The output defaults to the serial console and if you don't like that just change the config in `/boot/cmdline.txt`.
The most interesting thing can be found in `/boot/config.txt` with the following contents:
```sh
[pi1]
#kernel=vmlinuz-3.18.0-trunk-rpi
#initramfs initrd.img-3.18.0-trunk-rpi followkernel
kernel=vmlinuz-3.18.10+
initramfs initrd.img-3.18.10+ followkernel
#initramfs initrd.img-3.18.10+-crash followkernel
```
The 3.18.10+ kernel is installed from `rpi-update` which ran around 2015-04-04T13:19:00.
As you can see there's also a commented out initramfs line for the 3.18.10+ kernel and as the name suggests, that is the initramfs which crashed. So if you want to see it crash, remove the '#' in front of that line and put a '#' in front of the other. That way you can toggle between a working and a crashing system. Both initramfs files are present in `/boot/` so that can be inspected as well.

If you have any further questions, just let me know and I'll answer them as best as I can.

diederikdehaas, 2015-04-05:
It took a while and **many** tries, but there is now also an image from a kernel panic on a RPi 1B+ with Raspbian Jessie as OS version: http://cknow.org/rpi/raspbian-jessie-rpi1b+-kernel-panic.img.tar.gz

Apart from now also having a documented kernel crash on a RPi 1B+ (next to a 1B) another (potential) relevant difference is a newer version (from Jessie) of `initramfs-tools` used to generate the initramfs.
Other then the above described differences, the images are the same.
Hope this further helps to find the cause and solution for the reported issue.

pelwell, 2015-04-07:
Sadly that image is too large for the 8GB cards I have at my disposal:

NOOBS - 15415296 (eb3800) sectors = 7,892,631,552 bytes
Transcend - 15523840 (0xece000) sectors = 7,948,206,080 bytes
but
Image - 15644672 (eeb800) sectors = 8,010,072,064 bytes

So the image is at least 61MB too large.

I'll bring in a larger card tomorrow and see if that works.

diederikdehaas, 2015-04-08:
I've made them smaller, so they now they should (really) fit on a 8GB card:
http://cknow.org/rpi/raspbian-wheezy-rpi1b-kernel-panic-6GB.img.tar.gz
http://cknow.org/rpi/raspbian-jessie-rpi1b+-kernel-panic-6GB.img.tar.gz

pelwell, 2015-04-08:
I've downloaded both images, and both produce crashes on a B and a B+. Interestingly, they crash in different locations on the two models, but between runs and the two images, all crashes are remarkably consistent.

On the B+, the abridged callstack looks like this:
```
[    1.931867] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[    1.940006] pgd = c0004000
[    1.942882] [00000000] *pgd=00000000
...
[    1.966102] PC is at strcmp+0x18/0x40
[    1.969792] LR is at of_bus_isa_match+0x20/0x30
[    1.974340] pc : [<c02f8034>]    lr : [<c04512ec>]    psr: a0000013
[    1.974340] sp : da83bc50  ip : da83bc60  fp : da83bc5c
[    1.985842] r10: c083874c  r9 : 00000000  r8 : dbff0444
[    1.991079] r7 : 00000200  r6 : c0000808  r5 : dbfef120  r4 : c0838730
[    1.997619] r3 : c04512cc  r2 : dbfef120  r1 : c06e65a0  r0 : 00000000
...
[    2.273602] [<c02f8034>] (strcmp) from [<c04512ec>] (of_bus_isa_match+0x20/0x30)
[    2.281038] [<c04512ec>] (of_bus_isa_match) from [<c045112c>] (of_match_bus+0x2c/0x70)
[    2.288994] [<c045112c>] (of_match_bus) from [<c04516a8>] (__of_translate_address+0xa0/0x294)
[    2.297557] [<c04516a8>] (__of_translate_address) from [<c0451dac>] (of_address_to_resource+0x7c/0x100)
[    2.306998] [<c0451dac>] (of_address_to_resource) from [<c0314418>] (bcm2835_pinctrl_probe+0x54/0x48c)
[    2.316347] [<c0314418>] (bcm2835_pinctrl_probe) from [<c0369b78>] (platform_drv_probe+0x3c/0x6c)
[    2.325271] [<c0369b78>] (platform_drv_probe) from [<c0368110>] (really_probe+0x88/0x21c)
[    2.333490] [<c0368110>] (really_probe) from [<c03683b0>] (__driver_attach+0xa8/0xac)
[    2.341361] [<c03683b0>] (__driver_attach) from [<c03664bc>] (bus_for_each_dev+0x70/0xa4)
```
while on a B you get:
```
[    1.813861] Unable to handle kernel paging request at virtual address 0012a42c
[    1.821107] pgd = c0004000
[    1.823824] [0012a42c] *pgd=00000000
...
[    1.847020] PC is at pcpu_alloc+0x1cc/0x83c
[    1.851220] LR is at pcpu_alloc+0x188/0x83c
[    1.855418] pc : [<c01032cc>]    lr : [<c0103288>]    psr: 00000093
[    1.855418] sp : cac3bce0  ip : cac3bce0  fp : cac3bd34
[    1.866918] r10: c07f93c4  r9 : cac3a000  r8 : 00000000
[    1.872155] r7 : 0000000c  r6 : 00000004  r5 : 00000060  r4 : 0012a420
[    1.878694] r3 : cbfff080  r2 : 00000000  r1 : c07f93c4  r0 : 00000001
...
[    2.113744] [<c01032cc>] (pcpu_alloc) from [<c0103980>] (__alloc_percpu+0x20/0x24)
[    2.121365] [<c0103980>] (__alloc_percpu) from [<c00d91c8>] (perf_pmu_register+0x30/0x2c0)
[    2.129682] [<c00d91c8>] (perf_pmu_register) from [<c0018410>] (armpmu_register+0xa4/0xd4)
[    2.137992] [<c0018410>] (armpmu_register) from [<c0019310>] (cpu_pmu_device_probe+0xf8/0x39c)
[    2.146647] [<c0019310>] (cpu_pmu_device_probe) from [<c0369b78>] (platform_drv_probe+0x3c/0x6c)
[    2.155488] [<c0369b78>] (platform_drv_probe) from [<c0368110>] (really_probe+0x88/0x21c)
[    2.163709] [<c0368110>] (really_probe) from [<c03683b0>] (__driver_attach+0xa8/0xac)
[    2.171581] [<c03683b0>] (__driver_attach) from [<c03664bc>] (bus_for_each_dev+0x70/0xa4)
[    2.179796] [<c03664bc>] (bus_for_each_dev) from [<c0367c9c>] (driver_attach+0x28/0x30)
[    2.187838] [<c0367c9c>] (driver_attach) from [<c0367930>] (bus_add_driver+0x154/0x200)
```
Getting different failures like this suggests that both code paths stumble across memory corruption of some kind, but that's just a hunch.

By correlating multiple runs it appears that the B+ code is failing after accessing [r0], while the B code is reading [r4 + 0xc]. The next step would be to extract the kernel and see if the source of the two addresses can be determined by disassembly.

pelwell, 2015-04-08:
[ The comments above refer to the larger images, after extracting them onto a 32GB card. ]

diederikdehaas, 2015-04-08:
> but between runs and the two images, all crashes are remarkably consistent

That's what I noticed too. In [this comment](https://github.com/debian-pi/raspbian-ua-netinst/issues/199#issuecomment-84358071) there are a number of logs from boot till the kernel crash and they all look very much alike.

> [ The comments above refer to the larger images, after extracting them onto a 32GB card. ]

The new images are exactly the same as the old ones. The only thing I did was further shrink the root partition with gparted.

diederikdehaas, 2015-04-14:
Interesting development.

While preparing a new release for 'my' installer, I did some testing and now I can reproduce the crash ALWAYS on a Pi 1B, but not on a Pi 1B+. 
That installer is using the raspbian.org kernel `3.18.5-1~exp1+rpi16 (2015-03-28)`.
For 'kicks' (and to hopefully make it easier for you), I added the `kernel.img` from the `raspberrypi-bootloader_1.20150214-2_armhf` package ... and it crashed again.
Then I took the `kernel.img` from the `raspberrypi-bootloader_1.20150411~172641~nightly_armhf` package (from [this directory](http://nightly.raspberrypi.org/nightlyfirmware/pool/main/r/raspberrypi-firmware/) based on [plugwash's post in the forums](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=80023)) ... and it crashed again.

So now I have uploaded installer files which contain all 3 kernels with various (commented out) lines for those kernels in `config.txt` so you can easily switch between them.

http://cknow.org/rpi/raspbian-ua-netinst-20150414-git6afb9a1.img
http://cknow.org/rpi/raspbian-ua-netinst-20150414-git6afb9a1.img.bz2
http://cknow.org/rpi/raspbian-ua-netinst-20150414-git6afb9a1.img.xz

They are all the same, just different compression formats.
I always use the *.img.xz file and the official instruction for writing it from a linux host system can be found [here](https://github.com/debian-pi/raspbian-ua-netinst#sd-card-image-for-linux) (if needed)

diederikdehaas, 2015-04-14:
Here are the dumps of the crashes from all 3 kernels: 
https://gist.github.com/diederikdehaas/9756b355e5ae81184912

diederikdehaas, 2015-04-14:
Based on shiftplusone's comment on IRC, I also tested it with the kernel from `rpi-update` (20150414-1338 UTC) and got the same result (ie crash).
I updated the previously mentioned gist with the dump from the rpi-update-kernel

diederikdehaas, 2015-04-20:
Maybe it isn't the initramfs after all.
Today I did some tests by using the master files from the firmware repo ... and the crashes went away :-)

diederikdehaas, 2015-04-21:
After some more testing, I just got a case in which it did crash :-/
Disabling DeviceTree in `/boot/config.txt` fixed it (again) though. 
That 'trick' was the reason for me trying out new firmware files.

diederikdehaas, 2015-04-25:
I'm now rather convinced the problem is related to DeviceTree.

When using the installer image from https://github.com/raspberrypi/linux/issues/914#issuecomment-92817668 but adding `device_tree=` to the `[pi1]` section of `config.txt` (to disable DeviceTree) then the installer does work.
Later on with an installed system with DeviceTree enabled, I got a kernel crash on boot again. 
Disabled DeviceTree and the system booted again.

georgewhr, 2015-09-23:
guys, I have a very similar issue. I'm using multi image(kernel+ramdisk+dtb) to bootup my system, which is armv7 based on 3.14.43 kernel, and it shows up the crash log as below(the crash happened randomly)

    1.127886] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[    1.131799] pgd = c0204000
[    1.139948] [00000000] *pgd=00000000
[    1.145851] Internal error: Oops: 5 [#1] PREEMPT SMP ARM
[    1.146116] Modules linked in:
[    1.154360] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 3.14.43 #2
[    1.154454] task: cf078000 ti: cf076000 task.ti: cf076000
[    1.160536] PC is at strcmp+0x0/0x34
[    1.165828] LR is at __of_find_property+0x2c/0x54
[    1.169471] pc : [<c03f4ac4>]    lr : [<c0579698>]    psr: a0000093
[    1.169471] sp : cf077dd8  ip : c0842b94  fp : c0a1557c
[    1.174078] r10: cfdedae8  r9 : 00000001  r8 : cfdf8e08
[    1.185354] r7 : 00000000  r6 : c08bf1d6  r5 : 00000000  r4 : cfdedb44
[    1.190564] r3 : 0000006c  r2 : 00000063  r1 : c08bf1d6  r0 : 00000000
[    1.197162] Flags: NzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
[    1.203674] Control: 10c5387d  Table: 8020406a  DAC: 00000015
[    1.211221] Process swapper/0 (pid: 1, stack limit = 0xcf076238)
[    1.216866] Stack: (0xcf077dd8 to 0xcf078000)
[    1.222942] 7dc0:                                                       c08678da 00000000
[    1.227207] 7de0: 00000000 c057a1f4 cfdedae8 00000000 c08678da 60000013 cfdf8e08 00000001
[    1.235368] 7e00: cf1e9540 c057a380 000003e8 d0232000 c0a1557c cf1e84c0 00000000 c04e5c18
[    1.243526] 7e20: 00000026 00000073 000000ea c030bad8 cf1ea000 cebaa090 cf077e48 cf1e84c0
[    1.251687] 7e40: cf1ea4c0 cf131cd4 cebaa000 c0885eee cf131ca8 cebaa090 00000001 cf134810
[    1.259845] 7e60: c09b5490 c09b5490 00000000 c095fef4 c09de010 00000000 cf076000 c045fdc0
[    1.268005] 7e80: c045fd94 cf134810 00000000 c045e864 00000000 cf134810 cf134844 c09b5490
[    1.276165] 7ea0: 00000000 c045ea64 00000000 c09b5490 c045e9fc c045d108 cf06f85c cf1304b4
[    1.284326] 7ec0: c09b5490 ceba9400 c09ad364 c045e074 c09b55d0 c09b55d1 00000000 c09b5490
[    1.292486] 7ee0: c0976778 c097d458 000000e5 c045f04c c045f8e4 00000006 c0976778 c0214408
[    1.300645] 7f00: cf119280 cf119500 00000000 cf119500 c0806f00 c0a06c48 c0919058 c0302c90
[    1.308803] 7f20: c0804b47 cf119280 000000e5 cfdfc297 00000000 c0246658 00000000 c0919058
[    1.316963] 7f40: c0918208 000000e5 00000006 00000006 00000000 00000006 c0976778 c097d458
[    1.325124] 7f60: 000000e5 c09de018 c09de010 c0976784 00000000 c0949c1c 00000006 00000006
[    1.333282] 7f80: c0949504 cf076000 00000000 c020cb80 00000000 00000000 00000000 00000000
[    1.341441] 7fa0: 00000000 c020cb88 00000000 c0208d18 00000000 00000000 00000000 00000000
[    1.349600] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    1.357761] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000 fef0f078 fcf4e6f9
[    1.365937] [<c03f4ac4>] (strcmp) from [<c0579698>] (__of_find_property+0x2c/0x54)
[    1.374087] [<c0579698>] (__of_find_property) from [<c057a1f4>] (__of_device_is_compatible+0x34/0x104)
[    1.381551] [<c057a1f4>] (__of_device_is_compatible) from [<c057a380>] (of_find_compatible_node+0x48/0x74)
[    1.390842] [<c057a380>] (of_find_compatible_node) from [<c04e5c18>] (edma_axi_probe+0x520/0x96c)
[    1.400478] [<c04e5c18>] (edma_axi_probe) from [<c045fdc0>] (platform_drv_probe+0x2c/0x5c)
[    1.409416] [<c045fdc0>] (platform_drv_probe) from [<c045e864>] (driver_probe_device+0xb8/0x20c)
[    1.417576] [<c045e864>] (driver_probe_device) from [<c045ea64>] (__driver_attach+0x68/0x8c)
[    1.426516] [<c045ea64>] (__driver_attach) from [<c045d108>] (bus_for_each_dev+0x70/0x84)
[    1.434934] [<c045d108>] (bus_for_each_dev) from [<c045e074>] (bus_add_driver+0xd8/0x1c8)
[    1.443007] [<c045e074>] (bus_add_driver) from [<c045f04c>] (driver_register+0x9c/0xe0)
[    1.451172] [<c045f04c>] (driver_register) from [<c0214408>] (do_one_initcall+0x94/0x130)
[    1.458986] [<c0214408>] (do_one_initcall) from [<c0949c1c>] (kernel_init_freeable+0x108/0x1d4)
[    1.467319] [<c0949c1c>] (kernel_init_freeable) from [<c020cb88>] (kernel_init+0x8/0xfc)
[    1.475821] [<c020cb88>] (kernel_init) from [<c0208d18>] (ret_from_fork+0x14/0x3c)

Ruffio, 2016-08-14:
@diederikdehaas has your issue been resolved? If yes, then please close this issue.

diederikdehaas, 2016-08-14:
No. 
But I always suspected that the early/premature stage of DT on kernel 3.18 was the problem and I don't think anyone cares about this (anymore), so I'll close it anyway.

closed 2016-08-14 by diederikdehaas