raspberrypi/linux #920: "wilist wlan0 scan" doesn't work when wifi associated

pvouzis, 2015-04-04:
Hi,

Up until version 3.12.36+ #737 (jan 14) (https://github.com/Hexxeh/rpi-firmware/commits/master?page=2) I could have my WiFi associated and run a "iwlist wlan0 scan" and get the list of detected access points.

However, after that version when the WiFi is connected and I run the command "sudo iwlist wlan0 scan" I get

"wlan0 Interface doesn't support scanning : Device or resource busy"

Not sure what is the problem, but I suspect something changed with the WiFi drivers.

My WiFi dongle is an ASUS N53.

I found this https://wikidevi.com/wiki/ASUS_USB-N53 which says the drivers used is the rt2800usb. From what I see the drivers are installed in my RPi. And, again, it baffles me because because it used to work just fine, but it looks like after version 3.12.36+ #737 (jan 14) for the specific card I have, scanning doesn't work when the WiFi is associated. All I did was to update my RPi to the latest release by using apt-get.

The scanning command works just fine when I use other WiFi dongles (that use other drivers). It looks like the problem is with the particular driver.

pi@172.30.0.98 :~:$ lsusb
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp.
Bus 001 Device 004: ID 0bda:8176 Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter
Bus 001 Device 005: ID 04ca:004b Lite-On Technology Corp.

MrEngman, 2015-04-04:
The output from command `lsusb` you show is confusing because it is not for the Asus N53, is it?

pvouzis, 2015-04-04:
You are right MrEngman. I corrected it. I didn't know I would find you here as well! I initially posted this issue  at https://github.com/Hexxeh/rpi-firmware/ and I was told that this is the best place to put it.

Well, if you have an update please let me know.

MrEngman, 2015-04-07:
Just got myself an Asus N53 and connected to my Pi B2 it came up straight away, however there is an oddity with the scan command. If I use command `iwlist wlan0 scan` it displays scan results. If I use command `sudo iwlist wlan0 scan` I get an error `wlan0     Failed to read scan data : Resource temporarily unavailable`. But I'm fairly certain it does the scan because if I use the command `iwlist wlan0 scan` again it displays scan results but they are different to the previous display using that command. Then repeating that command and the results never change.

Using the command `iwlist wlan0 scan` it displays results stored somewhere on the SD card, don't know where, when it last did a scan. If I repeat the command the results stay the same. If I then use the command `sudo iwlist wlan0 scan` it returns the error I said above but if I then use the command `iwlist wlan0 scan` it displays the results and at least for me they are different to the previous results that were displayed. Repeating `iwlist wlan0 scan` and the scan results are listed and they do not change no matter how many times I repeat the command. Running `sudo iwlist wlan0 scan` again it returns the error but then using the command `iwlist wlan0 scan` again the scan results are displayed but are now different to the previous displays. But again repeating the command and the results do not change until I run the command again with `sudo`

For me it seems `sudo iwlist wlan0 scan` does a scan but for some reason will not display them but they are stored somewhere on the SD card. The command `iwlist wlan0 scan` then displays the stored results and repeating the command will always return the same results. Running the command with sudo will then scan again and store the data but give the error. Then using `iwlist wlan0 scan` again will now show new data from the scan using `sudo`.

Why using `sudo iwlist wlan0 scan` returns an error I do not know. Possibly a problem with the driver, I'm not sure. If I use a different wifi adapter then the command `sudo iwlist wlan0 scan` works OK and displays the scan results.

pvouzis, 2015-04-07:
I see what you are saying. I assume that the behavior your described is while the WiFi is associated. 

In my case, when the WiFi is not associated "sudo iwlist wlan0 scan" works fine and doesn't return an error.

It would be interesting to try this on an RPi with kernel 3.12.36+ #737 (I guess that cannot be an RPi 2). In my case "sudo iwlist wlan0 scan" returns results just fine when it's associated.

I don't know if there is a way to run kernel RPi 2 with 3.12.36+ #737. Can this be done?

Thanks for looking into this.

MrEngman, 2015-04-07:
You won't be able to run 3.12.36+ on a Pi B2.

Yes, The wifi was associated when I tried the scans. I'm wondering whether it is some kind of timing problem because I have a wifi module, Edimax EW-7811UTC, that works fine on the older Pi's but on a Pi B2 it causes the kernel to crash. So some issue with the Pi B2 maybe.

pvouzis, 2015-04-15:
I found this post about a change in the driver and I am wondering whether it's applicable in this case:

http://stackoverflow.com/questions/18277465/why-does-iwlist-scan-return-cached-results-almost-every-time

pvouzis, 2015-04-15:
MrEngman it looks like it's a timing issue. Here is what I did. 

I downloaded the source code of wireless tools v30 from here: https://code.google.com/p/rtl8192cu/source/browse/wireless_tools/wireless_tools.30.rtl.tar.gz?r=bdd3a2265bdd6a92f24cef3d52fa594b2844c9c1&spec=svnd65922d040ce25481b2a12740eec74fdbb17e76e

If you extract it you will find the file iwlist.c under wireless_tools/wireles_tools.30.rtl/. Edit it and change 

int timeout = 15000000;

to

int timeout = 30000000;

Compile with make and try to run scan while the wireless interface is associated. You will see the results this time.

Ruffio, 2016-08-14:
@pvouzis has your issue been resolved? If yes, then please close this issue.

closed 2017-05-17 by JamesH65