raspberrypi/linux #921: bcm2708-dmaengine and DMA channel types

notro, 2015-04-06:
I'm looking further into DMA channels in the Device Tree world.
In rpi-3.19.y, bcm2708-dmaengine has 5 LITE channels available to hand out.
However if it is built with CONFIG_ARCH_BCM2835, it will have the same channels as arch/arm/mach-bcm2708/dma.c.
But AFAICT it will not discriminate between the channel types. It will just hand out any channel sequentially.

This is probably not what we want.
bcm2708_fb for instance asks specifically for BCM_DMA_FEATURE_BULK.
A spi dma controller driver I use, asks for BCM_DMA_FEATURE_FAST.

So we need a way to specify channel type, right?

Here's some relevant parts from the dma drivers:

arch/arm/mach-bcm2708/dma.c: vc_dmaman_init():
```c
	dmaman->has_feature[BCM_DMA_FEATURE_FAST_ORD] = 0x0c;  /* chans 2 & 3 */
	dmaman->has_feature[BCM_DMA_FEATURE_BULK_ORD] = 0x01;  /* chan 0 */
	dmaman->has_feature[BCM_DMA_FEATURE_NORMAL_ORD] = 0xfe;  /* chans 1 to 7 */
	dmaman->has_feature[BCM_DMA_FEATURE_LITE_ORD] = 0x7f00;  /* chans 8 to 14 */
```
Are channels 2 and 3 special since they are both FAST and NORMAL? (0x0c & 0xfe == 0x0c)

drivers/dma/bcm2708-dmaengine.c
bcm2835_dma_probe():
```c
#ifndef CONFIG_ARCH_BCM2835
	for (i = 0; i < 5; i++) {
		chan_id = bcm_dma_chan_alloc(BCM_DMA_FEATURE_LITE, &chan_base, &irq);
		if (chan_id < 0)
			break;

		rc = bcm2708_dma_chan_init(od, chan_base, chan_id, irq);
	}
#else
// resources from DT are bank 1 interrupts 16-28, the same as arch/arm/mach-bcm2708/dma.c uses
	for (i = 0; i < pdev->num_resources; i++) {
		irq = platform_get_irq(pdev, i);
		if (irq < 0)
			break;

		if (chans_available & (1 << i)) {
			rc = bcm2835_dma_chan_init(od, i, irq);
			if (rc)
				goto err_no_dma;
		}
	}
#endif
```
This function is called when parsing DT property: dmas = <&dma 11>;
```c
static struct dma_chan *bcm2835_dma_xlate(struct of_phandle_args *spec,
					   struct of_dma *ofdma)
{
	struct bcm2835_dmadev *d = ofdma->of_dma_data;
	struct dma_chan *chan;

	chan = dma_get_any_slave_channel(&d->ddev);
	if (!chan)
		return NULL;

	/* Set DREQ from param */
	to_bcm2835_dma_chan(chan)->dreq = spec->args[0];

	return chan;
}
```

popcornmix, 2015-04-06:
I believe DMA channels 2&3 are identical to other normal dma channels, but they have their interrupts routed to arm (rather than as generic GPU interrupts):
https://github.com/raspberrypi/linux/blob/rpi-3.18.y/arch/arm/mach-bcm2709/include/mach/platform.h#L172

This means you can identify the interrupt source with few register reads, and as such they take precedence (when two interrupt are both pending). I'm not certain this provides any real benefit.

notro, 2015-04-06:
Does DREQ specify a specific channel or can a DREQ be used with any channel?

notro, 2015-04-06:
I can answer my own question, found this is the datasheet:

> The allocation of peripherals to DMA channels is programmable. 

This is the current situation:
```
		dma: dma@7e007000 {
			compatible = "brcm,bcm2835-dma";
			#dma-cells = <1>;
		};

		mmc: mmc@7e300000 {
			compatible = "brcm,bcm2835-mmc";
			reg = <0x7e300000 0x100>;
			interrupts = <2 30>;
			clocks = <&clk_mmc>;
			dmas = <&dma 11>,
			       <&dma 11>;
			dma-names = "tx", "rx";
			status = "disabled";
		};
```

Maybe we should change #dma-cells to 2 and use the second argument for channel type?
```
		dma: dma@7e007000 {
			compatible = "brcm,bcm2835-dma";
			#dma-cells = <2>;
		};

		mmc: mmc@7e300000 {
			compatible = "brcm,bcm2835-mmc";
			reg = <0x7e300000 0x100>;
			interrupts = <2 30>;
			clocks = <&clk_mmc>;
			dmas = <&dma 11 DMA_LITE>,
			       <&dma 11 DMA_LITE>;
			dma-names = "tx", "rx";
			status = "disabled";
		};
```

@popcornmix
Do you agree that we need a way to specify the channel type in the Device Tree?

notro, 2015-04-10:
@P33M  Can you comment on this?

It might not be a problem for bcm2708-dmaengine since it will be replaced by bcm2835-dma, but it will probably be for bcm2835-dma.
bcm2835-dma hands out channels sequentially from NORMAL down to LITE channels. There is no way to control who get the fast ones.

BCM2835 ARM Peripherals
4.5 DMA LITE Engines
```
The Lite engine will have about half the bandwidth of a normal DMA engine, and are intended for low bandwith peripheral servicing.
```
And which devices would benefit from using a normal channel instead of a lite channel?

Some other questions I have:

**Do ARM only have access to 2 normal channels?**

Device Tree
```
brcm,dma-channel-mask = <0x7f35>;
```
drivers/dma/bcm2835-dma.c
```
#define BCM2835_DMA_BULK_MASK   BIT(0)
#define BCM2835_DMA_FIQ_MASK    (BIT(2) | BIT(3))

of_property_read_u32(pdev->dev.of_node, "brcm,dma-channel-mask", &chans_available))

chans_available &= ~(BCM2835_DMA_FIQ_MASK | BCM2835_DMA_BULK_MASK);

```
This leaves only 2 normal channels:
```
0x7f35 & ~((1<<2) | (1<<3) | (1<<0)) = 0b111 1111 0011 0000
```
Currently the mmc driver will pick the normal channels (when using bcm2835-dma):
```
[    2.442686] bcm2835-dma 20007000.dma: Allocating DMA channel 4
[    2.448632] bcm2835-dma 20007000.dma: Allocating DMA channel 5
[    2.454644] mmc-bcm2835 20300000.mmc: DMA channels allocated
```

**Which are the LITE channels?**

AFAICT from the datasheet, channels 0-6 are normal and 7-14 are LITE.
This does not match arch/arm/mach-bcm2708/dma.c and drivers/dma/bcm2708-dmaengine.c which treats channels 8-> as LITE.

BCM2835 ARM Peripherals
4.2.1.2 Register Map
```
0_DEBUG 1_DEBUG 2_DEBUG 3_DEBUG 4_DEBUG 5_DEBUG 6_DEBUG Register
Synopsis DMA Debug register.

28	LITE	DMA Lite					RO	0x0
		Set if the DMA is a reduced performance LITE
		engine.

7_DEBUG 8_DEBUG 9_DEBUG 10_DEBUG 11_DEBUG 12_DEBUG 13_DEBUG 14_DEBUG Register
Synopsis DMA Lite Debug register.

28	LITE	DMA Lite					RO	0x1
		Set if the DMA is a reduced performance LITE
		engine.
```
[drivers/dma/bcm2708-dmaengine.c](https://github.com/raspberrypi/linux/blob/rpi-3.19.y/drivers/dma/bcm2708-dmaengine.c#L428)
```c
	if (c->ch >= 8) /* we have a LITE channel */
		max_size = MAX_LITE_TRANSFER;
	else
		max_size = MAX_NORMAL_TRANSFER;
```
[arch/arm/mach-bcm2708/dma.c](https://github.com/raspberrypi/linux/blob/rpi-3.18.y/arch/arm/mach-bcm2708/dma.c#L158)
```c
	dmaman->has_feature[BCM_DMA_FEATURE_BULK_ORD] = 0x01;  /* chan 0 */
	dmaman->has_feature[BCM_DMA_FEATURE_NORMAL_ORD] = 0xfe;  /* chans 1 to 7 */
	dmaman->has_feature[BCM_DMA_FEATURE_LITE_ORD] = 0x7f00;  /* chans 8 to 14 */
```

**And why the special treatment of channel 0?**
Can't find anything in the datasheet about this. I can see that bcm2708_fb asks for this channel.

**What's the LITE channel max transfer size, 32k or 64k?**

BCM2835 ARM Peripherals
4.5 DMA LITE Engines
```
3. The DMA length register is now 16 bits, limiting the maximum transferrable length to 65536 bytes.
```
[drivers/dma/bcm2708-dmaengine.c](https://github.com/raspberrypi/linux/blob/rpi-3.19.y/drivers/dma/bcm2708-dmaengine.c#L147)
```c
#define MAX_LITE_TRANSFER 32768
```

**Number of channels available**

There are 13 elements in bcm_dma_irqs. There are 15 bits in chans_available (0x7f35 & 0x7f00 = 0b111 1111 0000 0000).
Couldn't we end up outside the array here?

arch/arm/mach-bcm2708/dma.c
```c
// module parameter is set to this value
dmachans = 0x7f35

static void vc_dmaman_init(struct vc_dmaman *dmaman, void __iomem *dma_base,
			   u32 chans_available)
{
	dmaman->chan_available = chans_available; // which is the value of dmachans
	dmaman->has_feature[BCM_DMA_FEATURE_LITE_ORD] = 0x7f00;  /* chans 8 to 14 */
}

static unsigned char bcm_dma_irqs[] = {
	IRQ_DMA0,
[...]
	IRQ_DMA12
};

extern int bcm_dma_chan_alloc(unsigned preferred_feature_set,
			      void __iomem **out_dma_base, int *out_dma_irq)
{
		rc = vc_dmaman_chan_alloc(dmaman, preferred_feature_set);
			*out_dma_irq = bcm_dma_irqs[rc];
}

static int vc_dmaman_chan_alloc(struct vc_dmaman *dmaman,
				unsigned preferred_feature_set)
{
	chans = dmaman->chan_available;
	for (feature = 0; feature < BCM_DMA_FEATURE_COUNT; feature++)
		/* select the subset of available channels with the desired
		   feature so long as some of the candidate channels have that
		   feature */
		if ((preferred_feature_set & (1 << feature)) &&
		    (chans & dmaman->has_feature[feature]))
			chans &= dmaman->has_feature[feature];

	if (chans) {
		int chan = 0;
		/* return the ordinal of the first channel in the bitmap */
		while (chans != 0 && (chans & 1) == 0) {
			chans >>= 1;
			chan++;
		}
		/* claim the channel */
		dmaman->chan_available &= ~(1 << chan);
		return chan;
}
```

P33M, 2015-04-10:
Channel 0 is a "fat" channel. It has all the properties and features of the normal channels 1-7, but has oversized FIFOs which allow for longer burst lengths to/from SDRAM.

I'd have to check the regmap to see which channels are "lite" or not - I believe there's 8 normal and 8 lite (so Lite starting at channel 8).

Lite channels have a maximum transfer length of 2^16-1 bytes as the transfer length register field width is 16 bits.

Channel 15 (with a different base address) is another special channel - it's a Normal channel that's closer in the bus matrix to the VPU which makes it more suitable for low latency, fast transfers driven from the VPU.

In terms of bandwidth, there are few peripherals that care about the difference in throughput between channels. Normal channels could probably get close to saturating SDRAM bandwidth. Lite channels can certainly satisfy e.g. SDHCI at 20MB/s+. A rough guesstimate would put the max throughput of a Lite channel at about 400MB/s, based on what happens when you turn the number of wait cycles up to maximum.

popcornmix, 2015-04-10:
Some DMA info I wrote in a recent email:

DMA channel 0 is provided with an 8 entry read FIFO which allows it to perform efficient SDRAM burst reads. This increases its performance when writing out to high bandwidth peripherals.
DMA channels 1-7 are normal DMA engines.
DMA channels 8-14 are reduced capability DMA-LITE engines. These have reduced internal storage and cannot perform 2D transfers. They are intended for low bandwith peripherals.
DMA Channel 15 also has an 8 entry read FIFO, but is located nearer to the L2 Cache and the SDRAM, and so is better suited to bulk memory transfers. It has no access to the peripheral bus.

DMA LITE Engines

Several of the DMA engines are of the LITE design. This is a reduced specification engine designed to save space. The engine behaves in the same way as a normal DMA engine except for the following differences.
1. The internal data structure is 128 bits instead of 256 bits. This means that if you do a 128 bit wide read burst of more than 1 beat, the DMA input register will be full and the read bus will be stalled.
The normal DMA engine can accept a read burst of 2 without stalling. If you do a narrow 32 bit read burst from the peripherals then the lite engine can cope with a burst of 4 as opposed to a burst of 8
for the normal engine. Note that stalling the read bus will potentially reduce the overall system performance, and may possible cause a system lockup if you end up with a conflict where the DMA
cannot free the read bus as the read stall has prevented it writing out its data due to some circular system relationship.

2. The Lite engine does not support 2D transfers. The TDMODE, S_STRIDE, D_STRIDE and YLENGTH registers will all be removed. Setting these registers will have no effect.
3. The DMA length register is now 16 bits, limiting the maximum transferrable length to 65536 bytes.
4. Source ignore (SRC_IGNORE) and destination ignore (DEST_IGNORE) modes are removed.
The Lite engine will have about half the bandwidth of a normal DMA engine, and are intended for low bandwith peripheral servicing.

notro, 2015-04-13:
OK, thanks.
I guess channel type is of concern when doing mem to mem copy. Only the framebuffer driver does this, but it will go away eventually.

closed 2015-04-13 by notro