raspberrypi/linux #924: module rp_usbdisplay

peepo, 2015-04-08:
please could you include module for RoboPeak usb display?

it seems the 'notro' drivers are already in the official linux sources

https://github.com/robopeak/rpusbdisp.git

I regularly use rpi-update, but cannot fathom another simple way to ensure that this display continues to work.

this effects all RPi models including 2

popcornmix, 2015-04-08:
In general we don't include out of tree drivers in our kernel tree as they introduce a maintenance headache and the code hasn't been reviewed.
Notro's usb display drivers were different as they had been accepted upstream and we were just pulling them in early.
If the driver is accepted upstream, then we may accept it with a suitable PR. If not, then it needs to be submitted upstream first.
http://kernelnewbies.org/UpstreamMerge

peepo, 2015-04-08:
sorry, I'm not clear, is it that the Notro usb display drivers should be sufficient?
I did not find a way to test this.
or are they only necessary?
ie one has to also load module rp_usbdisplay

I really only need a usb framebuffer /dev/fb1 at this stage

popcornmix, 2015-04-08:
As far as I can see this is not supported in upstream kernel (hence the kernel out-of-tree kernel driver in github). Looks like there is a custom image here:
http://www.raspberrypi.org/forums/viewtopic.php?f=45&t=102623

peepo, 2015-04-08:
okay, tx I'm following that and a pimoroni link, both are for Pi2, arrives tomorrow.
I also need on Pi1 not clear to me it will work, but hacking...

closed 2015-04-08 by peepo