raspberrypi/linux #925: linux kernal 3.19.y make error "fs/btrfs/super.c  no such file or directory"

Ashu1504, 2015-04-08:
I am trying to install kernal 3.19.y on my Raspberry pi 2

Downloaded the file form here  https://github.com/raspberrypi/linux/tree/rpi-3.19.y

I am currently running kernal 3.18.7 -v7+

when I run the "make" command i executes properly until this

    make[1]: 'include/generated/mach-types.h' is up to date.
    CALL    scripts/checksyscalls.s
    CHK     include/generated/compile.h
    CHK     kernel/config_data.h
    CC [M]  fs/btrfs/super.o

    fs/btrfs/super.c:45:27: fatal error: delayed-inode.h: No such file or directory
    compilation terminated
    scripts/Makefile.build:257: recipe for target'fs/btrfs/super.o' failed
    make[2]: *** [fs/btrfs/super.o] Error 1
    scripts/Makefile.build:402: recipe for target 'fs/btrfs' failed
    make[1]: *** [fs/btrfs] Error 2
    Makefile:939: recipe for target 'fs' failed
    make: *** [fs] Error 2

can anyone help me to get this thing working?

popcornmix, 2015-04-08:
Are you using the recommended bcmrpi_defconfig/bcm2709_defconfig config files?

Ashu1504, 2015-04-09:
I followed the steps fro building kernel from the official raspberry pi wesite
And got everything working but after rebooting the system when i try uname -a i get
    Linux raspberrypi 3.18.11-v7+

Why is that? I was installing 3.19

popcornmix, 2015-04-09:
Are you on Pi2? Did you update kernel7.img?

popcornmix, 2015-04-09:
Duplicate of: https://github.com/raspberrypi/linux/issues/927

closed 2015-04-09 by popcornmix

Ashu1504, 2015-04-09:
yes am on pi2

I used these commands for kernel7.img

    rm -f /boot/kernel7.img.old
    mv /boot/kernel7.img /boot/kernel7.img.old 
    cp arch/arm/boot/Image /boot/kernel7.img