raspberrypi/linux #926: DPMS signal to turn off monitor results in FBIOBLANK: Operation not permitted

doozan, 2015-04-09:
This may be a dup of #487 but that was closed a year ago and that patch looks like it's still in the latest kernel.  I'm seeing this on a RPi 2 with kernel 3.18.11-v7+

Under Xorg, running "xset dpms force off" results in the monitor going dark but never powering off.  Xorg.0.log shows:

(EE) FBDEV(0): FBIOBLANK: Operation not permitted

Running "tvservice -o" will properly turn off the monitor, FWIW.

asgh, 2015-04-24:
I can confirm this. I have the exact same problem.

hunternet93, 2015-05-22:
Same issue here running `3.18.0-trunk-rpi2`. Looks like the function `bcm2708_fb_blank` changed significantly after the commit fixing #487, maybe a bug was introduced somehow.

Ruffio, 2016-08-15:
@doozan has your issue been resolved? If so, please close this issue. Thanks.

JamesH65, 2017-05-17:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.

closed 2017-05-17 by JamesH65

popcornmix, 2017-05-17:
For info. Setting ``hdmi_blanking=1`` in config.txt will remove hdmi output on a console blank.
It is not a default option as it is unwanted for some use cases (e.g. omxplayer).