raspberrypi/linux #927: Unable to install kernel 3.19.0 properly

Ashu1504, 2015-04-09:
I am having raspberry pi 2
MY previous kernel version was 3.18.7-v7+
and I downloaded 3.19.y and complied it following the steps form here http://rglinuxtech.com/?p=1324
after rebooting the system when i type uname -a I get 
    Linux raspberrypi 3.18.11-v7+ #1 SMP PREEMPT Thu Apr 9 06:32:41 MVT 2015 armv7l GNU/Linux

Why am I not getting 3.19.0 ?

popcornmix, 2015-04-09:
what does
```md5sum arch/arm/boot/Image /boot/kernel7.img```
show you? I suspect you haven't copied the kernel correctly.

Note, the linked instructions need "sudo" added to the commands that copy files into system locations (like /boot).

Ashu1504, 2015-04-09:
yeah u were right, I haven't copied the kernel correctly as the above commands gives me the output as

    md5sum: arch/arm/boot/Image: No such file or directory
    593c0e0cc6cab1a3fe8da6dc90081bef  /boot/kernel7.img

So what should I do? will I have to start again from the beginning or their is a way around?

Ashu1504, 2015-04-09:
and yes i added sudo to the linked instructions

popcornmix, 2015-04-09:
I expect you had some error in one of the commands entered and so it hasn't produced a kernel.
You need to check the output of each command you typed in and pay attention if it failed.
The ```cp arch/arm/boot/Image /boot/kernel7.img``` command for example would have reported ```No such file or directory``` 
But this github issue tracker is for reporting kernel bugs, not for advice on how to build. Every post here will spam a large number of developers who follow this repo.
I'd suggest you open a post on the forum asking for advice on building the kernel.

closed 2015-04-09 by popcornmix

pelwell, 2015-04-09:
@popcornmix is right - use the forum. But here are some *final* suggestions:

What do the final lines of the kernel build output show?

On a side note, you would be better off changing the line that just says "make" to "make zImage", and then copying zImage to kernel7.img, because zImage is a compressed kernel that will load quicker.

Also, if you want to use Device Tree on the new kernel (and you should), just copying the zImage isn't enough. You need to get the mkknlimg utility from the raspberrypi/tools repo and replace "sudo cp arch/arm/boot/zImage /boot/kernel7.img" with "sudo mkknlimg arch/arm/boot/zImage /boot/kernel7.img".

Ashu1504, 2015-04-09:
I used "make zImage" this time which gave the final output lines as 
    GZIP    kernel/config_data.gz
    CHK     kernel/config_data.h
    Kernel: arch/arm/boot/Image is ready
    Kernel: arch/arm/boot/zImage is ready

which i guess where the same when I used  "make"

Thanks a lot for the help and I'll definitely  try what you said
Just for the reference if u can just verify if the steps in the link are correct or not
Should i follow that only (using sudo wherever required)+ your advice ?