raspberrypi/linux #929: New kernel module and overly for sound card

jgrulich, 2015-04-14:
Is there some central repo where to find the source files and step by step tutorial how to customize the kernel module and overlay for new sound card?

pelwell, 2015-04-14:
The source for the existing overlays is in this tree - go to arch/arm/boot/dts and look for *-overlay.dts .

There is no "step by step tutorial", but there is documentation of Device Tree and its usage on Raspberry Pi [here](https://www.raspberrypi.org/documentation/configuration/device-tree.md). You can get support in the forums - [here](https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=99784&sid=11ce65be15cc5b303b10cbb30c05f3e1)'s a reasonable starting point.

jgrulich, 2015-04-15:
I've already looked around such a pages as a first step. Hence I've not found any tutorial for beginner, I've tried to ask for some support. First of all, I need the kernel driver and later on the device tree overlay. I guess that only overlay is not sufficient. Is this correct?
Seems that I need to find someone who is skilled in kernell programming, but don't know where to look.

pelwell, 2015-04-15:
So, you will need one kernel module for the DAC, one for the sound card as a whole, and an overlay. I had some success getting a basic configuration working with just DAC module + overlay, using a DAC module that exposed some ALSA parameters, but in the end I ran into some restrictions.

A few questions:
1. Is there already a driver for the DAC/codec in the tree?
2. Is there another RPi sound card already using that DAC?
3. Aside from the DAC, is there any other component that requires programming? Is that boot-time initialisation or is there a dynamic element (crystal switching, for example)?

If your answers are Yes, Yes and No then it should be a trivial cut-and-paste job to construct a driver for your board. Answering Yes to 3 makes it marginally harder, but not much.

jgrulich, 2015-04-15:
Yes, I already have working card with the driver and overlay which is already in the kernel.
1. Yes, there is driver and overlay hifiberry-dacplus.
2. Yes, hifiberry-dac+ sound card
3. Additionally there is headphone amplifier on my board which enables I2C control - volume and enable. The hifiberry driver need to be only slightly modified and added one more slider for volume control and one ON/OFF switch.

jgrulich, 2015-04-16:
As a conclusion of before mentioned, there is already codec for DAC and  I need soc-bcm module and corresponding overlay, both tailored to my card.
I need some help with this, because I'm not skilled in the kernel programing.

pelwell, 2015-04-16:
It is something I could do, and if nobody else comes forward I could probably help you through it.

However, I have pinged somebody I know who frequents these parts and who has the skills. Should they want to get in touch, how should they reach you?

jgrulich, 2015-04-16:
Thank you for help. How to send you personal message to provide more detailed info?
Or you may join my web: grulich.eu where I have some of my designs and contact e-mail.

pelwell, 2015-04-16:
My name is Phil, I work for Raspberry Pi, and I have the logical email address.

jgrulich, 2015-04-16:
Hi Phil, please contact me directly on my e-mail: jan@grulich.eu.

closed 2015-04-23 by jgrulich