raspberrypi/linux #932: watchdog: Report reason for last reset

julianscheel, 2015-04-17:
Would it be possible to properly report if a watchdog timeout caused the last reset or a normal powercycle? Or maybe even detect resets due to voltage dropout, etc?

I had hoped the PM_RSTS register described in the watchdog section of `bcm2708/include/mach/platform.h` would be helpful. Reading return `0x0` only for me, though. And I couldn't find any documentation on that register.

Maybe someone can provide details?

popcornmix, 2015-04-17:
PM_RSTS is reset when read. But we do cache the original version away on GPU side.

In one of my scripts I so this:
```
 # check if we rebooted through good or bad path
 RSTS=`vcgencmd get_rsts`
 RSTS=${RSTS: -3}
 # 1000 is power on reset, so treat an run again
 if [ "$RSTS" == "1000" ]; then
  POWERON=1
 elif [ "$RSTS" == "574" ]; then
  PASS=1
 else
  PASS=0
 fi
```

julianscheel, 2015-04-17:
Ah nice, then one more question: Is there some documentation on the possible RSTS values?
Some are understandable from your snippet and the platform.h (like HADPOR), but not all...

nbrachet, 2015-04-17:
Wouldn't it be better to use:
```
RSTS=`vcgencmd get_rsts`
RSTS=${RSTS#*=}
```
```${RSTS: -3}``` will only select the last 3 charaters won't it? And so RSTS wouldn't be equal to 1000 but to 000.

popcornmix, 2015-04-17:
PM_RSTC
```
31-24 PASSWD Power Manager password “5a” W
23-13 - Unused R
21-20 HRCFG Hard reset configuration R/W
Not used on BCM2708A0.
19-18 - Unused R
17-16 FRCFG Full reset configuration R/W
Not used on BCM2708A0.
15-14 - Unused R
13-12 QRCFG Quick reset configuration R/W
0 = do not reset PLLs
1 = reset PLLs
2 = do not reset PLLs
3 = reset PLLs
11-10 - Unused R
9-8 SRCFG Software reset configuration R/W
0 = no reset
1 = quick reset
2 = full reset
3 = hard reset
7-6 - Unused R
5-4 WRCFG Watchdog reset configuration R/W
0 = no reset
1 = quick reset
2 = full reset
3 = hard reset
3-2 - Unused R
1-0 DRCFG Debugger reset configuration R/W
0 = no reset
1 = quick reset
2 = full reset
3 = hard reset
```

PM_RSTS
```
31-24 PASSWD Power Manager password “5a” W
23-13 - Unused R
12 HADPOR Had a power-on reset R/W
Clear this flag by writing 0 to this field.
11 - Unused R
10 HADSRH Had a software hard reset R/W
Clear this flag by writing 0 to this field.
9 HADSRF Had a software full reset R/W
Clear this flag by writing 0 to this field.
8 HADSRQ Had a software quick reset R/W
Clear this flag by writing 0 to this field.
7 - Unused R
6 HADWRH Had a watchdog hard reset R/W
Clear this flag by writing 0 to this field.
5 HADWRF Had a watchdog full reset R/W
Clear this flag by writing 0 to this field.
4 HADWRQ Had a watchdog quick reset R/W
Clear this flag by writing 0 to this field.
3 - Unused R
2 HADDRH Had a debugger hard reset R/W
Clear this flag by writing 0 to this field.
1 HADDRF Had a debugger full reset R/W
Clear this flag by writing 0 to this field.
0 HADDRQ Had a debugger quick reset R/W
Clear this flag by writing 0 to this field.
```

Note: NOOBS uses these bits for indicating the partition the next boot should occur from, so beware - you could get unexpected behaviour if you set some of these.

julianscheel, 2015-04-17:
Thanks a lot!

closed 2015-04-17 by julianscheel

popcornmix, 2015-04-17:
@nbrachet thanks, that looks right.

Rosentorp, 2019-03-21:
Hi all!
Anyone that knows the reason for a Raspberry Pi restart with result 1020? Eg. both 
HADWRF Had a watchdog full reset
and
12 HADPOR Had a power-on reset
?

pelwell, 2022-06-10:
It's actually available from Device Tree, as `/chosen/pm_rsts`.
The documentation has just been updated - see https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/configuration/device-tree.adoc#firmware-parameters. I'm not sure why it isn't live yet.

pelwell, 2022-06-10:
It's live now : https://www.raspberrypi.com/documentation/computers/configuration.html#part4

LonerDan, 2022-06-10:
At least what version of firmware I need to have those parameters supplied to me? Because I am not seeing them in my dtb.