raspberrypi/linux #934: rpi-4.0.y: stuck on boot

msperl, 2015-04-17:
booting a freshly built 4.0.y on a rpi2 results in the following:
```
...
[    2.396735] Initializing XFRM netlink socket
[    2.405533] NET: Registered protocol family 17
[    2.414623] Key type dns_resolver registered
[    2.423941] Registering SWP/SWPB emulation handler
[    2.434178] registered taskstats version 1
[    2.443049] vc-sm: Videocore shared memory driver
[    2.452243] [vc_sm_connected_init]: start
[    2.461484] [vc_sm_connected_init]: end - returning 0
[    2.467620] Indeed it is in host mode hprt0 = 00021501
[    2.482252] Waiting for root device /dev/mmcblk0p2...
[    2.647513] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    2.658797] Indeed it is in host mode hprt0 = 00001101
[    2.867824] usb 1-1: New USB device found, idVendor=0424, idProduct=9514
[    2.879188] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.891829] hub 1-1:1.0: USB hub found
[    2.900307] hub 1-1:1.0: 5 ports detected
[    3.187561] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    3.297806] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00
[    3.309343] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumbe0
[    3.324075] smsc95xx v1.0.4
[    3.391677] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-bcm2708_usb-7
[    3.487566] usb 1-1.4: new high-speed USB device number 4 using dwc_otg
[    3.598256] usb 1-1.4: New USB device found, idVendor=0bda, idProduct=8179
[    3.609968] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumbe3
[    3.622023] usb 1-1.4: Product: 802.11n NIC
[    3.630851] usb 1-1.4: Manufacturer: Realtek
[    3.639757] usb 1-1.4: SerialNumber: 00E04C0001
[    3.887554] usb 1-1.5: new low-speed USB device number 5 using dwc_otg
[    4.016791] usb 1-1.5: New USB device found, idVendor=0c40, idProduct=8000
[    4.028391] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumbe0
[    4.040429] usb 1-1.5: Product: 2.4GHz receiver
[    4.049661] usb 1-1.5: Manufacturer: ELMCU
[    4.073357] input: ELMCU 2.4GHz receiver as /devices/platform/bcm2708_usb/us0
[    4.147785] hid-generic 0003:0C40:8000.0001: input,hidraw0: USB HID v1.11 Ke0
[    4.184441] input: ELMCU 2.4GHz receiver as /devices/platform/bcm2708_usb/us1
[    4.257924] hid-generic 0003:0C40:8000.0002: input,hidraw1: USB HID v1.11 Mo1
[   13.510731] random: nonblocking pool is initialized
```

There is no means to recover from this besides pulling out the sd-card...

msperl, 2015-04-17:
kernel is built via: ```make bcm2709_defconfig clean;make```

popcornmix, 2015-04-17:
I'm currently running rpi-4.0.y built with bcm2709_defconfig (compiled today).
I assume you have updated bcm2709-rpi-2-b.dtb?
Can you try with any USB (or other external hardware) removed?
Can you try with default config.txt/cmdline.txt?

msperl, 2015-04-17:
I will try - but only after I have confirmed that it works with a rebuilt 3.18.y.

Is there a change with the device-tree? Because it should be backwards-compatible with any earlier versions...

popcornmix, 2015-04-17:
https://github.com/raspberrypi/linux/pull/909 breaks compatibility (3.19 and 4.0 have this, 3.18 doesn't). It stops mmc from working if you mismatched kernel and DT blob.

msperl, 2015-04-17:
I guess I can only try it tomorrow morning as I have to rebuild (again)

msperl, 2015-04-18:
OK - compilation ran thru the night and with the new DTB it works...

closed 2015-04-18 by msperl