raspberrypi/linux #937: fiq_enable=0 fails with usb keyboard/mouse on both Pi1 and Pi2

notro, 2015-04-21:
If I set ```dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0``` and have a keyboard or mouse connected, it fails to boot. Without keyboard or mouse it boots fine.
Using latest rpi-update kernel.

Pi2 kernel messages with fiq_enable=1 and usb keyboard and mouse connected, which works fine:
```
[    0.000000] Linux version 3.18.11-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #780 SMP PREEMPT Mon Apr 20 14:59:10 BST 2015
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B
...
[    3.412545] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-bcm2708_usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:6e:52:28
[    3.507998] usb 1-1.2: new low-speed USB device number 4 using dwc_otg
[    3.642349] usb 1-1.2: New USB device found, idVendor=046d, idProduct=c313
[    3.653490] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.653502] usb 1-1.2: Product: USB Multimedia Keyboard
[    3.653519] usb 1-1.2: Manufacturer: LITEON Technology
[    3.680167] udevd[175]: starting version 175
[    3.691277] input: LITEON Technology USB Multimedia Keyboard as /devices/platform/bcm2708_usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:046D:C313.0001/input/input0
[    3.709933] hid-generic 0003:046D:C313.0001: input,hidraw0: USB HID v1.10 Keyboard [LITEON Technology USB Multimedia Keyboard] on usb-bcm2708_usb-1.2/input0
[    3.739751] input: LITEON Technology USB Multimedia Keyboard as /devices/platform/bcm2708_usb/usb1/1-1/1-1.2/1-1.2:1.1/0003:046D:C313.0002/input/input1
[    3.758210] hid-generic 0003:046D:C313.0002: input,hidraw1: USB HID v1.10 Device [LITEON Technology USB Multimedia Keyboard] on usb-bcm2708_usb-1.2/input1
[    3.867971] usb 1-1.3: new low-speed USB device number 5 using dwc_otg
[    3.999340] usb 1-1.3: New USB device found, idVendor=045e, idProduct=0084
[    4.011364] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    4.023444] usb 1-1.3: Product: Basic Optical Mouse
[    4.031375] usb 1-1.3: Manufacturer: Microsoft
[    4.051708] input: Microsoft Basic Optical Mouse as /devices/platform/bcm2708_usb/usb1/1-1/1-1.3/1-1.3:1.0/0003:045E:0084.0003/input/input2
[    4.072647] hid-generic 0003:045E:0084.0003: input,hidraw2: USB HID v1.10 Mouse [Microsoft Basic Optical Mouse] on usb-bcm2708_usb-1.3/input0
```

Pi2 kernel messages with fiq_enable=0 and usb keyboard connected, just reboots:
```
Uncompressing Linux... done, booting the kernel.
...
[    1.673419] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
[    1.881058] Core Release: 2.80a
[    1.885693] Setting default values for core params
[    1.891993] Finished setting default values for core params
[    2.099450] Using Buffer DMA mode
[    2.104231] Periodic Transfer Interrupt Enhancement - disabled
[    2.111610] Multiprocessor Interrupt Enhancement - disabled
[    2.118738] OTG VER PARAM: 0, OTG VER FLAG: 0
[    2.124647] Dedicated Tx FIFOs mode
[    2.129981] dwc_otg bcm2708_usb: DWC OTG Controller
[    2.136400] dwc_otg bcm2708_usb: new USB bus registered, assigned bus number 1
[    2.145193] dwc_otg bcm2708_usb: irq 75, io mem 0x00000000
[    2.152236] Init: Port Power? op_state=1
[    2.157643] Init: Power Port (0)
[    2.162694] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    2.171132] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.179995] usb usb1: Product: DWC OTG Controller
[    2.186321] usb usb1: Manufacturer: Linux 3.18.11-v7+ dwc_otg_hcd
[    2.194054] usb usb1: SerialNumber: bcm2708_usb
[    2.201065] hub 1-0:1.0: USB hub found
[    2.206438] hub 1-0:1.0: 1 port detected
...
[    2.587991] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    2.596470] Indeed it is in host mode hprt0 = 00001101
[    2.798283] usb 1-1: New USB device found, idVendor=0424, idProduct=9514
[    2.806988] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.816854] hub 1-1:1.0: USB hub found
[    2.822598] hub 1-1:1.0: 5 ports detected
Uncompressing Linux... done, booting the kernel.
```

Pi2 kernel messages with fiq_enable=0 and usb mouse connected, crashes:
```
[    2.618125] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    2.626579] Indeed it is in host mode hprt0 = 00001101
[    2.828384] usb 1-1: New USB device found, idVendor=0424, idProduct=9514
[... 98 lines cut ...]
[    4.398054] [<80014a74>] (secondary_start_kernel) from [<000086a4>] (0x86a4)
[    4.407302] CPU2: stopping
[    4.412144] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G      D        3.18.11-v7+ #780
[    4.424004] [<800172bc>] (unwind_backtrace) from [<80012f7c>] (show_stack+0x20/0x24)
[    4.436029] [<80012f7c>] (show_stack) from [<80520970>] (dump_stack+0x98/0xd8)
[    4.445462] [<80520970>] (dump_stack) from [<80014f24>] (handle_IPI+0x234/0x268)
[    4.457137] [<80014f24>] (handle_IPI) from [<80008618>] (do_IPI+0x18/0x1c)
[    4.466221] [<80008618>] (do_IPI) from [<805261b4>] (__irq_svc+0x34/0x14c)
[    4.475310] Exception stack(0xb98e5f70 to 0xb98e5fb8)
[    4.482531] 5f60:                                     807d9ccc 00000000 00000000 00000000
[    4.494945] 5f80: b98e4000 00000002 807d8dd4 80817808 808128dc 807d8d74 8052b828 b98e5fc4
[    4.507353] 5fa0: b98e5fb8 b98e5fb8 8000fc78 8000fc7c 60000013 ffffffff
[    4.516136] [<805261b4>] (__irq_svc) from [<8000fc7c>] (arch_cpu_idle+0x30/0x4c)
[    4.527725] [<8000fc7c>] (arch_cpu_idle) from [<8005b690>] (cpu_startup_entry+0x1e4/0x264)
[    4.540186] [<8005b690>] (cpu_startup_entry) from [<80014a74>] (secondary_start_kernel+0x144/0x164)
[    4.553499] [<80014a74>] (secondary_start_kernel) from [<000086a4>] (0x86a4)
[    4.562753] ---[ end Kernel panic - not syncing: Fatal exception in interrupt
```

On Pi1 it crashes both with only keyboard and only mouse (but working without)
```
[    3.212733] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-bcm2708_usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:72:5d:d1
[    3.323902] usb 1-1.2: new low-speed USB device number 4 using dwc_otg
[    3.332822] Unable to handle kernel NULL pointer dereference at virtual address 0000003c
```

P33M, 2015-06-08:
https://github.com/raspberrypi/linux/commit/927804bbcde70f07217b1a5f68e6ea2da805b5e1

Should fix this issue.

popcornmix, 2015-06-08:
Cherry-picked to rpi-4.0.y tree.

notro, 2015-06-08:
With a nudge it's working with ARCH_BCM2835 also.

config: ```CONFIG_USB_DWCOTG=y```
Diff: https://gist.github.com/notro/71c6f6a39562a5ec36cc

closed 2015-06-08 by notro