raspberrypi/linux #941: bcm2835-v4l2 driver does not advertise V4L2_CAP_TIMEPERFRAME

beldenfox, 2015-04-23:
The v4l driver for the Raspberry Pi camera does not set the V4L2_CAP_TIMEPERFRAME capability bit even though it appears to correctly implement the related VIDIOC_S_PARM call. I've inspected the rpi-4.0y source on github to verify.

$ modinfo bcm2835-v4l2
filename:       /lib/modules/3.18.7-v7+/kernel/drivers/media/platform/bcm2835/bcm2835-v4l2.ko
version:        0.0.2
license:        GPL
author:         Vincent Sanders
description:    Broadcom 2835 MMAL video capture
srcversion:     36FE81D45FFC9A17867108D
depends:        videobuf2-core,videodev,videobuf2-vmalloc,v4l2-common
intree:         Y
vermagic:       3.18.7-v7+ SMP preempt mod_unload modversions ARMv7 
parm:           debug:int
parm:           bcm2835_v4l2_debug:Debug level 0-2
parm:           max_video_width:Threshold for video mode (int)
parm:           max_video_height:Threshold for video mode (int)
parm:           gst_v4l2src_is_broken:If non-zero, enable workaround for Gstreamer (int)

popcornmix, 2015-04-23:
ping @6by9

6by9, 2015-04-23:
Which capability field are you thinking of? 
Spec: http://hverkuil.home.xs4all.nl/spec/media.html#v4l2-captureparm
"Drivers support this function only when they set the V4L2_CAP_TIMEPERFRAME flag in the capability field."
https://github.com/raspberrypi/linux/blob/rpi-4.0.y/drivers/media/platform/bcm2835/bcm2835-camera.c#L1291
```
static int vidioc_g_parm(struct file *file, void *priv,
			  struct v4l2_streamparm *parm)
{
..
	parm->parm.capture.capability   = V4L2_CAP_TIMEPERFRAME;
```
So on VIDIOC_G_PARM we do set the capability flag. Compare to any other V4L2 driver, and they do likewise (mainly as I copy/pasted it from another driver!)

beldenfox, 2015-04-23:
Whoops. I'm looking at the return value from VIDIOC_QUERYCAP which is wrong. Sorry to waste your time.

closed 2015-04-23 by beldenfox

6by9, 2015-04-23:
No problem. I had to go and check the spec in detail :-)