raspberrypi/linux #942: u-boot and devicetree overlays

wbx-github, 2015-04-24:
Hi,

we are using U-Boot as secondary bootloader so that we can produce a updateable firmware
with fallback partition. When an firmware update fails, we just boot into the old partition.
For 3.12.x based kernels this worked fine, but now we are trying to use 3.18.x based with device-tree support. We have a rpi-proto chip connected via I2S to the raspberry pi board. 
U-Boot only supports a DTB file, the kernel boots up fine, but the audio chip driver does not work.

What are my best options here? Disable device-tree entirely? Add static device informations to the Raspberry PI DTB avoiding the overlay mechanism?
I got no answers from the u-boot developer mailing list, if devicetree overlay support is planned.

Thanks in advance for any advise.
best regards
 Waldemar

pelwell, 2015-04-24:
What is the firmware for? If it is essentially a consumer device with a known, static configuration then your best option is probably to create a single DTB containing all the devices you need.

wbx-github, 2015-04-24:
It is a commercial appliance with a static hardware configuration.
Okay, so I go the single DTB way. Do I need to add I2S/I2C/SPI to the DTB, too?
Are they enabled via overlays?

pelwell, 2015-04-24:
Yes, but it's trivial. For the interfaces, just locate the `status="disabled"` properties and change them to `status="okay"`.

Expanding overlays into a tree by hand is actually quite straightforward. For each fragment, use the `target=<&xxx>` or `target-path="/yyy"` properties to locate a label (xxx) or absolute path (yyy) in the base .dts file, and paste the contents of the `__overlay__` node into there, taking care to remove any existing properties with the same names.

wbx-github, 2015-04-24:
Great thanks. With your really good documentation I get it working with the first try compiling a kernel
using this patch:
diff -Nur linux-3.18.10.orig/arch/arm/boot/dts/bcm2708.dtsi linux-3.18.10/arch/arm/boot/dts/bcm2708.dtsi
--- linux-3.18.10.orig/arch/arm/boot/dts/bcm2708.dtsi   2015-04-24 17:21:52.736448819 +0200
+++ linux-3.18.10/arch/arm/boot/dts/bcm2708.dtsi        2015-04-24 17:31:43.812768908 +0200
@@ -44,7 +44,7 @@
                        //dmas = <&dma 2>,
                        //       <&dma 3>;
                        dma-names = "tx", "rx";
-                       status = "disabled";
+                       status = "okay";
                };
 
                spi0: spi@7e204000 {
@@ -54,7 +54,7 @@
                        clocks = <&clk_spi>;
                        #address-cells = <1>;
                        #size-cells = <0>;
-                       status = "disabled";
+                       status = "okay";
                };
 
                i2c0: i2c@7e205000 {
@@ -64,7 +64,7 @@
                        clocks = <&clk_i2c>;
                        #address-cells = <1>;
                        #size-cells = <0>;
-                       status = "disabled";
+                       status = "okay";
                };
 
                i2c1: i2c@7e804000 {
@@ -74,7 +74,14 @@
                        clocks = <&clk_i2c>;
                        #address-cells = <1>;
                        #size-cells = <0>;
-                       status = "disabled";
+                       status = "okay";
+
+                       wm8731@1a {
+                               #sound-dai-cells = <0>;
+                               compatible = "wlf,wm8731";
+                               reg = <0x1a>;
+                               status = "okay";
+                       };
                };
 
                leds: leds {
diff -Nur linux-3.18.10.orig/arch/arm/boot/dts/bcm2708-rpi-b.dts linux-3.18.10/arch/arm/boot/dts/bcm2708-rpi-b.dts
--- linux-3.18.10.orig/arch/arm/boot/dts/bcm2708-rpi-b.dts      2015-04-24 17:21:52.736448819 +0200
+++ linux-3.18.10/arch/arm/boot/dts/bcm2708-rpi-b.dts   2015-04-24 17:31:44.372769196 +0200
@@ -19,6 +19,9 @@
        };
 
        sound: sound {
+               compatible = "rpi,rpi-proto";
+               i2s-controller = <&i2s>;
+               status = "okay";
        };
 };
 
You made my day. Have a nice weekend.

closed 2015-05-06 by wbx-github

jampy, 2016-01-25:
Hi!

I'm having the same problem and a static DTB would be fine for me too.  Unfortunately I don't understans how to do that - could you please explain a bit more?

Thsnks a lot

pelwell, 2016-01-25:
You would be better asking this kind of question in the Device Tree section of the Raspberry Pi Forum.