raspberrypi/linux #944: BUG: workqueue leaked lock or atomic: kworker/0:2/0xffffffff/4586

marc40000, 2015-04-26:
Hi !

I get a crash every one or two days on one of my rpis. Everything works fine but at some point, I get repeating entries multiple times in /var/log/syslog and /var/log/Messages multiple times per second:

Apr 26 17:17:12 gardeneast kernel: [160403.252871] BUG: workqueue leaked lock or atomic: kworker/0:2/0xffffffff/4586
Apr 26 17:17:12 gardeneast kernel: [160403.252871]      last function: od_dbs_timer
Apr 26 17:17:12 gardeneast kernel: [160403.285169] CPU: 0 PID: 4586 Comm: kworker/0:2 Tainted: G      D W      3.18.7+ #755
Apr 26 17:17:12 gardeneast kernel: [160403.313727] Workqueue: events od_dbs_timer
Apr 26 17:17:12 gardeneast kernel: [160403.327924] [<c00151fc>] (unwind_backtrace) from [<c0012710>] (show_stack+0x20/0x24)
Apr 26 17:17:12 gardeneast kernel: [160403.356381] [<c0012710>] (show_stack) from [<c0555b30>] (dump_stack+0x20/0x28)
Apr 26 17:17:12 gardeneast kernel: [160403.374797] [<c0555b30>] (dump_stack) from [<c0039d60>] (process_one_work+0x360/0x444)
Apr 26 17:17:12 gardeneast kernel: [160403.402947] [<c0039d60>] (process_one_work) from [<c003a998>] (worker_thread+0x190/0x580)
Apr 26 17:17:12 gardeneast kernel: [160403.431557] [<c003a998>] (worker_thread) from [<c003edf8>] (kthread+0xd8/0xf4)
Apr 26 17:17:12 gardeneast kernel: [160403.449926] [<c003edf8>] (kthread) from [<c000e988>] (ret_from_fork+0x14/0x20)
Apr 26 17:17:12 gardeneast kernel: [160403.469339] BUG: workqueue leaked lock or atomic: kworker/0:2/0xffffffff/4586
Apr 26 17:17:12 gardeneast kernel: [160403.469339]      last function: BlinkWorkItemCallback [8192cu]
Apr 26 17:17:12 gardeneast kernel: [160403.503255] CPU: 0 PID: 4586 Comm: kworker/0:2 Tainted: G      D W      3.18.7+ #755
Apr 26 17:17:12 gardeneast kernel: [160403.531949] Workqueue: events BlinkWorkItemCallback [8192cu]
Apr 26 17:17:12 gardeneast kernel: [160403.547943] [<c00151fc>] (unwind_backtrace) from [<c0012710>] (show_stack+0x20/0x24)
Apr 26 17:17:12 gardeneast kernel: [160403.576332] [<c0012710>] (show_stack) from [<c0555b30>] (dump_stack+0x20/0x28)
Apr 26 17:17:12 gardeneast kernel: [160403.594707] [<c0555b30>] (dump_stack) from [<c0039d60>] (process_one_work+0x360/0x444)
Apr 26 17:17:12 gardeneast kernel: [160403.623210] [<c0039d60>] (process_one_work) from [<c003a998>] (worker_thread+0x190/0x580)
Apr 26 17:17:12 gardeneast kernel: [160403.651869] [<c003a998>] (worker_thread) from [<c003edf8>] (kthread+0xd8/0xf4)
Apr 26 17:17:12 gardeneast kernel: [160403.670260] [<c003edf8>] (kthread) from [<c000e988>] (ret_from_fork+0x14/0x20)
Apr 26 17:17:12 gardeneast kernel: [160403.688909] BUG: workqueue leaked lock or atomic: kworker/0:2/0xffffffff/4586
Apr 26 17:17:12 gardeneast kernel: [160403.688909]      last function: fb_flashcursor
Apr 26 17:17:12 gardeneast kernel: [160403.721667] CPU: 0 PID: 4586 Comm: kworker/0:2 Tainted: G      D W      3.18.7+ #755
Apr 26 17:17:12 gardeneast kernel: [160403.750935] Workqueue: events_power_efficient fb_flashcursor
Apr 26 17:17:12 gardeneast kernel: [160403.767903] [<c00151fc>] (unwind_backtrace) from [<c0012710>] (show_stack+0x20/0x24)
Apr 26 17:17:12 gardeneast kernel: [160403.796472] [<c0012710>] (show_stack) from [<c0555b30>] (dump_stack+0x20/0x28)
Apr 26 17:17:12 gardeneast kernel: [160403.815150] [<c0555b30>] (dump_stack) from [<c0039d60>] (process_one_work+0x360/0x444)
Apr 26 17:17:12 gardeneast kernel: [160403.843399] [<c0039d60>] (process_one_work) from [<c003a998>] (worker_thread+0x190/0x580)
Apr 26 17:17:12 gardeneast kernel: [160403.872901] [<c003a998>] (worker_thread) from [<c003edf8>] (kthread+0xd8/0xf4)
Apr 26 17:17:12 gardeneast kernel: [160403.891996] [<c003edf8>] (kthread) from [<c000e988>] (ret_from_fork+0x14/0x20)
Apr 26 17:17:12 gardeneast kernel: [160403.912141] BUG: workqueue leaked lock or atomic: kworker/0:2/0xffffffff/4586
Apr 26 17:17:12 gardeneast kernel: [160403.912141]      last function: BlinkWorkItemCallback [8192cu]
Apr 26 17:17:12 gardeneast kernel: [160403.946287] CPU: 0 PID: 4586 Comm: kworker/0:2 Tainted: G      D W      3.18.7+ #755
Apr 26 17:17:12 gardeneast kernel: [160403.975256] Workqueue: events BlinkWorkItemCallback [8192cu]
Apr 26 17:17:12 gardeneast kernel: [160403.991175] [<c00151fc>] (unwind_backtrace) from [<c0012710>] (show_stack+0x20/0x24)
Apr 26 17:17:12 gardeneast kernel: [160404.020100] [<c0012710>] (show_stack) from [<c0555b30>] (dump_stack+0x20/0x28)
Apr 26 17:17:12 gardeneast kernel: [160404.038625] [<c0555b30>] (dump_stack) from [<c0039d60>] (process_one_work+0x360/0x444)
Apr 26 17:17:12 gardeneast kernel: [160404.067273] [<c0039d60>] (process_one_work) from [<c003a998>] (worker_thread+0x190/0x580)
Apr 26 17:17:12 gardeneast kernel: [160404.095978] [<c003a998>] (worker_thread) from [<c003edf8>] (kthread+0xd8/0xf4)
Apr 26 17:17:12 gardeneast kernel: [160404.114522] [<c003edf8>] (kthread) from [<c000e988>] (ret_from_fork+0x14/0x20)
Apr 26 17:17:12 gardeneast kernel: [160404.133444] BUG: workqueue leaked lock or atomic: kworker/0:2/0xffffffff/4586
Apr 26 17:17:12 gardeneast kernel: [160404.133444]      last function: fb_flashcursor
Apr 26 17:17:12 gardeneast kernel: [160404.165922] CPU: 0 PID: 4586 Comm: kworker/0:2 Tainted: G      D W      3.18.7+ #755
Apr 26 17:17:12 gardeneast kernel: [160404.194346] Workqueue: events_power_efficient fb_flashcursor
Apr 26 17:17:12 gardeneast kernel: [160404.210311] [<c00151fc>] (unwind_backtrace) from [<c0012710>] (show_stack+0x20/0x24)
Apr 26 17:17:12 gardeneast kernel: [160404.239152] [<c0012710>] (show_stack) from [<c0555b30>] (dump_stack+0x20/0x28)
Apr 26 17:17:13 gardeneast kernel: [160404.257793] [<c0555b30>] (dump_stack) from [<c0039d60>] (process_one_work+0x360/0x444)
Apr 26 17:17:13 gardeneast kernel: [160404.289065] [<c0039d60>] (process_one_work) from [<c003a998>] (worker_thread+0x190/0x580)
Apr 26 17:17:13 gardeneast kernel: [160404.318293] [<c003a998>] (worker_thread) from [<c003edf8>] (kthread+0xd8/0xf4)
Apr 26 17:17:13 gardeneast kernel: [160404.337813] [<c003edf8>] (kthread) from [<c000e988>] (ret_from_fork+0x14/0x20)
Apr 26 17:17:13 gardeneast kernel: [160404.358465] BUG: workqueue leaked lock or atomic: kworker/0:2/0xffffffff/4586
Apr 26 17:17:13 gardeneast kernel: [160404.358465]      last function: od_dbs_timer
Apr 26 17:17:13 gardeneast kernel: [160404.392035] CPU: 0 PID: 4586 Comm: kworker/0:2 Tainted: G      D W      3.18.7+ #755
Apr 26 17:17:13 gardeneast kernel: [160404.420648] Workqueue: events od_dbs_timer
Apr 26 17:17:13 gardeneast kernel: [160404.436581] [<c00151fc>] (unwind_backtrace) from [<c0012710>] (show_stack+0x20/0x24)
Apr 26 17:17:13 gardeneast kernel: [160404.465130] [<c0012710>] (show_stack) from [<c0555b30>] (dump_stack+0x20/0x28)
Apr 26 17:17:13 gardeneast kernel: [160404.483737] [<c0555b30>] (dump_stack) from [<c0039d60>] (process_one_work+0x360/0x444)
Apr 26 17:17:13 gardeneast kernel: [160404.512941] [<c0039d60>] (process_one_work) from [<c003a998>] (worker_thread+0x190/0x580)
Apr 26 17:17:13 gardeneast kernel: [160404.542122] [<c003a998>] (worker_thread) from [<c003edf8>] (kthread+0xd8/0xf4)
Apr 26 17:17:13 gardeneast kernel: [160404.560879] [<c003edf8>] (kthread) from [<c000e988>] (ret_from_fork+0x14/0x20)

This continues for a few minutes and afterwards it crashes:

Apr 26 17:17:14 gardeneast kernel: [160405.761998] CPU: 0 PID: 4586 Comm: kworker/0:2 Tainted: G      D W      3.18.7+ #755
Apr 26 17:17:14 gardeneast kernel: [160405.793514] Workqueue: events od_dbs_timer
Apr 26 17:17:14 gardeneast kernel: [160405.808272] [<c00151fc>] (unwind_backtrace) from [<c0012710>] (show_stack+0x20/0x24)
[... 56 lines cut ...]
Apr 26 17:17:14 gardeneast kernel: [160405.958977] [<c001dd08>] (bcm_mailbox_property) from [<c03298c4>] (bcm2708_fb_blank+0x60/0x98)
Apr 26 17:17:14 gardeneast kernel: [160405.959001] [<c03298c4>] (bcm2708_fb_blank) from [<c03229c4>] (fb_blank+0x64/0xb8)
Apr 26 17:17:14 gardeneast kernel: [160405.959019] [<c03229c4>] (fb_blank) from [<c031e368>] (fbcon_blank+0x1f0/0x298)
Apr 26 17:17:14 gardeneast kernel: [160405.959042] [<c031e368>] (fbcon_blank) from [<c0347dd0>] (do_blank_screen+0x1c4/0x27c)
Apr 26 17:17:14 gardeneast kernel: [160405.959065] [<c0347dd0>] (do_blank_screen) from [<c034ae2c>] (console_callback+0x84/0x160)
Apr 26 17:17:14 gardeneast kernel: [160405.959091] [<c034ae2c>] (console_callback) from [<c0039b3c>] (process_one_work+0x13c/0x444)
Apr 26 17:17:14 gardeneast kernel: [160405.959111] [<c0039b3c>] (process_one_work) from [<c003a998>] (worker_thread+0x190/0x580)
[... cut ...]
Apr 26 17:17:19 gardeneast rsyslogd: [origin software="rsyslogd" swVersion="5.8.11" x-pid="2052" x-info="http://www.rsyslog.com"] start
Apr 26 17:17:19 gardeneast kernel: [    0.000000] Booting Linux on physical CPU 0x0
Apr 26 17:17:19 gardeneast kernel: [    0.000000] Initializing cgroup subsys cpu
Apr 26 17:17:19 gardeneast kernel: [    0.000000] Initializing cgroup subsys cpuacct

It's a rpi b+ with kernel 3.18.7 #755. I use the raspbian Image. There is a EDIMAX EW-7811UN that connects the rpi to the lan and Internet.

What causes this? Is it really a kernel bug? Can I do something about it?

Ruffio, 2016-08-14:
@marc40000 has your issue been resolved? If yes, then please close this issue.

marc40000, 2016-08-14:
No. I still have issues with wifi rpis crashing every few days.

popcornmix, 2016-08-15:
@marc40000 what does ``vcgencmd version`` and ``uname -a`` report?

marc40000, 2016-08-15:
pi@gardeneast ~ $  vcgencmd version
Mar 15 2016 14:47:28
Copyright (c) 2012 Broadcom
version 1bf9a9a77026af9128a339c82d72e331d3532ee4 (clean) (release)

pi@gardeneast ~ $ uname -a
Linux gardeneast 4.1.19+ #858 Tue Mar 15 15:52:03 GMT 2016 armv6l GNU/Linux

I have several rpis running and they all work without issues. Only the 2 
that are connected by wifi regularly crash. I can't get them running in 
a stable way. I updated several times and sometimes it feels like it 
crashes less often and then a few weeks later after the next update it 
feels liek it gets worse again. Could be random though. I never had a 
version that didn't crash.

marc40000, 2016-08-15:
Also, I don't know if it helps, but it feels like it always happens if there is a lot of wifi udp traffic going on.

popcornmix, 2016-08-15:
The kernel for jessie is using 4.4 tree. 
You really should upgrade if you are having problems.
I'm not saying for certain it is fixed, but we can't do anything to help problems with an old kernel/firmware.

marc40000, 2016-08-15:
Ok I'll upgrade

marc40000, 2016-09-03:
Ok I upgraded. And I still have this issue. Random crashes. twice a weak on average.

popcornmix, 2016-09-04:
What wireless devices are you using?
Are you able to connect ethernet to one of the Pi's just to confirm it is solved when wireless is not used?
An updated dmesg log would be useful (the line numbers in back trace don't match current kernel).

JamesH65, 2017-05-17:
@marc40000 There have been some fixes for wireless and USB drivers in the latest kernel (4.9). Can you try those and see if the issue is fixed?

closed 2017-12-04 by JamesH65