raspberrypi/linux #946: Commit  https://github.com/raspberrypi/linux/commit/0ed1e9cb588e is BAD

MrEngman, 2015-04-28:
re: https://github.com/raspberrypi/linux/pull/939

I have just updated my Linux source to the latest commit to compile several wifi modules for `3.18.12+ #782`. Commit https://github.com/raspberrypi/linux/commit/0ed1e9cb588e causes problems compiling all wifi modules I have tried to compile, e.g. 8188eu, 8192eu, 8812au and mt7601. With all previous 3.18 versions all compiled OK.

The items suggested by this commit are already all included in the rtl8192cu source as far as I can make out and in all the other driver sources I have.

Especially noticable, config AP_MODE generates a value CONFIG_AP_MODE which is already defined in the rtl8192cu source and all other wifi module sources I've seen and including it in Kconfig ends up with loads of errors regarding CONFIG_AP_MODE being redefined when attempting to compile any of the drivers I use.

This commit really needs removing ASAP.

popcornmix, 2015-04-29:
Ping @syntheticpp for comments.

popcornmix, 2015-05-05:
@MrEngman okay to close?

MrEngman, 2015-05-05:
I guess so.

closed 2015-05-05 by popcornmix