raspberrypi/linux #951: blk_update_request: critical target error on USB HDD after kernel upgrade

bosscyril, 2015-05-04:
As I can't reopen https://github.com/raspberrypi/linux/issues/703 so it's more "visible", I open a new thread and copy the last post. I apologize if this isn't the right way of doing this.

I think this issue have risen again, the only difference I see is that the log message changed from "end_request" to "blk_update_request", but behaviour seems the same:

```
Apr 14 13:00:39 raspberrypi kernel: [  561.226469] blk_update_request: critical target error, dev sda, sector 404211137
Apr 14 13:00:39 raspberrypi kernel: [  561.226579] Aborting journal on device sda2-8.
Apr 14 13:00:40 raspberrypi kernel: [  563.043697] EXT4-fs error (device sda2): ext4_journal_check_start:56: Detected aborted journal
Apr 14 13:00:40 raspberrypi kernel: [  563.057031] EXT4-fs (sda2): Remounting filesystem read-only
Apr 14 13:00:51 raspberrypi kernel: [  574.083665] EXT4-fs error (device sda2): ext4_put_super:789: Couldn't clean up the journal
```
If I try to fsck:

```
Apr 14 12:45:24 raspberrypi kernel: [ 1603.653290] EXT4-fs error (device sda2): ext4_put_super:789: Couldn't clean up the journal
Apr 14 12:45:30 raspberrypi kernel: [ 1610.138494] blk_update_request: critical target error, dev sda, sector 0
Apr 14 12:45:30 raspberrypi kernel: [ 1610.197294] blk_update_request: critical target error, dev sda, sector 0
Apr 14 12:45:30 raspberrypi kernel: [ 1610.203760] blk_update_request: critical target error, dev sda, sector 0
Apr 14 12:46:41 raspberrypi kernel: [ 1681.278773] blk_update_request: critical target error, dev sda, sector 0
Apr 14 12:46:41 raspberrypi kernel: [ 1681.282161] blk_update_request: critical target error, dev sda, sector 0
Apr 14 12:46:41 raspberrypi kernel: [ 1681.283998] blk_update_request: critical target error, dev sda, sector 0
```

Based on my tests, last working commit is https://github.com/Hexxeh/rpi-firmware/commit/f74b92120e0d469fc5c2dc85b2b5718d877e1cbb

Broken from https://github.com/Hexxeh/rpi-firmware/commit/cad071ac50ddf8a9f3c7a5d808cfa0daf5effb45 and onwards.

popcornmix, 2015-05-05:
Sounds like this: https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1418339
which we do have the fix for:
https://github.com/raspberrypi/linux/commit/458919c470d3316ab86f5de005bc37ca9ae32a86

I assume you are seeing this issue with latest (3.18.12) kernel?

bosscyril, 2015-05-05:
Yes, this happens with 3.8.12 kernel also
https://github.com/Hexxeh/rpi-firmware/commit/2f4703cbee9801ef04adcf63d74823961cc3c275

Ruffio, 2016-08-15:
@bosscyril has your issue been resolved? If so, please close this issue. Thanks.

bosscyril, 2016-08-16:
Yes, it has been fixed at some point. Now running 4.1.19 without issues. Closing...

closed 2016-08-16 by bosscyril