raspberrypi/linux #952: rpi-4.0.y: missing dependency for drivers/mmc/host/bcm2835-sdhost.ko

msperl, 2015-05-04:
There is a missing dependency on of_dma_request_slave_channel in drivers/mmc/host/bcm2835-sdhost.c

Which results in a compile issue:
```
  Building modules, stage 2.
  Kernel: arch/arm/boot/Image is ready
  Kernel: arch/arm/boot/zImage is ready
  MODPOST 1381 modules
ERROR: "of_dma_request_slave_channel" [drivers/mmc/host/bcm2835-sdhost.ko] undefined!
scripts/Makefile.modpost:90: recipe for target '__modpost' failed
```
at least when compiling with:
CONFIG_MMC_BCM2835_SDHOST=m

As of_dma_request_slave_channel does not get EXPORT_SYMBOL_GPL exported!

So module is not an option...

introduced with: da9969b2

pelwell, 2015-05-05:
The same issue affects bcm2835-mmc.c. dma_request_slave_channel looks like a suitable alternative - it calls of_dma_request_slave_channel and is exported - the only difference being that it takes a dev rather than a node.

@notro - do you agree?

pelwell, 2015-05-05:
@msperl Can you test and close if fixed?

msperl, 2015-05-05:
tested and fixed

closed 2015-05-05 by msperl