raspberrypi/linux #953: 4.0.y: dma-controller not found

msperl, 2015-05-05:
When running:
```dma_request_slave_channel(dev, "tx”);```

it returns NULL and I get the following with debugfs and dynamic debug enabled:
```
[  112.352434] of_dma_find_controller: can't find DMA controller /soc/dma@7e007000
[  112.352458] spi-bcm2835 3f204000.spi: no tx-dma found - not using dma mode
```

the device contains the following in the overlay:
```
                        dmas = <&dma 6>, <&dma 7>;
                        dma-names = "tx", "rx";
```

See also: https://github.com/msperl/linux-rpi/commit/00014a07bed0a3b2ed47a7b3829dda58519da4ac for the test-patch of the (still unfinished) dma support for the spi-bcm2835 driver.

here also the "effective" device-tree (via dtc -I fs /proc/device-tree), which looks ok:
```
/dts-v1/;

/ {
...
        soc {
...
                spi@7e204000 {
                        reg = <0x7e204000 0x1000>;
                        dmas = <0x3 0x6 0x3 0x7>;
                        interrupts = <0x2 0x16>;
                        pinctrl-0 = <0x6>;
                        compatible = "brcm,bcm2835-spi";
                        clocks = <0x5>;
                        status = "okay";
                        #address-cells = <0x1>;
                        phandle = <0x10>;
                        #size-cells = <0x0>;
                        dma-names = "tx", "rx";
                        pinctrl-names = "default";
                        linux,phandle = <0x10>;
...
                };
...
                dma@7e007000 {
                        compatible = "brcm,bcm2835-dma";
                        #dma-cells = <0x1>;
                        phandle = <0x3>;
                        linux,phandle = <0x3>;
                };
...
        };
...
};
```

We had something similar (but only on reload of the module) with the upstream 4.1-rc1 and this
fixed it:
https://github.com/torvalds/linux/commit/63f89caad0e32dcfa17b2d17919816253de48996

pelwell, 2015-05-05:
Have you seen @notro's patch (1c3f58d69121cbd5d90ec8f279d7831b95e93ec1
)? It rationalises the DMA driver situation, merging some and removing others. A consequence of this patch is that list of irqs must now be provided in the DTB.

See his preparatory patch (0454dbb6f0239d820c57e56aa0f6d94cd4018c6d) for an example of what is required.

notro, 2015-05-05:
I have built a rpi-4.0.y kernel with bcmrpi_defconfig.
I reverted this commit as mentioned in the comments: https://github.com/raspberrypi/linux/commit/39315a8ea76cfb3aae6853aedd409e89886d8fd4
I haven't looked more into this issue, what might be wrong. Let me know if you can do without it.

I have added the jiffies patch and your rc2 dma patch.
I patched the dts with this:
``` 
		spi0: spi@7e204000 {
			compatible = "brcm,bcm2835-spi";
			reg = <0x7e204000 0x1000>;
			interrupts = <2 22>;
			clocks = <&clk_spi>;
			#address-cells = <1>;
			#size-cells = <0>;
//			status = "disabled";

			dmas = <&dma 7>,
			       <&dma 6>;
			dma-names = "rx", "tx";
		};
```
When I boot, I get:
```
[    6.042520] spi spi0.0: setting up native-CS0 as GPIO 8
[    6.131558] spi spi0.1: setting up native-CS1 as GPIO 7
```
So it seems it could get the dma channels.
I haven't tried using the driver yet, but will do later.

notro, 2015-05-05:
Since this is DT only, I would suggest using of_dma_request_slave_channel() directly which will also give you the error code on failure.

This is the call chain on a Device Tree node:
dma_request_slave_channel() -> dma_request_slave_channel_reason() -> of_dma_request_slave_channel()

pelwell, 2015-05-05:
But as Martin has found (https://github.com/raspberrypi/linux/issues/952), `of_dma_request_slave_channel` isn't an exported symbol.

notro, 2015-05-05:
Yes you're right. Sorry I missed that.
I can't see any in-tree users of of_dma_request_slave_channel(), so it's intented to be internal to the subsystem I guess: http://lxr.free-electrons.com/ident?i=of_dma_request_slave_channel

pelwell, 2015-05-05:
Isn't the problem here that Martin is using a custom DTB that doesn't include the irq data for the dma controller?

notro, 2015-05-05:
You're right again. I should not try and give answer 5 minutes after I got out of bed :-)

@msperl there will still be changes to the Device Tree files in the coming weeks, so make sure you also use the latest DT when you try the kernel.

pelwell, 2015-05-05:
I re-learnt yesterday that github and childcare don't mix well. :-)

msperl, 2015-05-05:
isn't DT supposed to be an API, so anything older should also work?
But yes: with the original DT it works.

closed 2015-05-05 by msperl

notro, 2015-05-05:
> isn't DT supposed to be an API, so anything older should also work?

Yes in mainline it's considered to ABI.
But it's impossible for us to enforce this, since we are in the process of moving from devices added in the platform file to adding them in the Device Tree instead.
If we were to support older DTB's, we would end up with several devices being added twice, cluttering up the boot messages.
And as @popcornmix previously pointed out, this will bite developers only. End users always get a matching DTB.

pelwell, 2015-05-05:
I agree it's not ideal, but I think the end justifies the means.