raspberrypi/linux #956: Unable to enumerate USB device - microphone SF-555 L5YG

exander77, 2015-05-05:
I bought USB microphone SF-555 L5YG. On linux desktop it works without problems, on Raspberry Pi Model 2 updated (rpi-update) it gives in dmesg:

```
[   72.007935] usb 1-1.5: new full-speed USB device number 9 using dwc_otg
[   72.087942] usb 1-1.5: device descriptor read/64, error -32
[   72.277952] usb 1-1.5: device descriptor read/64, error -32
[   72.467959] usb 1-1.5: new full-speed USB device number 10 using dwc_otg
[   72.547921] usb 1-1.5: device descriptor read/64, error -32
[   72.737939] usb 1-1.5: device descriptor read/64, error -32
[   72.927913] usb 1-1.5: new full-speed USB device number 11 using dwc_otg
[   73.347828] usb 1-1.5: device not accepting address 11, error -32
[   73.427941] usb 1-1.5: new full-speed USB device number 12 using dwc_otg
[   73.847848] usb 1-1.5: device not accepting address 12, error -32
[   73.848181] usb 1-1-port5: unable to enumerate USB device
```
With dwc_otg.speed=1 it gives:
```
[   40.518008] usb 1-1.5: new full-speed USB device number 9 using dwc_otg
[   40.937794] usb 1-1.5: device not accepting address 9, error -71
[   41.018019] usb 1-1.5: new full-speed USB device number 10 using dwc_otg
[   41.437797] usb 1-1.5: device not accepting address 10, error -71
[   41.518036] usb 1-1.5: new full-speed USB device number 11 using dwc_otg
[   41.598006] usb 1-1.5: device descriptor read/64, error -71
[   41.788016] usb 1-1.5: device descriptor read/64, error -71
[   41.978046] usb 1-1.5: new full-speed USB device number 12 using dwc_otg
[   42.058022] usb 1-1.5: device descriptor read/64, error -71
[   42.248016] usb 1-1.5: device descriptor read/64, error -71
[   42.358039] usb 1-1-port5: unable to enumerate USB device
```

exander77, 2015-05-05:
On Raspberry Pi Model B not working either.

Ruffio, 2016-08-14:
@exander77 has your issue been resolved? If yes, then please close this issue.

P33M, 2017-05-04:
Closing USB issues as OP has not updated the thread. Further comments by the OP stating that the issue is still present on latest rpi-update kernel will lead to a review.

closed 2017-05-04 by P33M