raspberrypi/linux #957: Star printer SP500 - usblp hangs

exander77, 2015-05-05:
I have this printer, when I write to /dev/usb/lp0, it works for a few times and then stop working. I have to turn off and on the printer, then it continue printing.

Sometimes unloading and reloading usblp module works.

P33M, 2015-05-06:
Can you capture the output of `lsusb -v` and `dmesg` after the device has "broken" and post it here?

exander77, 2015-05-06:
There is no message in dmesg.

```
Bus 001 Device 008: ID 10d5:55a4 Uni Class Technology Co., Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x10d5 Uni Class Technology Co., Ltd
  idProduct          0x55a4 
  bcdDevice            1.00
  iManufacturer           1 No brand
  iProduct                2 4 Port KVMSwicther
  iSerial                 3 2t?
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          4 HID Mouse
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              7 Keyboard
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      65
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              6 Mouse
        HID Device Descriptor:
[... 592 lines cut ...]
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0008
    Ganged power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        1 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered
```

Ruffio, 2016-08-13:
@exander77 has this issue been resolved? If it has, then please close this issue.

exander77, 2016-08-13:
@Ruffio  Sorry, I cannot verify, I no longer have this model.

Ruffio, 2016-08-13:
@exander you didn't test the fix? Anyway, you are the only one with this issue. It should be closed and if someone experience this again, it can be reopened.

exander77, 2016-08-13:
@Ruffio No, I could not make it work, so I bought different printer. You are probably right.

closed 2016-08-14 by popcornmix