raspberrypi/linux #961: Kernel BUG in drivers/video/fbdev/bcm2708_fb.c:329!

huaraz, 2015-05-08:
Hi

  I upgraded my with a new kernel und run now into the following issue:

raspberrypi login: ▒▒[225060.073616] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[225061.883660] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[225061.925903] systemd-journald[88]: Received SIGTERM from PID 1 (systemd-shutdow).
[225061.983990] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[225062.018718] systemd-shutdown[1]: Unmounting file systems.
[225062.026081] systemd-shutdown[1]: Unmounting /sys/fs/fuse/connections.
[225062.033154] systemd-shutdown[1]: Unmounting /sys/kernel/config.
[225062.039323] systemd-shutdown[1]: Unmounting /dev/mqueue.
[225062.045006] systemd-shutdown[1]: Unmounting /sys/kernel/debug.
[225062.106181] EXT4-fs (sda1): re-mounted. Opts: (null)
[225062.117704] EXT4-fs (sda1): re-mounted. Opts: (null)
[225062.122979] EXT4-fs (sda1): re-mounted. Opts: (null)
[225062.128117] systemd-shutdown[1]: All filesystems unmounted.
[225062.133927] systemd-shutdown[1]: Deactivating swaps.
[225062.139474] systemd-shutdown[1]: All swaps deactivated.
[225062.145098] systemd-shutdown[1]: Detaching loop devices.
[225062.190936] systemd-shutdown[1]: All loop devices detached.
[225062.196840] systemd-shutdown[1]: Detaching DM devices.
[225062.203056] systemd-shutdown[1]: All DM devices detached.
[225062.260208] systemd-shutdown[1]: Rebooting.
[225062.265938] reboot: Restarting system
Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.18.7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #755 PREEMPT Thu Feb 12 17:14:31 GMT 2015
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine model: Raspberry Pi Model B
[    0.000000] early_vc_cma_mem(0/0x4c00000@0xa000000)
[    0.000000]  -> initial 0, size 4c00000, base a000000[    0.000000] cma: Reserved 76 MiB at 0x0a000000
[    0.000000] cma: Reserved 8 MiB at 0x08800000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 55848
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708.boardrev=0x2 bcm2708.serial=0x96445531 smsc95xx.macaddr=B8:27:EB:44:55:31 bcm2708_fb.fbswap=1 sdhci-bcm2708.emmc_clock_freq=250000000 vc-cma-mem=0/0x4c00000@0xa000000 mem=0x9000000@0x0 mem=0x4c00000@0xa000000 vc_mem.mem_base=0xec00000 vc_mem.mem_size=0x10000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/sda1     rootfstype=ext4 elevator=deadline rootwait rootdelay=5 coherent_pool=6M  smsc95xx.turbo_mode=N
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 127784K/225280K available (5926K kernel code, 358K rwdata, 1876K rodata, 340K init, 734K bss, 97496K reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
[    0.000000]     vmalloc : 0xcf000000 - 0xff000000   ( 768 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xcec00000   ( 236 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc07a6ad8   (7803 kB)
[    0.000000]       .init : 0xc07a7000 - 0xc07fc000   ( 340 kB)
[    0.000000]       .data : 0xc07fc000 - 0xc085588c   ( 359 kB)
[    0.000000]        .bss : 0xc085588c - 0xc090d128   ( 735 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:522
[    0.000024] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483648000ns
[    0.000075] Switching to timer-based delay loop, resolution 1000ns
[    0.000354] Console: colour dummy device 80x30
[    0.001536] console [tty1] enabled
[    0.001581] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.001657] pid_max: default: 32768 minimum: 301
[    0.002030] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.002095] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.003093] Initializing cgroup subsys memory
[    0.003185] Initializing cgroup subsys devices
[    0.003244] Initializing cgroup subsys freezer
[    0.003297] Initializing cgroup subsys net_cls
[    0.003349] Initializing cgroup subsys blkio
[    0.003474] CPU: Testing write buffer coherency: ok
[... 184 lines cut ...]
    #sdtv_mode=2

    #uncomment to overclock the arm. 700 MHz is the default.
    #arm_freq=800

    #memory split options
    gpu_mem_256=112
    gpu_mem_512=368
    cma_lwm=16
    cma_hwm=32
    cma_offline_start=16

    # for more options see http://elinux.org/RPi_config.txt
    gpu_mem=64
     

cmdline.txt was:

dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/sda1   rootfstype=ext4 elevator=deadline rootwait rootdelay=5 coherent_pool=6M  smsc95xx.turbo_mode=N

The last kernel working for me was:

Linux raspberrypi 3.12.32+ #721 PREEMPT Fri Nov 7 16:50:31 GMT 2014 armv6l GNU/Linux

Markus

notro, 2015-05-08:
@popcornmix 
This commit https://github.com/notro/linux/commit/5b103c323095098b1170ec50bb32a8e0bc555ede from #949 removes that nasty BUG statement in the fbdev driver letting the kernel continue booting when it fails probing.

pelwell, 2015-05-08:
That's an old kernel and firmware. There have been several CMA-related changes since then - you could try a "sudo rpi-update" after you've temporarily commented out the CMA settings. But to keep things simple, try adding "cma=0" to your cmdline.txt and rebooting.

huaraz, 2015-05-08:
When you say old kernel do you mean 
Linux raspberrypi 3.12.32+ #721 PREEMPT Fri Nov 7 16:50:31 GMT 2014 armv6l GNU/Linux ?

The latest I tested and from which the kernel bug messages is is 

 3.18.7-v7+

Markus

pelwell, 2015-05-08:
Both. The rpi-3.18.y tree is up to 3.18.11.

huaraz, 2015-05-08:
Ok. I did a rpi-update and added cma=0 which worked.

Thank you
Markus

closed 2015-05-08 by huaraz