raspberrypi/linux #962: WiFi driver issues after kernel: Bump to 3.18.4 Jan 28

pvouzis, 2015-05-11:
Hi,

I am using wpa_supplicant to connect my RPi to different access points. I had no problem connecting to 2.4 GHz and 5GHz AP up until version 3.12.36 Jan 14 (https://github.com/Hexxeh/rpi-firmware/commits/master) 

After moving to 3.18.4 Jan 28 weird things started happening (some good):
1) I can't connect to 5.4 GHz AP
2) iwlist wlan0 scan detects some 5GHz AP (but fewer than version 3.12.36)
3) (The good one): The metrics of signal strength and link quality reported by iwconfig are much closer to values I get by using other devices to measure WiFi signal strength.

Given that I see a note kernel on Jan 23 (Update with addtional modules (including brcmfmac wifi driver popcornmix) I suspect that something happened between 3.12.36 and 3.18.4 that messed up the WiFi drivers.

I am using an ASUS N-53 dongle.

It's great that the new drivers report far more accurate metrics, but not being able to connect to certain AP (that I could earlier) is a big problem.

If you need more information for troubleshooting please let me know.

Thanks,
Panos

P.S. This may be related to a previous issue: https://github.com/raspberrypi/linux/issues/920

popcornmix, 2015-05-12:
Can you identify the breaking update exactly?
There are 8 updates between Jan 14 and Jan 28.

pvouzis, 2015-05-12:
It's the "3.18 kernel with device tree enabled … Jan 20 ( cad071a)"

It looks like the driver cannot "see" 5.4GHz channels above a certain threshold. 

To test that you can do a 

sudo iwlist wlan0 scan 

on a version earlier that the one above and most likely you will see SSIDs with channels above 150. E.g. in my office I can see channels 153 and above. Then, when I switch to version cad071a I can see see up to channel 108 (that may be not the exact cut off, but it's what I see at my office with around 30 access points around me).

So, the driver might be able to connect to some 5GHz channels, but up to a certain threshold.

popcornmix, 2015-05-13:
What is the output of:

```
lsmod
ldusb
```

pvouzis, 2015-05-13:
Have you confirmed that the problem is also there for you?  Do you have the same WiFi dongle I have (ASUS N-53), by any chance?

pi@ :~:$ lsmod
Module                  Size  Used by
sg                     21787  0
ctr                     3993  1
ccm                     8238  1
bcm2708_wdog            3894  1
snd_bcm2835            21342  0
snd_pcm                93100  1 snd_bcm2835
snd_seq                61097  0
snd_seq_device          7209  1 snd_seq
snd_timer              23007  2 snd_pcm,snd_seq
snd                    67211  5 snd_bcm2835,snd_timer,snd_pcm,snd_seq,snd_seq_device
arc4                    1964  2
rt2800usb              18970  0
rt2800lib              81833  1 rt2800usb
rt2x00usb              12510  1 rt2800usb
rt2x00lib              48797  3 rt2x00usb,rt2800lib,rt2800usb
mac80211              557746  3 rt2x00lib,rt2x00usb,rt2800lib
cfg80211              471916  2 mac80211,rt2x00lib
crc_ccitt               1612  1 rt2800lib
rfkill                 22347  2 cfg80211
uio_pdrv_genirq         3666  0
uio                     9897  1 uio_pdrv_genirq
pi@ :~:$ lsusb
Bus 001 Device 002: ID 0424:9512 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp.
Bus 001 Device 004: ID 0b05:179d ASUSTek Computer, Inc. USB-N53 802.11abgn Network Adapter [Ralink RT3572]

P.S. Maybe of help: we had a long discussion with MrEngmann for driver issues on this post: https://github.com/raspberrypi/linux/issues/920

popcornmix, 2015-05-13:
The point where the behaviour changed was a move from 3.12 to 3.18 kernel.
So this isn't something in our pi specific code, but an upstream change.

It would be interesting to see if you get the same behaviour on another linux platform that uses 3.18 kernel. It may be worth searching around for similar reports for this dongle on other linux platforms. We are unlikely to be able to fix this issue ourselves, but can pull in a patch if one is found.

If you can build your own kernel, it might be interesting to try the rpi-4.0.y branch (i.e. linux 4.0.2).

pvouzis, 2015-05-13:
Unfortunately, I don't have access to any other Linux machine with 3.18.

Also, if there is a way to download a precompiled rpi-4.0 I could easily try it. Can I find an image somewhere?

pvouzis, 2015-06-12:
I tried it on a BananaPi with kernel 3.19. Does that mean that when 3.19 get released for RPi it will work? Any idea about the release timeline of 3.19?

popcornmix, 2015-06-12:
We will be moving to 4.0 next. Probably within the next few weeks.

pvouzis, 2015-06-12:
It looks like the instructions at https://www.raspberrypi.org/documentation/linux/kernel/building.md are not valid any more.

From the commands:
$ make zImage modules dtbs
$ sudo make modules_install
$ sudo cp arch/arm/boot/dts/*.dtb /boot/
$ sudo cp arch/arm/boot/dts/overlays/*.dtb* /boot/overlays/
$ sudo scripts/mkknlimg arch/arm/boot/zImage /boot/$KERNEL.img

The first 3 work fine. But #4 says that the "overlays" doesn't exist, and script "mkknlimg" is removed after kernel 3.19. Where can I find the latest compilation and update procedure?

Thanks!

pelwell, 2015-06-13:
Those steps are correct for the latest kernels (top of tree on 3.18.y and 4.0.y). For older kernels it is slightly more awkward, since there are two different sets of .dtb files in the same directory that have to go to different destinations (hence the reorganisation).

Try
```
$ sudo cp arch/arm/boot/dts/bcm*.dtb /boot/
$ sudo cp arch/arm/boot/dts/overlays/*-overlay.dtb /boot/overlays/
```

pvouzis, 2015-06-13:
Just compiled kernel 4.0 and installed it without any problems. 

More importantly, the problem with the WiFi is fixed.

Is there a way to find where this problem on the upstream came from? If we know were it came from, it won't be repeated.

Thanks!

P.S. With 3.19.y the directory "arch/arm/boot/dts/overlays/" doesn't even exist.

pelwell, 2015-06-13:
That's correct. We'll be leapfrogging 3.19 and going straight to 4.0.

pvouzis, 2015-06-13:
Found another issue, which is not a biggie, but good to know about it since it may have implications. It has to do with the format of the output.

If you do a iwlist wlan0 scan here is the output you should get for each access point (truncated):
          Cell 01 - Address: 54:E4:3A:E8:47:48
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=67/70  Signal level=-43 dBm
                    Encryption key:on
                    ESSID:"Rudolf Media"

Now, for channels above 108 you get this:
          Cell 06 - Address: 44:94:FC:3F:83:2A
                    Channel:153
                    Frequency:5.765 GHz
                    Quality=41/70  Signal level=-69 dBm
                    Encryption key:on
                    ESSID:"NETGEAR53-5G"

Notice that on the "Frequency" line there is no display of the channel. It is only displayed on the previous line.

Again, not a biggie because the info is there (and can also be derived from the frequency), but it may break some people's scripts if they are parsing this output. 

However, it's a big improvement because before rpi-4.0.y channels above 108 weren't even detected.

pvouzis, 2015-06-15:
Here is another change in the output that might break parsing scripts (It broke mine):

Before 4.0 the command "iwconfig wlan0" would give the bit rate as follows:
Bit Rate=65 Mb/s

On 4.0 the same command gives:
Bit Rate:65 Mb/s

Thanks

popcornmix, 2015-06-16:
These are really questions for the upstream kernel. We haven't touched the drivers apart from updating to newer upstream versions. It is certainly not our place to meddle with this.

I'm afraid you'll just have to support either format in your script.

closed 2017-05-17 by JamesH65