raspberrypi/linux #963: SSH prompt lags with Raspian 2015-05-05

typhoon71, 2015-05-13:
With last Raspian, image 2015-05-05, there is a noticeable lag during ssh prompt for password, that previous images didn't exibit. 
It's something like 3 seconds, when it used to be istant ready. Once logged it's fine, no lag.
Btw, it's only ssh access that has this lag, the rest I use is fine (like transmission-daemon RPC)
My setup didn't change, and it seems other ppl noticed it (https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=109781&start=50)
I'm running it on a headless RPi, plugged to one LAN port of the router and connecting via its IP.
Both cli from linux distros or putty exibit this issue.

typhoon71, 2015-05-14:
I think avahi-daemon might be the culprit, if I do:
"sudo service avahi-daemon stop"
the lags disappears, whith no apparent loss of functionality (headless rpi).
But I suppose there was a reason for it to be there, so it'll be better someone more expert in the matter checks it.

XECDesign, 2015-05-14:
You're right. Google turns up a lot of results for avahi ssh lag.

typhoon71, 2015-05-14:
Didn't think of googling avahi-daemon...
I found that changing a line in /etc/nsswitch.conf fixes it, with avahi-daemon running.

"hosts: files mdns4_minimal [NOTFOUND=return] dns mdns4"

should become:

"host: files dns"

Restarting avahi-daemon service fixes it. No need to reboot (works after reboot anyway, tested)

Now while this fixes what I want fixed I suppose it would be good to check if nothing is broken.
If mdns4 is what I think (related to wins) maybe setting a short timeout could help.

XECDesign, 2015-05-14:
So here's the relevant debian bug report:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=414569
Fixed in version nss-mdns/0.10-4
Jessie is on 0.10-6
Wheezy is on 0.10-3.2
The fix in 0.10-4 is:
  * On new installations, do not add "mdns4" to nsswitch.conf, only
    "mdns4_minimal [NOTFOUND=return]". This means we can't
    perform reverse DNS using mDNS for addresses outside 169.254.x.x and
    fe80::/10, but avoids a 5 second delay if such addresses do not
    have a PTR record in DNS (Closes: #412714, #414569, #561622 for
    new installations).

Closing this as 'fixed', since the next image should be a jessie one. If the next image is still wheezy for whatever reason, I'll add that fix in. Thanks for reporting and looking into this.

closed 2015-05-14 by XECDesign

typhoon71, 2015-05-14:
OK.