raspberrypi/linux #964: Pi2 stuck at 600mhz

ric96, 2015-05-14:
i am running raspbian on my pi 2 and it never seems to reach 900mhz and is stuck at 600, power supply is a 2.1A capable of running a hdd.
i have performed cpu intensive tasks like BOINC and compiling tangram with -j4, all of which result in a 100% cpu usage but frequency is 600mhz across all cores.

pelwell, 2015-05-14:
This isn't a kernel issue, but a help request. Please search the forums for "overclocking", like this:

https://www.raspberrypi.org/forums/search.php?keywords=overclocking

closed 2015-05-14 by pelwell

ric96, 2015-05-14:
hmm... its not about overclocking as it turns out on a pi 2 the cpu will be stuck at 600mhz until a package "cpufrequtils" is installed, only then it will scale as it should.
this probably happens because the cpu governor is set to power save by default which makes little sense as it will almost never let the cpu run at full speed

pelwell, 2015-05-14:
It's a Raspbian issue.

On a Pi 1, the default CPU frequencies are min=700, max=700. Therefore to make use of the ondemand or performance governor you have to overclock. If you use raspi-config to enable overclocking then it will also enable the raspi-config service (sudo service raspi-config start) which switches to the ondemand governor (unless Shift is held during boot).

On a Pi 2, the default frequencies are min=600, max=900. So it makes sense to always start the raspi-config service on a Pi 2 (or to change the kernel default).

@XECDesign is looking at the best way to resolve this.

XECDesign, 2015-05-14:
The thing about it being enabled if you overclock seems to be wrong. I'll post back here when I get to the bottom of this.

pelwell, 2015-05-14:
Also being discussed on the Forums: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=110464&p=758755#p758755

XECDesign, 2015-05-14:
should be fixed now, please run:
```
sudo apt-get update
sudo apt-get upgrade
sudo reboot
```

ric96, 2015-05-14:
yes, that works very well..
Thank You