raspberrypi/linux #968: Please enable CONFIG_CIFS_UPCALL

mruzicka, 2015-05-17:
* What led up to the situation?
  I wanted to mount a CIFS share using kerberos authentication.

* What exactly did you do (or not do) that was effective (or ineffective)?
  ```
  mount -t cifs -osec=krb5 //intranet/videos /mnt
  ```

* What was the outcome of this action?
  The following errors messages were printed in the console:
  ```
  mount error(38): Function not implemented
  Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
  ```

  In addition to that dmesg contained this information:
  ```
  CIFS VFS: Kerberos negotiated but upcall support disabled!
  CIFS VFS: Send error in SessSetup = -38
  CIFS VFS: cifs_mount failed w/return code = -38
  ```

* What outcome did you expect instead?
  Successful completion of the mount command without any error messages.

popcornmix, 2015-05-18:
Added in latest rpi-update firmware. Please test.

mruzicka, 2015-05-19:
Works for me:
```
root@videoberry:~# cat /proc/version 
Linux version 3.18.13+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #785 PREEMPT Mon May 18 17:39:07 BST 2015
root@videoberry:~# /bin/mount -t cifs -o sec=krb5 //intranet/videos /mnt
root@videoberry:~# cat /proc/mounts 
rootfs / rootfs rw 0 0
/dev/root / ext4 rw,noatime,data=ordered 0 0
devtmpfs /dev devtmpfs rw,relatime,size=218620k,nr_inodes=54655,mode=755 0 0
tmpfs /run tmpfs rw,nosuid,noexec,relatime,size=44580k,mode=755 0 0
tmpfs /run/lock tmpfs rw,nosuid,nodev,noexec,relatime,size=5120k 0 0
proc /proc proc rw,nosuid,nodev,noexec,relatime 0 0
sysfs /sys sysfs rw,nosuid,nodev,noexec,relatime 0 0
tmpfs /run/shm tmpfs rw,nosuid,nodev,noexec,relatime,size=89140k 0 0
devpts /dev/pts devpts rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000 0 0
/dev/mmcblk0p1 /boot vfat rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro 0 0
//intranet/videos /mnt cifs rw,relatime,vers=1.0,sec=krb5,cache=strict,username=root,uid=0,noforceuid,gid=0,noforcegid,addr=192.168.1.4,unix,posixpaths,serverino,acl,rsize=1048576,wsize=65536,actimeo=1 0 0
```
Thank you!
When will there be an updated raspberrypi-bootloader deb package? :wink:

popcornmix, 2015-05-20:
@XECDesign may be able to answer that.

XECDesign, 2015-05-20:
There aren't any plan to bump the bootloader package before the jessie image, which is still some time off. In the meanwhile, if you need the latest firmware in a .deb package, you could use this repo http://nightly.raspberrypi.org/nightlyfirmware-v2/pool/main/r/raspberrypi-firmware/

mruzicka, 2015-05-21:
OK, many thanks.

closed 2015-05-21 by mruzicka