raspberrypi/linux #971: Crackling sound on USB soundcard

ArthurJam, 2015-05-19:
Hello,

I have a raspberry pi 2 with a USB soundcard and i can't get clean sound on it.
After many search and meny try, i dont know what to do.
If someone can help me, i would appreciate.

I post an audio file to show you the problem : https://clyp.it/j3hnfvvt

```
pi@raspberrypi ~ $ lsusb
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp.
Bus 001 Device 004: ID 08bb:2704 Texas Instruments Japan
Bus 001 Device 005: ID 148f:7601 Ralink Technology, Corp.
```

```
pi@raspberrypi ~ $ aplay -l
**** Liste des Périphériques Matériels PLAYBACK ****
carte 0: DAC [USB Audio DAC], périphérique 0: USB Audio [USB Audio]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0
carte 1: ALSA [bcm2835 ALSA], périphérique 0: bcm2835 ALSA [bcm2835 ALSA]
  Sous-périphériques: 8/8
  Sous-périphérique #0: subdevice #0
  Sous-périphérique #1: subdevice #1
  Sous-périphérique #2: subdevice #2
  Sous-périphérique #3: subdevice #3
  Sous-périphérique #4: subdevice #4
  Sous-périphérique #5: subdevice #5
  Sous-périphérique #6: subdevice #6
  Sous-périphérique #7: subdevice #7
carte 1: ALSA [bcm2835 ALSA], périphérique 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0
```

```
pi@raspberrypi ~ $ uname -a
Linux raspberrypi 3.18.13-v7+ #784 SMP PREEMPT Sat May 9 15:57:36 BST 2015 armv7l GNU/Linux
```

Thank you.

ArthurJam, 2015-05-24:
I dont know why but the sound going more bad when i touch the piscreen.
I got more noise and crack.

notro, 2015-05-24:
Since you mention piscreen, I assume you use the fbtft drivers. This is a similar case: https://github.com/raspberrypi/linux/issues/888#issuecomment-81884235

notro, 2015-05-24:
You can verify if this is the problem by not loading the piscreen device.

ArthurJam, 2015-05-25:
Hello,

Yes i use the fbtft driver.
I still have the problem when i remove the piscreen.
But the fbtft driver us maybe loaded.
Can you give me a method to unload the driver correctly.

Thx notro for your answer and for yours works.

notro, 2015-05-25:
> Can you give me a method to unload the driver correctly.

That depends on how you load it.
If you use a Device Tree overlay, just uncomment it in /boot/config.txt
If you use fbtft_device, uncomment it in /etc/modules or wherever it is loaded.
Verify that the driver is not loaded by looking in the kernel log: ```dmesg```
This line should not be present:
```
graphics fb1: XXXX frame buffer, XXX KiB video memory, ...
```

Or you can unload the module after it is loaded:
fbtft_device: ```sudo modprobe -r fbtft_device```
DT overlay: ```sudo modprobe -r fb_xxxx``` (use lsmod to find the name)

ArthurJam, 2015-05-25:
Thx i will try.

ArthurJam, 2015-05-25:
The sound going better after unload the fbtft.
It s not perfect when i stress the raspberry like starting a browser but the sound is more clear.
Do you know how i can use my piscreen, it s for a personnal project ?

notro, 2015-05-25:
You can try txbuflen=128 (256,512,1024,2048). If you use fbtft_device, you can use it directly as an argument.
If you use a DT overlay, you need to customize the source and add: txbuflen = <128>;
Lowering this value will slow down your display.

You will see that the change has taken effect in the log:
graphics fb1: XXX frame buffer, X KiB video memory, **X KiB buffer memory**,

https://github.com/notro/fbtft/wiki/fbtft_device
https://www.raspberrypi.org/documentation/configuration/device-tree.md

ArthurJam, 2015-05-25:
Ok, i will try to make my own overlay.
I found the solution on this page : https://github.com/notro/fbtft/wiki/FBTFT-RPI-overlays

Can you tell me in which section I must add this line : txbuflen = <128>;
My overlay will be based on : https://github.com/notro/fbtft/blob/master/dts/overlays/rpi/piscreen-overlay.dts

Thx.

notro, 2015-05-25:
Add it alongside fps, buswidth and regwidth:
```
			piscreen: piscreen@0{
				compatible = "ilitek,ili9486";
				reg = <0>;
				pinctrl-names = "default";
				pinctrl-0 = <&piscreen_pins>;

				spi-max-frequency = <32000000>;
				rotate = <270>;
				bgr;
				fps = <30>;
				buswidth = <8>;
				regwidth = <16>;
				txbuflen = <128>;
```
If you add this to overrides (DT overlay argument), then you can tweak the value easily:
```
	__overrides__ {
[...]
		txbuflen =		<&piscreen>,"txbuflen:0";
	};
```
dtoverlay=piscreen,txbuflen=256

ArthurJam, 2015-05-25:
Ok,
I make the custom overlay and i have a better sound now.
I try with multiple txbuflen value.
Do you know what the default value with piscreen orginal file ?

notro, 2015-05-25:
The fbtft default is 4k=4096. txbuflen is the size of the transmit buffer, how much is transferred in one SPI message.

ArthurJam, 2015-05-25:
Thanks.
Is it a material restriction ?
Or should I have a hope to get a greater buffer one day ?

closed 2015-05-25 by ArthurJam

notro, 2015-05-25:
It's probably a buffer underun in the usb audio device. But I know to little about the processor internals to say anything about where the bottleneck is.