raspberrypi/linux #972: Failure to clean up during upgrade of cgroups-bin breaks systemd

vasyugan, 2015-05-19:
Issue was originally reported to Debian, however, it is raspbian specific, see 

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=785484

Essentially, cgroups-bin, which is a transitional package in Jessie, still leaves /etc/init.d/cgroup-bin around. This script tries to mount /sys/fs/cgroup/memory but seems to fail and it then unmounts /sys/fs/cgroup.

Therefore systemd cannot create new cgroups or spawning processes cannot attach to them.

XECDesign, 2015-05-19:
Unless they are foundation-repo or spindle specific, Raspbian bugs generally go here. https://bugs.launchpad.net/raspbian
This github repo is for the kernel only.

However, I'm struggling to understand how this isn't an upstream bug. What is raspbian doing differently from debian that makes this a raspbian bug rather then a debian bug?

vasyugan, 2015-05-19:
The file in question was never shipped by Debian, see here: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=785484#76

XECDesign, 2015-05-19:
Ah then yes, this is our problem. Thanks.

XECDesign, 2015-05-19:
On the other hand, we don't pretend to support jessie/systemd yet. so This will have to wait a bit.

XECDesign, 2015-05-19:
Could you please open the issue here? https://github.com/RPi-Distro/repo/issues

closed 2015-05-19 by XECDesign