raspberrypi/linux #983: How to download linux 3.18.11 kernel?

abhi-84, 2015-05-25:
Hi
For some application, i need to have linux 3.18.11 kernel but currently it is giving 3.8.14 kernel. How can I get my desired version number ?
please help

6by9, 2015-05-25:
Look at https://github.com/Hexxeh/rpi-firmware/commits/master and you'll see commit 4e2845e0cbe7e701f0431e66a26f730de264e7a6 was the first that jumped to 3.18.11. 

```sudo rpi-update 4e2845e0cbe7e701f0431e66a26f730de264e7a6``` 
will sync your Pi to that commit. You could adopt any of the 5 other commits on 3.18.11 if you need any of the features those added.

mavirroco, 2015-07-16:
any ideas how to install kernel-devel for this one 3.18.11+ ?

6by9, 2015-07-17:
@mavirroco There isn't a kernel-devel package for Raspbian - you're duplicating https://github.com/raspberrypi/linux/issues/1044, or https://github.com/raspberrypi/linux/issues/382

You can get the full kernel source from this repo, and that is a superset of everything that would be in kernel-devel.

closed 2015-07-18 by popcornmix